margelo/react-native-vision-camera #2536: 🐛 App crash VisionCameraScheduler.trigger, com.facebook.jni.CppException - std::exception

hardikhere, 2024-02-10:
### What's happening?

App is crashing at Fatal Exception: com.facebook.jni.CppException: std::exception

### Reproduceable Code

```tsx
const REMOVE_INFRACTION_BRANDS = [
  'samsung',
  'oneplus',
  'redmi',
  'mi',
  'oppo',
  'xiaomi',
  'vivo',
];

  const format = getCameraFormat(device, [
    {videoResolution: {width: 480, height: 720}},
    {fps: 30},
  ]);

  const checkDeviceOEM = async () => {
    try {
      const deviceOem = DeviceInfo.getBrand();

      if (REMOVE_INFRACTION_BRANDS.includes(deviceOem)) {
        setShouldEnableFrameProccessing(false);
      }
    } catch (err) {}
  };

        <Camera
          ref={camera}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isFocussed}
          onInitialized={onInitialized}
          onError={onError}
          enableZoomGesture={false}
          photo={true}
          video={true}
          audio={false}
          format={format}
          pixelFormat="yuv"
          orientation="portrait"
          enableHighQualityPhotos={false}
          frameProcessor={shouldEnableFrameProccessing ? frameProcessor : null}
        />
```

### Relevant log output

```shell
Fatal Exception: com.facebook.jni.CppException: std::exception
       at com.mrousavy.camera.frameprocessor.VisionCameraScheduler.trigger(VisionCameraScheduler.java)
       at com.mrousavy.camera.frameprocessor.VisionCameraScheduler.$r8$lambda$Rh5vFX2pR6HZnaz5K-D0xg2S108()
       at com.mrousavy.camera.frameprocessor.VisionCameraScheduler$$ExternalSyntheticLambda0.run(:2)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loopOnce(Looper.java:233)
       at android.os.Looper.loop(Looper.java:344)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
const devices = Camera.getAvailableCameraDevices();
  const device = devices.find(d => d.position === 'front');
```

### Device

OnePlus 7 Oppo Vivo

### VisionCamera Version

3.6.6

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bglgwyng, 2024-02-12:
Maybe duplicated #2153

mrousavy, 2024-02-13:
Please upgrade to 3.9.x and let me know if that fixed your issue.

Also, this is not enough logs for me to really investigate the issue, as this is only a stacktace.

mrousavy, 2024-02-14:
Hey - I just released [VisionCamera 3.9.0-beta.4](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.4) which probably fixes this issue - can you try that latest beta and let me know? :)

closed 2024-02-14 by mrousavy

hardik-nextlevel, 2024-02-19:
> Hey - I just released [VisionCamera 3.9.0-beta.4](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.4) which probably fixes this issue - can you try that latest beta and let me know? :)

Issue persists in 3.9.0 beta.4

bglgwyng, 2024-02-19:
@hardikhere  @hardik-nextlevel I made a reproducing step [here](https://github.com/mrousavy/react-native-vision-camera/issues/2153#issuecomment-1950038141)

Could you put that code in the example app and see the crash happen? It would be helpful to figure out if we are talking about the same behavior.

hardik-nextlevel, 2024-02-19:
@bglgwyng this happens in some specific devices in android not all. had to test from my side

debdeep-ganguly, 2024-05-03:
To give more context on the crash, it is coming on 98% of devices that were in the background.