margelo/react-native-vision-camera #2540: Failed to configure CameraSession! Error: Invalid pixel format 35 🐛

IvanByRool, 2024-02-12:
### What's happening?

After updating to 3.9.0-beta.3 the camera preview shows a black screen when I set pixelFormat = yuv
The error displayed below

### Reproduceable Code

```tsx
<Camera
        frameProcessor={frameProcessor}
        onInitialized={() => {
          setIsCameraReady(true);
        }}
        onError={err => {
          console.log('camera has error');
        }}
        device={device}
        pixelFormat={'yuv}
        format={format}
        video={true}
        isActive={true}
      />
```

### Relevant log output

```shell
Creating outputs for Camera #0...
2024-02-12 10:47:43.185  1969-27694 CameraSession           com.carooltruck.dev                  I  Adding 720x480 Video Output in YUV_420_888...
2024-02-12 10:47:43.185  1969-27694 VideoPipeline           com.carooltruck.dev                  I  Initializing 720 x 480 Video Pipeline (format: YUV)
2024-02-12 10:47:43.185  1969-27694 VideoPipeline           com.carooltruck.dev                  I  Using ImageReader round-trip (format: #35)
2024-02-12 10:47:43.186  1969-27694 VideoPipeline           com.carooltruck.dev                  I  Using API 29 for GPU ImageReader...
2024-02-12 10:47:43.186  1969-27694 CameraSession           com.carooltruck.dev                  E  Failed to configure CameraSession! Error: Invalid pixel format 35, isRunning: false, Config-Diff: Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
                                                                                                    java.lang.IllegalArgumentException: Invalid pixel format 35
                                                                                                    	at android.hardware.HardwareBuffer.isSupported(HardwareBuffer.java:202)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.supportsHardwareBufferFlags(VideoPipeline.kt:231)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.getRecommendedHardwareBufferFlags(VideoPipeline.kt:213)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.<init>(VideoPipeline.kt:84)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:285)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.access$configureOutputs(CameraSession.kt:44)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$configure$2$1.invokeSuspend(CameraSession.kt:135)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$configure$2$1.invoke(Unknown Source:8)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$configure$2$1.invoke(Unknown Source:2)
                                                                                                    	at com.mrousavy.camera.core.PersistentCameraCaptureSession.withConfiguration(PersistentCameraCaptureSession.kt:79)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:127)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$createPreviewOutput$1.invokeSuspend(CameraSession.kt:205)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:883)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loop(Looper.java:237)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2024-02-12 10:47:43.186  1969-27694 CameraView              com.carooltruck.dev                  E  invokeOnError(...):
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  java.lang.IllegalArgumentException: Invalid pixel format 35
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at android.hardware.HardwareBuffer.isSupported(HardwareBuffer.java:202)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.VideoPipeline.supportsHardwareBufferFlags(VideoPipeline.kt:231)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.VideoPipeline.getRecommendedHardwareBufferFlags(VideoPipeline.kt:213)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.VideoPipeline.<init>(VideoPipeline.kt:84)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:285)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession.access$configureOutputs(CameraSession.kt:44)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession$configure$2$1.invokeSuspend(CameraSession.kt:135)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession$configure$2$1.invoke(Unknown Source:8)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession$configure$2$1.invoke(Unknown Source:2)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.PersistentCameraCaptureSession.withConfiguration(PersistentCameraCaptureSession.kt:79)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:127)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at com.mrousavy.camera.core.CameraSession$createPreviewOutput$1.invokeSuspend(CameraSession.kt:205)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at android.os.Handler.handleCallback(Handler.java:883)
2024-02-12 10:47:43.186  1969-27694 System.err              com.carooltruck.dev                  W  	at android.os.Handler.dispatchMessage(Handler.java:100)
[... 30 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung SM-G960U1

### VisionCamera Version

3.9.0-beta.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

IvanByRool, 2024-02-12:
I've been testing this on several old Samsung S9 / S8 and always getting the same error. Thx

mrousavy, 2024-02-13:
Which Android version is your Samsung SM-G960U1?

IvanByRool, 2024-02-13:
Version android : 10

mrousavy, 2024-02-13:
Found a fix for this issue! https://github.com/mrousavy/react-native-vision-camera/pull/2547 :) 💪

IvanByRool, 2024-02-13:
Thanks, I appreciate, as always. 
It was my last fix to migrate from 2.x to 3.x 🔥

closed 2024-02-13 by mrousavy