margelo/react-native-vision-camera #2542: iPhone 14 does not switch between zoom levels 🐛

griffinbaker12, 2024-02-12:
### What's happening?

When switching between zoom levels with the triple-camera, nothing happens on the iPhone 14. This does not happens on the iPhone 15 Pro. Unsure what other devices it may or may not occur on. The user using two fingers to zoom in / out as opposed to the button did not work either.

[image]

### Reproduceable Code

```tsx
useEffect(() => {
        const priorZoom = zoom.value;
        if (cameraPos === "front") {
            zoom.value = withTiming(minZoom, { duration: 300 });
        } else {
            if (includeUltraWide) {
                zoom.value = withTiming(minZoom, { duration: 300 });
            } else {
                zoom.value = withTiming(neutralZoom, { duration: 300 });
            }
        }
        if (!sent && isCameraInit && phone && numberToCheck.has(phone)) {
            const devices = Camera.getAvailableCameraDevices();
            const deviceNames = devices.map((device) => device.name);
            const deviceNamesString = "Devices: " + deviceNames.join(", ");

            const zoomString = `Min: ${device?.minZoom}` + " " + `Neutral: ${device?.neutralZoom}` + " " + `Max: ${device?.maxZoom}`;

            trackError(axios.authAxios, deviceNamesString + " " + zoomString);
            sent = true;
        }
        const newZoom = zoom.value;
        if (numberToCheck.has(phone)) {
            trackError(axios.authAxios, `Zoom went from ${priorZoom} to ${newZoom}; IncludeUltraWide is ${includeUltraWide}`);
        }
    }, [cameraPos, includeUltraWide]);
```

### Relevant log output

```shell
This were the logs I received via text from a user experiencing this issue:

-- Device Information --
Devices: Back Camera, Front Camera, Front TrueDepth Camera, Back Dual Wide Camera, Back Ultra Wide Camera Min: 1 Neutral: 2 Max: 123.75

-- When the user was toggling between the 0.5x and 1x cam --
Error: Zoom went from 1 to 1; IncludeUltraWide is false
Error: Zoom went from 2 to 2; IncludeUltraWide is true
Error: Zoom went from 1 to 1; IncludeUltraWide is false
Error: Zoom went from 2 to 2; IncludeUltraWide is false
```

### Camera Device

```json
{
  "name": "Back Camera",
  "hasFlash": true,
  "position": "back",
  "hardwareLevel": "full",
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-right",
  "isMultiCam": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "formats": [],
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "hasTorch": true,
  "maxZoom": 123.75,
  "minZoom": 1
}
 LOG  {
  "sensorOrientation": "landscape-right",
  "hasTorch": false,
  "isMultiCam": false,
  "minZoom": 1,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "hardwareLevel": "full",
  "position": "front",
  "supportsLowLightBoost": false,
  "name": "Front Camera",
  "hasFlash": true,
  "maxZoom": 128.875,
  "formats": []
}
```

### Device

iPhone 14

### VisionCamera Version

3.6.8

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stevengoldberg, 2024-02-12:
Hmm, can't really see what's happening here without knowing how `includeUltraWide` gets set, but FWIW zooming works as expected on my iPhone 14 Pro. Switching between 0.5 and 1 zoom is essentially just setting `zoom.value = device.minZoom` or `zoom.value = device.neutralZoom`.

griffinbaker12, 2024-02-12:
The weird thing is that this works on iPhone 14 / 15 Pro, but not on the base 14. Include ultrawide gets set when people press the 0.5x and 1x button.

mrousavy, 2024-04-22:
Maybe a rounding issue. Or the iPhone just decides to not yet switch to the Camera (the new iOS macro lens feature). Either way, nothing I can do about it.

closed 2024-04-22 by mrousavy