margelo/react-native-vision-camera #2543: 🐛 Android 3.9.0-beta.3 - Deformed preview image

vitzaoral, 2024-02-12:
### What's happening?

On some Android (Xiaomi Mi A3) preview image is deformed.
But on some Android it works well.
It always works well on iPhone.
Same problem in example app...
[image]

### Reproduceable Code

```tsx
const device = useCameraDevice("back", {
    physicalDevices: [
      "ultra-wide-angle-camera",
      "wide-angle-camera",
      "telephoto-camera"
    ]
  })

const photoAspectRatio = 4 / 3

const selectedFormat = useCameraFormat(device, [
    { videoResolution: "max" },
    { photoAspectRatio: photoAspectRatio },
    { photoResolution: { width: 1200 } }
  ])

<CameraFrame width={width} height={width * photoAspectRatio}>
<Camera
          ref={camera}
          style={
            StyleSheet.absoluteFill
          }
          device={device}
          isActive={true}
          zoom={zoom}
          flash={supportsFlash ? flash : "off"}
          photo={true}
           format={selectedFormat}
          orientation={cameraOrientation}
        />
</>
```

### Relevant log output

```shell
2024-02-12 18:32:22.937  3552-3717  CameraSession           com.punkhive.beentry                 I  Successfully configured Session with 1 outputs for Camera #0!
2024-02-12 18:32:22.937  3552-3717  Persistent...ureSession com.punkhive.beentry                 D  --> setRepeatingRequest(...)
2024-02-12 18:32:22.937  3552-3717  Persistent...ureSession com.punkhive.beentry                 D  --> setIsActive(false)
2024-02-12 18:32:22.937  3552-3717  Persistent...ureSession com.punkhive.beentry                 D  Configure() with isActive: false, ID: 0, device: null, session: null
2024-02-12 18:32:22.937  3552-3717  Persistent...ureSession com.punkhive.beentry                 I  Creating new device...
2024-02-12 18:32:22.937  3552-3717  CameraManager           com.punkhive.beentry                 I  Camera #0: Opening...
2024-02-12 18:32:22.942  3552-3717  libc                    com.punkhive.beentry                 E  Access denied finding property "persist.vendor.camera.privapp.list"
2024-02-12 18:32:22.933  3552-3552  sionCamera.main         com.punkhive.beentry                 W  type=1400 audit(0.0:7179745): avc: denied { read } for name="u:object_r:vendor_persist_camera_prop:s0" dev="tmpfs" ino=21288 scontext=u:r:untrusted_app:s0:c236,c257,c512,c768 tcontext=u:object_r:vendor_persist_camera_prop:s0 tclass=file permissive=0 app=com.punkhive.beentry
2024-02-12 18:32:22.950  3552-4561  libc                    com.punkhive.beentry                 E  Access denied finding property "vendor.camera.aux.packagelist"
2024-02-12 18:32:22.950  3552-4561  CameraManagerGlobal     com.punkhive.beentry                 E  Camera 61 is not available. Ignore physical camera status change
2024-02-12 18:32:22.950  3552-4561  libc                    com.punkhive.beentry                 E  Access denied finding property "vendor.camera.aux.packagelist"
2024-02-12 18:32:22.952  3552-3720  CameraDevices           com.punkhive.beentry                 I  Camera #0 is now unavailable.
2024-02-12 18:32:22.933  3552-3552  Binder:3552_6           com.punkhive.beentry                 W  type=1400 audit(0.0:7179746): avc: denied { read } for name="u:object_r:vendor_persist_camera_prop:s0" dev="tmpfs" ino=21288 scontext=u:r:untrusted_app:s0:c236,c257,c512,c768 tcontext=u:object_r:vendor_persist_camera_prop:s0 tclass=file permissive=0 app=com.punkhive.beentry
2024-02-12 18:32:22.961   723-723   mm-camera               and...e.camera.provider@2.4-service  E  client_name com.punkhive.beentry all_log_disable 0
2024-02-12 18:32:22.943  3552-3552  Binder:3552_6           com.punkhive.beentry                 W  type=1400 audit(0.0:7179747): avc: denied { read } for name="u:object_r:vendor_persist_camera_prop:s0" dev="tmpfs" ino=21288 scontext=u:r:untrusted_app:s0:c236,c257,c512,c768 tcontext=u:object_r:vendor_persist_camera_prop:s0 tclass=file permissive=0 app=com.punkhive.beentry
2024-02-12 18:32:22.994  3552-3718  CameraManager           com.punkhive.beentry                 I  Camera #0: Opened!
2024-02-12 18:32:22.994  3552-3717  Persistent...ureSession com.punkhive.beentry                 I  Creating new session...
2024-02-12 18:32:22.996  3552-3717  CreateCaptureSession    com.punkhive.beentry                 I  Camera #0: Creating Capture Session #5... (Hardware Level: 3 | Outputs: [PHOTO (4000x3000 in JPEG)])
2024-02-12 18:32:22.997  3552-3717  CreateCaptureSession    com.punkhive.beentry                 I  Using new API (>=28)
2024-02-12 18:32:23.001  3552-3552  CameraView              com.punkhive.beentry                 I  Updating CameraSession...
2024-02-12 18:32:23.003  3552-3552  PreviewView             com.punkhive.beentry                 I  PreviewView is 720x960, rendering 720x1411 content. Resizing to: 1881x960 (COVER)
2024-02-12 18:32:23.015  3552-3717  CameraSession           com.punkhive.beentry                 I  configure { ... }: Waiting for lock...
2024-02-12 18:32:23.016  3552-3718  CreateCaptureSession    com.punkhive.beentry                 I  Camera #0: Successfully created CameraCaptureSession #5!
[... 73 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Xiaomi Mi A3

### VisionCamera Version

3.9.0-beta.3

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

nica0012, 2024-02-12:
I am having the same issue, on IOS everything works fine.

On android, my preview is slightly zoomed in vs the output which looks normal.

nonam4, 2024-02-12:
This patch fixed issue for me:

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
index fef6998..c339ea7 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt
@@ -313,7 +313,7 @@ class CameraSession(private val context: Context, private val cameraManager: Cam
       )
       outputs.add(output)
       // Size is usually landscape, so we flip it here
-      previewView?.setSurfaceSize(size.width, size.height)
+      previewView?.setSurfaceSize(size.height, size.width)
     }
 
     // CodeScanner Output
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
index c862be7..2d26fe9 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
@@ -43,13 +43,13 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) : SurfaceV
     holder.addCallback(callback)
   }
 
-  suspend fun setSurfaceSize(width: Int, height: Int) {
+  suspend fun setSurfaceSize(height: Int, width: Int) {
     withContext(Dispatchers.Main) {
-      size = Size(width, height)
+      size = Size(height, width)
       Log.i(TAG, "Setting PreviewView Surface Size to $size...")
       requestLayout()
       invalidate()
-      holder.resize(width, height)
+      holder.resize(height, width)
     } 
   }
 
@@ -73,7 +73,7 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) : SurfaceV
 
   private fun getSize(contentSize: Size, containerSize: Size, resizeMode: ResizeMode): Size {
     // TODO: Take sensor orientation into account here
-    val contentAspectRatio = contentSize.height.toDouble() / contentSize.width
+    val contentAspectRatio = contentSize.width.toDouble() / contentSize.height
     val containerAspectRatio = containerSize.width.toDouble() / containerSize.height
 
     val widthOverHeight = when (resizeMode) {
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt
index ecd525c..e758c45 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getPreviewSize.kt
@@ -9,7 +9,7 @@ fun getMaximumPreviewSize(): Size {
   // See https://developer.android.com/reference/android/hardware/camera2/params/StreamConfigurationMap
   // According to the Android Developer documentation, PREVIEW streams can have a resolution
   // of up to the phone's display's resolution, with a maximum of 1920x1080.
-  val display1080p = Size(1920, 1080)
+  val display1080p = Size(1080, 1920)
   val displaySize = Size(
     Resources.getSystem().displayMetrics.widthPixels,
     Resources.getSystem().displayMetrics.heightPixels
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt
index cc0fe9c..6ef3880 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/SurfaceHolder+resize.kt
@@ -7,7 +7,7 @@ import kotlin.coroutines.resume
 import kotlinx.coroutines.suspendCancellableCoroutine
 
 @UiThread
-suspend fun SurfaceHolder.resize(width: Int, height: Int) {
+suspend fun SurfaceHolder.resize(height: Int, width: Int) {
   return suspendCancellableCoroutine { continuation ->
     val currentSize = this.surfaceFrame
     if (currentSize.width() == width && currentSize.height() == height) {
```

I suppose this is happeing because width/height is inverted

frenbergFNX, 2024-02-13:
Same issue on samsung s10

```
2024-02-13 13:29:53.315 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  ViewPostIme pointer 0
2024-02-13 13:29:53.428 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  ViewPostIme pointer 1
2024-02-13 13:29:53.707 18071-18071 SensorManager           se.fortnox.powerup.acce              D  unregisterListener ::   
2024-02-13 13:29:53.735 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  MSG_WINDOW_FOCUS_CHANGED 0 1
2024-02-13 13:29:53.735 18071-18071 InputMethodManager      se.fortnox.powerup.acce              D  prepareNavigationBarInfo() DecorView@d513ce7[MainActivity]
2024-02-13 13:29:53.735 18071-18071 InputMethodManager      se.fortnox.powerup.acce              D  getNavigationBarColor() -855310
2024-02-13 13:29:53.751 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  stopped(false) old=false
2024-02-13 13:29:53.756 18071-18071 SensorManager           se.fortnox.powerup.acce              D  registerListener :: 1, LSM6DSO Acceleration Sensor, 66667, 0,  
2024-02-13 13:29:53.772 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=6 res=0x1 s={true 492643340288} ch=false
2024-02-13 13:29:53.778 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  MSG_WINDOW_FOCUS_CHANGED 1 1
2024-02-13 13:29:53.778 18071-18071 InputMethodManager      se.fortnox.powerup.acce              D  prepareNavigationBarInfo() DecorView@d513ce7[MainActivity]
2024-02-13 13:29:53.778 18071-18071 InputMethodManager      se.fortnox.powerup.acce              D  getNavigationBarColor() -855310
2024-02-13 13:29:53.785 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=2 res=0x1 s={true 492643340288} ch=false
2024-02-13 13:29:53.883 18071-25901 DDN                     se.fortnox.powerup.acce              D  license valid
2024-02-13 13:29:54.350 18071-18071 PreviewView             se.fortnox.powerup.acce              I  Creating PreviewView...
2024-02-13 13:29:54.350 18071-18071 PreviewView             se.fortnox.powerup.acce              I  PreviewView is 0x0, rendering 720x1415 content. Resizing to: 0x0 (COVER)
2024-02-13 13:29:54.350 18071-18071 CameraView              se.fortnox.powerup.acce              I  Updating CameraSession...
2024-02-13 13:29:54.351 18071-19340 CameraSession           se.fortnox.powerup.acce              I  configure { ... }: Waiting for lock...
2024-02-13 13:29:54.351 18071-18071 CameraView              se.fortnox.powerup.acce              I  Updating CameraSession...
2024-02-13 13:29:54.351 18071-18071 SurfaceView             se.fortnox.powerup.acce              I  onWindowVisibilityChanged(0) true com.mrousavy.camera.core.PreviewView{bca0861 V.E...... ......I. 0,0-0,0} of ViewRootImpl@3143ee1[MainActivity]
2024-02-13 13:29:54.351 18071-19340 CameraSession           se.fortnox.powerup.acce              I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-02-13 13:29:54.351 18071-19340 CameraSession           se.fortnox.powerup.acce              I  Configuring inputs for CameraSession...
2024-02-13 13:29:54.352 18071-18071 PreviewView             se.fortnox.powerup.acce              I  PreviewView is 720x1351, rendering 720x1415 content. Resizing to: 2655x1351 (COVER)
2024-02-13 13:29:54.352 18071-19340 Persistent...ureSession se.fortnox.powerup.acce              D  --> setInput(0)
2024-02-13 13:29:54.352 18071-19340 CameraSession           se.fortnox.powerup.acce              I  Destroying previous outputs...
2024-02-13 13:29:54.353 18071-19340 CameraSession           se.fortnox.powerup.acce              I  Creating outputs for Camera #0...
2024-02-13 13:29:54.354 18071-19340 CameraSession           se.fortnox.powerup.acce              I  Adding 4032x1908 Photo Output in JPEG...
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.fortnox.powerup.acce              D  --> setOutputs([PHOTO (4032x1908 in JPEG)])
2024-02-13 13:29:54.354 18071-19340 CameraSession           se.fortnox.powerup.acce              I  Successfully configured Session with 1 outputs for Camera #0!
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.fortnox.powerup.acce              D  --> setRepeatingRequest(...)
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.fortnox.powerup.acce              D  --> setIsActive(false)
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.fortnox.powerup.acce              D  Configure() with isActive: false, ID: 0, device: null, session: null
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.fortnox.powerup.acce              I  Creating new device...
2024-02-13 13:29:54.354 18071-19340 CameraManager           se.fortnox.powerup.acce              I  Camera #0: Opening...
2024-02-13 13:29:54.356 18071-18071 ViewRootIm...nActivity] se.fortnox.powerup.acce              I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=4 res=0x1 s={true 492643340288} ch=false
[... cut ...]
2024-02-13 13:29:54.481 18071-19340 CameraSession           se.fortnox.powerup.acce              I  configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: true)
2024-02-13 13:29:54.481 18071-19340 CameraView              se.fortnox.powerup.acce              I  invokeOnStarted()
2024-02-13 13:29:54.759 18071-19248 CameraManagerGlobal     se.fortnox.powerup.acce              I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client se.fortnox.powerup.acce API Level 2
```

frenbergFNX, 2024-02-13:
Same issue on Samsung S10

```
2024-02-13 13:29:53.315 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  ViewPostIme pointer 0
2024-02-13 13:29:53.428 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  ViewPostIme pointer 1
2024-02-13 13:29:53.707 18071-18071 SensorManager           se.powerup.acce              D  unregisterListener ::   
2024-02-13 13:29:53.735 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  MSG_WINDOW_FOCUS_CHANGED 0 1
2024-02-13 13:29:53.735 18071-18071 InputMethodManager      se.powerup.acce              D  prepareNavigationBarInfo() DecorView@d513ce7[MainActivity]
2024-02-13 13:29:53.735 18071-18071 InputMethodManager      se.powerup.acce              D  getNavigationBarColor() -855310
2024-02-13 13:29:53.751 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  stopped(false) old=false
2024-02-13 13:29:53.756 18071-18071 SensorManager           se.powerup.acce              D  registerListener :: 1, LSM6DSO Acceleration Sensor, 66667, 0,  
2024-02-13 13:29:53.772 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=6 res=0x1 s={true 492643340288} ch=false
2024-02-13 13:29:53.778 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  MSG_WINDOW_FOCUS_CHANGED 1 1
2024-02-13 13:29:53.778 18071-18071 InputMethodManager      se.powerup.acce              D  prepareNavigationBarInfo() DecorView@d513ce7[MainActivity]
2024-02-13 13:29:53.778 18071-18071 InputMethodManager      se.powerup.acce              D  getNavigationBarColor() -855310
2024-02-13 13:29:53.785 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=2 res=0x1 s={true 492643340288} ch=false
2024-02-13 13:29:53.883 18071-25901 DDN                     se.powerup.acce              D  license valid
2024-02-13 13:29:54.350 18071-18071 PreviewView             se.powerup.acce              I  Creating PreviewView...
2024-02-13 13:29:54.350 18071-18071 PreviewView             se.powerup.acce              I  PreviewView is 0x0, rendering 720x1415 content. Resizing to: 0x0 (COVER)
2024-02-13 13:29:54.350 18071-18071 CameraView              se.powerup.acce              I  Updating CameraSession...
2024-02-13 13:29:54.351 18071-19340 CameraSession           se.powerup.acce              I  configure { ... }: Waiting for lock...
2024-02-13 13:29:54.351 18071-18071 CameraView              se.powerup.acce              I  Updating CameraSession...
2024-02-13 13:29:54.351 18071-18071 SurfaceView             se.powerup.acce              I  onWindowVisibilityChanged(0) true com.mrousavy.camera.core.PreviewView{bca0861 V.E...... ......I. 0,0-0,0} of ViewRootImpl@3143ee1[MainActivity]
2024-02-13 13:29:54.351 18071-19340 CameraSession           se.powerup.acce              I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-02-13 13:29:54.351 18071-19340 CameraSession           se.powerup.acce              I  Configuring inputs for CameraSession...
2024-02-13 13:29:54.352 18071-18071 PreviewView             se.powerup.acce              I  PreviewView is 720x1351, rendering 720x1415 content. Resizing to: 2655x1351 (COVER)
2024-02-13 13:29:54.352 18071-19340 Persistent...ureSession se.powerup.acce              D  --> setInput(0)
2024-02-13 13:29:54.352 18071-19340 CameraSession           se.powerup.acce              I  Destroying previous outputs...
2024-02-13 13:29:54.353 18071-19340 CameraSession           se.powerup.acce              I  Creating outputs for Camera #0...
2024-02-13 13:29:54.354 18071-19340 CameraSession           se.powerup.acce              I  Adding 4032x1908 Photo Output in JPEG...
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.powerup.acce              D  --> setOutputs([PHOTO (4032x1908 in JPEG)])
2024-02-13 13:29:54.354 18071-19340 CameraSession           se.powerup.acce              I  Successfully configured Session with 1 outputs for Camera #0!
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.powerup.acce              D  --> setRepeatingRequest(...)
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.powerup.acce              D  --> setIsActive(false)
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.powerup.acce              D  Configure() with isActive: false, ID: 0, device: null, session: null
2024-02-13 13:29:54.354 18071-19340 Persistent...ureSession se.powerup.acce              I  Creating new device...
2024-02-13 13:29:54.354 18071-19340 CameraManager           se.powerup.acce              I  Camera #0: Opening...
2024-02-13 13:29:54.356 18071-18071 ViewRootIm...nActivity] se.powerup.acce              I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=4 res=0x1 s={true 492643340288} ch=false
2024-02-13 13:29:54.360 18071-18071 SurfaceView             se.powerup.acce              I  surfaceCreated 1 #8 com.mrousavy.camera.core.PreviewView{bca0861 V.E...... ......ID -967,0-1688,1351}
2024-02-13 13:29:54.361 18071-18071 CameraSession           se.powerup.acce              I  PreviewView Surface created! Surface(name=null)/@0xf994f86
2024-02-13 13:29:54.361 18071-18071 CameraSession           se.powerup.acce              I  Setting Preview Output...
[... cut ...] d4c95e
2024-02-13 13:29:54.481 18071-19340 CameraSession           se.powerup.acce              I  configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: true)
2024-02-13 13:29:54.481 18071-19340 CameraView              se.powerup.acce              I  invokeOnStarted()
2024-02-13 13:29:54.759 18071-19248 CameraManagerGlobal     se.powerup.acce              I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client se.powerup.acce API Level 2
```

mrousavy, 2024-02-14:
Does @nonam4's patch work for everyone of you? Then it is definitely just an orientation issue.

vitzaoral, 2024-02-14:
I tried @nonam4 path on _Xiaomi Mi A3_ and _Motorola moto g(60)s_. With patch it works well, image isn't deformed. But is zoomed when compared with default device camera app.
[image]
[image]

nonam4, 2024-02-14:
> Can you please all post screenshots of how it looks?
> 
> Does it change the view after a fast refresh, or does it always stay the same?

I don't have any screenshot right now but before applying patch the preview image was in landscape mode (image width was screen height and image heigth was screen width).

Change camera's `resizeMode` to `contain` without applying patch and you'll see a landscape image without distortion but not filling screen height.

mrousavy, 2024-02-15:
I think I fixed this now. I don't have a device with a different sensor orientation than `landscapeLeft` here to test it, so could you guys please test if [v3.9.0-beta.5](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.5) fixed the issue for you guys? :)

vitzaoral, 2024-02-15:
I tried it in the pulled Example app ont the Xiaomi Mi A3 device. It is much more better, but still preview image is little bit stretched (horizontally) and zoomed in.
[image]
[image]

mrousavy, 2024-02-15:
@vitzaoral hmm, could you try to tweak the values in `PreviewView.kt`?

https://github.com/mrousavy/react-native-vision-camera/blob/c5646ca1e278b6c2afc6c259cb2e98980cc4f96d/package/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt#L105-L113

Maybe replace `width / ...` with `width * ...`, or `height / ...` or `height * ...`.

vitzaoral, 2024-02-16:
@mrousavy I tried to swap operators, values etc. (OK, without deeper knowledge of how all thing works) but unfortunately without result.. :( Still bad behavior (version Android 3.9.0-beta.6) on _Xiaomi Mi A3_, works well on _Motorola moto g(60)s_ 🤔

mrousavy, 2024-02-16:
Wtf, this is so weird 😂

vitzaoral, 2024-02-16:
Yep...unfortunately I only have these two Android devices to test.

mrousavy, 2024-02-16:
Can you share your adb logcat logs of the Xiaomi with me? There should be something along the lines of
```
PreviewView is ...x..., rendering ...x.. ....
```

in there which I need.

vitzaoral, 2024-02-16:
Of course. Here is the logcat from the Example app on the Xiaomi device. This lines are repeating forever, but I don't see _PreviewView_ or _rendering_ text. Am I doing something wrong?
[image]

mlecoq, 2024-02-17:
I have 

```
2024-02-17 23:25:33.469 29592-29592 CameraSession           com.myapp                I  PreviewView Surface updated! Surface(name=null)/@0x4271077 1280 x 960
2024-02-17 23:25:33.475 29592-25308 Persistent...ureSession com.myapp                 D  --> setOutputs([PHOTO (2976x2976 in JPEG), PREVIEW (1280 x 960)])
2024-02-17 23:25:33.481 29592-25308 CreateCaptureSession    com.myapp                 I  Camera #0: Creating Capture Session #25... (Hardware Level: 3 | Outputs: [PHOTO (2976x2976 in JPEG), PREVIEW (1280 x 960)])
2024-02-17 23:25:33.522   823-823   CHI                     ven...e.camera.provider@4.0-service  I  [SS_INFO]: [CHI         ]: chxusecaseutils.cpp: DefaultMatchingUsecase: 863: usecase UsecaseJPEGEncodePreviewSnapshot, pipelineName JPEGEncodePreview
2024-02-17 23:25:33.522   823-823   CHI                     ven...e.camera.provider@4.0-service  I  [SS_INFO]: [CHI         ]: chxusecase.cpp: Initialize: 2161: [Default] usecase name = UsecaseJPEGEncodePreviewSnapshot, pipeline name = JPEGEncodePreview
2024-02-17 23:25:33.748   823-823   CHI                     ven...e.camera.provider@4.0-service  I  [SS_INFO]: [CHI         ]: chxutils.cpp: FillTuningModeDataLog: 2103: [REAR_WIDE] CameraID 0 SensorSetting 0  Sensor:0  usecase:THIRD_PARTY_PREVIEW(3)  feature1:NONE(0)  feature2:HDR_OFF(0)  scene:NONE(0)
2024-02-17 23:26:12.815 29592-29592 PreviewView            com.myapp                 I  PreviewView is 1600x1539, rendering 960x1280 content (LANDSCAPE_LEFT). Resizing to: 1600x2133 (COVER)
```

mlecoq, 2024-02-17:
I have the following logs

```
2024-02-17 23:25:33.475 29592-25308 Persistent...ureSession com.myapp                D  --> setOutputs([PHOTO (2976x2976 in JPEG), PREVIEW (1280 x 960)])
2024-02-17 23:25:33.481 29592-25308 CreateCaptureSession    com.myapp                I  Camera #0: Creating Capture Session #25... (Hardware Level: 3 | Outputs: [PHOTO (2976x2976 in JPEG), PREVIEW (1280 x 960)])
2024-02-17 23:25:33.522   823-823   CHI                     ven...e.camera.provider@4.0-service  I  [SS_INFO]: [CHI         ]: chxusecaseutils.cpp: DefaultMatchingUsecase: 863: usecase UsecaseJPEGEncodePreviewSnapshot, pipelineName JPEGEncodePreview
2024-02-17 23:25:33.522   823-823   CHI                     ven...e.camera.provider@4.0-service  I  [SS_INFO]: [CHI         ]: chxusecase.cpp: Initialize: 2161: [Default] usecase name = UsecaseJPEGEncodePreviewSnapshot, pipeline name = JPEGEncodePreview
2024-02-17 23:25:33.748   823-823   CHI                     ven...e.camera.provider@4.0-service  I  [SS_INFO]: [CHI         ]: chxutils.cpp: FillTuningModeDataLog: 2103: [REAR_WIDE] CameraID 0 SensorSetting 0  Sensor:0  usecase:THIRD_PARTY_PREVIEW(3)  feature1:NONE(0)  feature2:HDR_OFF(0)  scene:NONE(0)
2024-02-17 23:26:12.815 29592-29592 PreviewView             com.myapp                I  PreviewView is 1600x1539, rendering 960x1280 content (LANDSCAPE_LEFT). Resizing to: 1600x2133 (COVER)
```

mlecoq, 2024-02-17:
The preview is more similar to taken photo with the following code, we'll make a pr

```
 return if (widthOverHeight) {
      // Scale by width to cover height
      val scaledWidth =  containerSize.height / contentSize.height  * contentSize.width
      Size(scaledWidth, containerSize.height)
    } else {
      // Scale by height to cover width
      val scaledHeight = containerSize.width / contentSize.width * contentSize.height
      Size(containerSize.width, scaledHeight)
    }
```

mlecoq, 2024-02-18:
to get the same result in preview and taken picture, I have just to modify `getSize` method in `PreviewView.kt` to return `containerSize`

```
  private fun getSize(contentSize: Size, containerSize: Size, resizeMode: ResizeMode): Size {
   return containerSize
  }

```

and I give the expected aspect ratio to my camera view

```

 const [layoutRect, setLayoutRect] = useState<LayoutRectangle | null>(null);

 const aspectRatio = layoutRect
    ? layoutRect.width / layoutRect.height
    : undefined;

  const format = useCameraFormat(device, [
    {
      videoResolution: { width: 1280, height: 720 },
      photoAspectRatio: aspectRatio,
      videoAspectRatio: aspectRatio,
    },
  ]);

  return <View
      {...props}
      onLayout={event => setLayoutRect(event.nativeEvent.layout)}
    >
     <Camera
            ref={camera}
            style={styles.camera}
            device={device}
            fps={30}
            format={format}/>
   </View>

```

it works on my redmi note and on my Galaxy Tab S5e