margelo/react-native-vision-camera #2544: 🐛 "no non-static method" error only on release mode

nonam4, 2024-02-13:
### What's happening?

I'm using expo SDK 50 managed workflow and while on dev mode (npx expo run:android -d) everything works really fine but after doing a release build my app crash when trying to open the camera view.

I'm using my own [face detector plugin](https://www.npmjs.com/package/react-native-vision-camera-face-detector).

I think this is something related to #2168

Full crashlytics error: 
```
Error: Exception in HostFunction: java.lang.NoSuchMethodError: no non-static method "Lcom/mrousavy/camera/frameprocessor/FrameProcessor;.<init>(Lcom/facebook/jni/HybridData;)V", js engine: hermes, stack:
setFrameProcessor@1:1674615
onViewReady@1:1674705
invokeGuardedCallbackImpl@1:371841
invokeGuardedCallback@1:371898
invokeGuardedCallbackAndCatchFirstError@1:371930
executeDispatch@1:372059
executeDispatchesAndReleaseTopLevel@1:376253
forEachAccumulated@1:373535
anonymous@1:376616
batchedUpdatesImpl@1:434780
batchedUpdates@1:376169
_receiveRootNodeIDEvent@1:376454
receiveEvent@1:429381
__callFunction@1:133914
anonymous@1:132354
__guard@1:133295
callFunctionReturnFlushedQueue@1:132312
       at com.facebook.react.modules.core.ExceptionsManagerModule.reportException(ExceptionsManagerModule.java:75)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:142)
       at com.facebook.react.bridge.JavaModuleWrapper.invoke(:21)
       at com.facebook.jni.NativeRunnable.run(SourceFile)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage()
       at android.os.Looper.loop(Looper.java:164)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(:37)
       at java.lang.Thread.run(Thread.java:764)
```

### Reproduceable Code

```tsx
import { 
  StyleSheet, 
  Text, 
  View 
} from 'react-native'
import { 
  useEffect, 
  useState 
} from 'react'
import {
  Camera,
  useCameraDevice,
  useFrameProcessor
} from 'react-native-vision-camera'
import { 
  detectFaces,
  DetectionResult 
} from 'react-native-vision-camera-face-detector'
import { Worklets } from 'react-native-worklets-core'

export default function App() {
  const device = useCameraDevice('front')

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission()
[... 170 lines cut ...]
  "hasFlash": false,
  "maxExposure": 20,
  "name": "FRONT (1)"
}
```

### Device

Iphone 11 with Ios 17.2.1 and Galaxy J5 with android 8

### VisionCamera Version

both 3.8.2 and 3.9.0-beta.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-13:
Are you using ProGuard?

mrousavy, 2024-02-13:
Thanks for the issue report, I found a fix: https://github.com/mrousavy/react-native-vision-camera/pull/2546 :) 💪

mrousavy, 2024-02-13:
[image]

This is not true. Please only post correct information

nonam4, 2024-02-13:
@mrousavy Thanks for quick response, I was writing a reply about ProGuard and you already found the fix XD.

> This is not true. Please only post correct information

Sorry, my bad.

closed 2024-02-13 by mrousavy

mrousavy, 2024-02-13:
[image]

nonam4, 2024-02-13:
I also noticed that this error happens only when opening app at first time, if I reopen it there's no crash but frame processors doesn't work.

mrousavy, 2024-02-13:
Unfortunately there's nothing I can do without logs and potentially a symbolicated stack trace. Please try to get those from the crash, then I can help you

nonam4, 2024-02-13:
First, I'm using proguard on my expo app (enabled it on [BuildProperties](https://docs.expo.dev/versions/latest/sdk/build-properties/#platform-compatibility))

I got some runtime logs, here they are:
```
31876-31951 unknown:ViewManagerPropertyUpdater  Could not find generated setter for class com.mrousavy.camera.CameraViewManager
31876-31951 CameraDevices                       Camera #0 is now available.
31876-31951 CameraDevices                       Camera #1 is now available.
31876-31948 ReactNativeJS                       Loading react-native-worklets-core...
31876-31948 ReactNativeJS                       Worklets loaded successfully
31876-31948 VisionCameraProxy                   Initializing VisionCameraProxy...
31876-31948 VisionCameraProxy                   Creating Worklet Context...
31876-31948 VisionCameraProxy                   Worklet Context created!
31876-31948 CameraManager                       getCameraCharacteristics : cameraId = 0
31876-31948 CameraManager                       getCameraCharacteristics : cameraId = 0
31876-31948 CameraManager                       getCameraCharacteristics : cameraId = 1
31876-31948 CameraManager                       getCameraCharacteristics : cameraId = 1
31876-31948 FrameProcessorPluginRegistry        Looking up Frame Processor Plugin "detectFaces"...
31876-31948 FrameProcessorPluginRegistry        Frame Processor Plugin "detectFaces" found! Initializing...
31876-31876 PreviewView                         Creating PreviewView...
31876-31876 PreviewView                         PreviewView is 0x0, rendering 720x1280 content. Resizing to: 0x0 (COVER)
31876-31876 CameraView                          Updating CameraSession...
31876-31940 CameraSession                       configure { ... }: Waiting for lock...
31876-31940 CameraSession                       configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
31876-31940 CameraSession                       Configuring inputs for CameraSession...
31876-31940 PersistentCameraCaptureSession      --> setInput(1)
31876-31951 zygote                              Deoptimizing void com.facebook.react.uimanager.n0.<init>() due to JIT inline cache
31876-31940 CameraSession                       Destroying previous outputs...
31876-31940 CameraManager                       getCameraCharacteristics : cameraId = 1
31876-31951 zygote                              Deoptimizing void com.facebook.react.uimanager.d1.t(com.facebook.react.uimanager.m0, int, com.facebook.react.uimanager.o0) due to JIT inline cache
31876-31940 CameraSession                       Creating outputs for Camera #1...
31876-31940 CameraSession                       Adding 2560x1440 Photo Output in JPEG...
31876-31940 Surface                             sf_framedrop debug : 0x4f4c, game : false, logging : 0
31876-31940 CameraSession                       Adding 1920x1080 Video Output in YUV_420_888...
31876-31940 VideoPipeline                       Initializing 1920 x 1080 Video Pipeline (format: YUV)
31876-31940 Surface                             sf_framedrop debug : 0x4f4c, game : false, logging : 0
31876-31940 VideoPipeline                       Using ImageReader round-trip (format: #35)
31876-31940 VideoPipeline                       Using legacy API for CPU ImageReader...
31876-31940 Surface                             sf_framedrop debug : 0x4f4c, game : false, logging : 0
31876-31951 zygote                              Deoptimizing void com.facebook.react.uimanager.d1.V(int, com.facebook.react.bridge.ReadableArray) due to JIT inline cache
31876-31951 zygote                              Deoptimizing void com.facebook.react.uimanager.n0.d(com.facebook.react.uimanager.n0, int) due to JIT inline cache
31876-31940 Surface                             sf_framedrop debug : 0x4f4c, game : false, logging : 0
31876-31940 PersistentCameraCaptureSession      --> setOutputs([PHOTO (2560x1440 in JPEG), VIDEO (1920x1080 in YUV)])
31876-31940 CameraSession                       Successfully configured Session with 2 outputs for Camera #1!
31876-31940 CameraSession                       Updating Video Outputs...
31876-31940 VideoPipeline                       Removing RecordingSession Output...
31876-31940 PersistentCameraCaptureSession      --> setRepeatingRequest(...)
31876-31940 PersistentCameraCaptureSession      --> setIsActive(false)
31876-31940 PersistentCameraCaptureSession      Configure() with isActive: false, ID: 1, device: null, session: null
31876-31940 PersistentCameraCaptureSession      Creating new device...
31876-31940 CameraManager                       Camera #1: Opening...
31876-31940 CameraManager                       getCameraCharacteristics : cameraId = 1
31876-31876 zygote                              Deoptimizing android.view.View com.facebook.react.uimanager.ViewManager.createViewInstance(int, com.facebook.react.uimanager.y0, com.facebook.react.uimanager.o0, com.facebook.react.uimanager.x0) due to JIT same target
31876-31951 CameraDevices                       Camera #1 is now unavailable.
31876-31951 CameraManager                       getCameraCharacteristics : cameraId = 1
31876-31945 CameraManager                       Camera #1: Opened!
31876-31940 PersistentCameraCaptureSession      Creating new session...
31876-31940 CameraManager                       getCameraCharacteristics : cameraId = 1
[... cut ...]
31876-31945 libc                                Fatal signal 11 (SIGSEGV), code 2, fault addr 0x74656d14 in tid 31945 (ionCamera.video)
```

On first app load (or if you clean app data and reopen it) there's a crash when trying to open camera view but after the crash if I reopen the app there's no crash but frame processors are not beeing called because none of my logs are printed, only `32675-32737 VideoPipeline                       ImageReader::onImageAvailable!`.

Can you say something with these logs? Is there something I can try to fix this?

nonam4, 2024-02-14:
@mrousavy I did a lot of testing in my app and I discovered that if I replace `runAsync` with `runAtTargetFps` in my frame processor the app doesn't crash but there's a side effect that camera preview freezes when my face detector plugin is processing.

I also tryed increasing and decreasing target fps value but when frame is beeing processed camera preview freezes.

> What does your Frame Processor look like?

[Here](https://www.npmjs.com/package/react-native-vision-camera-face-detector) is my plugin repo.

Crashing:
```
const handleDetectionWorklet = Worklets.createRunInJsFn( (
    result: DetectionResult
  ) => { 
    console.log( 'detection result', result )
  })
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    runAsync(frame, () => {
      'worklet'
      detectFaces(
        frame,
        handleDetectionWorklet, {
          // detection settings
        }
      )
    })
  }, [handleDetectionWorklet])
``` 

Not crashing but freezing:
```
const handleDetectionWorklet = Worklets.createRunInJsFn( (
    result: DetectionResult
  ) => { 
    console.log( 'detection result', result )
  })
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    runAtTargetFps(2, () => {
      'worklet'
      detectFaces(
        frame,
        handleDetectionWorklet, {
          // detection settings
        }
      )
    })
  }, [handleDetectionWorklet])
``` 

> Does it work if you use an empty Frame Processor? Or none at all?

If you mean this code, yes it works: 
```
const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    console.log('I'm running')
  }, [])
```

But if I add a `runAsync` (with `processNestedWorklets` in babel config) the app crash. 

Without `processNestedWorklets` I got `Regular javascript function cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet.` error related [here](https://github.com/mrousavy/react-native-vision-camera/issues/1791#issuecomment-1768616161).

If I use `runAtTargetFps` it works doesn't matter if `processNestedWorklets` is present or not on babel config but with preview frozing issue.

Any suggestion?

mrousavy, 2024-02-15:
1. `runAsync` and `runAtTargetFps` are two completely different things - `runAsync` schedules the function to run on a background Thread, with a maximum of one function at a time being able to run. `runAtTargetFps` runs on the same thread as the Frame Processor, but skips all Frames that come in more often than the target FPS rate. 
So by design, `runAtTargetFps` still freezes the Camera if your code is slow. This is how it works, by design.
2. Can you use `runAsync` in the Example app here? For me that works.

nonam4, 2024-02-16:
@mrousavy sorry for the delay in reply, I've been running some tests and I think I found the problem in my app, I suppose it's something related to `react-native-reanimated` and/or `react-native-svg`... 

I'm not sure yet what is really making it crash as I'm still doing some tests. I'll give you a feedback asap.

nonam4, 2024-02-16:
Ok, after a lot of tests I got a conclusion:

I have coppied whole example `CameraPage.tsx` code and remove everything I don't need, I left only most basic things. [Here](https://gist.github.com/nonam4/8b15f046978321d0e7de9233e62f6f66) is my current code.

The problem is: If I try to use `react-native-reanimated` there's no crash but `runAsync` method is NOT called. Also here's my whole `package.json`
```
{
  "name": "myApp",
  "version": "1",
  "private": true,
  "scripts": {
    "lint": "yarn test && eslint --quiet --fix --ext .js,.ts,.tsx,.jsx .",
    "test": "tsc",
    "prebuild": "npx expo prebuild",
    "prebuild:clean": "npx expo prebuild --clean",
    "android": "yarn prebuild && npx expo run:android -d",
    "android:prod": "yarn prebuild &&  npx expo run:android -d --variant release",
    "ios": "yarn prebuild && npx expo run:ios -d",
    "ios:prod": "yarn prebuild && npx expo run:ios -d --configuration Release",
    "android:clean": "yarn prebuild:clean && npx expo run:android -d --no-build-cache",
    "ios:clean": "yarn prebuild:clean && npx expo run:ios -d --no-build-cache",
    "start": "expo start --dev-client"
  },
  "main": "index.js",
  "dependencies": {
    "@react-native-community/hooks": "^3.0.0",
    "@react-navigation/native": "^6.1.9",
    "expo": "~50.0.6",
    "expo-application": "~5.8.3",
    "expo-build-properties": "~0.11.1",
    "expo-constants": "~15.4.5",
    "expo-dev-client": "~3.3.8",
    "react": "18.2.0",
    "react-native": "0.73.4",
    "react-native-gesture-handler": "~2.14.1",
    "react-native-hole-view": "^3.0.0-alpha4",
    "react-native-reanimated": "~3.7.0",
    "react-native-safe-area-context": "4.8.2",
    "react-native-svg": "^14.1.0",
    "react-native-vector-icons": "^10.0.3",
    "react-native-vision-camera": "3.9.0-beta.5",
    "react-native-vision-camera-face-detector": "^1.2.2",
    "react-native-worklets-core": "^0.3.0"
  },
  "devDependencies": {
    "@babel/core": "^7.23.9",
    "@types/react": "~18.2.48",
    "@typescript-eslint/eslint-plugin": "^6.19.1",
    "@typescript-eslint/parser": "^6.19.1",
    "babel-plugin-module-resolver": "^5.0.0",
    "eslint": "^8.56.0",
    "eslint-config-google": "^0.14.0",
    "eslint-config-next": "^14.1.0",
    "eslint-plugin-import": "^2.29.1",
    "eslint-plugin-react": "^7.33.2",
    "typescript": "^5.3.3"
  },
  "resolutions": {
    "@expo/config-plugins": "^7.8.0"
  }
}
``` 
No matter what I do in the code, If I run my app using `yarn android:prod` the app don't crash but frame processor is not called.

About the crashes.... the problem is `react-native-svg`. If I try to use any Svg with vision-camera the app crash. I tryed:
- create a aniamted svg:
```
import Svg from 'react-native-svg'
const AnimatedSvg = Animated.createAnimatedComponent( Svg )
```
- returning only a normal svg:
```
return (
  <Camera
    ....
  />
  <Svg>
    ...
  </Svg>
)
```
My SVG is rendered in front of camera preview because I'm creating a hole view (indication where user face must be).

@mrousavy do you have any thoughts on this?