margelo/react-native-vision-camera #2551: 💭 Difficulty with Barcode Detection in Specific Screen Area using React Native CLI

sanjayradadiya, 2024-02-13:
### Question

I am currently working on a project using React Native CLI and have encountered a challenge with barcode detection within a specific area of the screen. My goal is to align the scan area with the barcode frame accurately, but despite my efforts, I'm facing inconsistencies across different devices

React Native window resolution is set to 384 x 818.
The frame resolution from the camera is 1920 x 1080.

### What I tried

Approaches Attempted:
I have attempted to adjust the React Native coordinates to match the frame resolution by scaling them up based on the ratio between the React Native window and the frame resolution.
Conversely, I tried scaling down the frame coordinates to match the React Native window, also based on their ratio.
I implemented formatting using the useFormat hook to possibly address the issue.
Extensive calculations were performed to align the barcode frame with the scan area, but the results were not as expected and varied across devices.
Additionally, I experimented with rendering the camera within a fixed-width and height React Native view (260 x 260), yet the frame resolution remains at 1920 x 1080, and the scale ratio also varied when using the full screen.

Despite these efforts, the calculated values derived from the barcode frame do not seem to align with the scan area defined by one of the view elements.

Request for Assistance:
I would greatly appreciate any guidance or suggestions on how to accurately align the barcode scanning area with the frame across various devices.

Thank you for your time and assistance.

VisionCamera Version
3.8.2
Worklet Core Version
0.3.0
Reanimated Version
3.5.4

### VisionCamera Version

3.8.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-04-22:
You need to map from native coordinates to view coordinates yourself.

closed 2024-04-22 by mrousavy

pke, 2024-05-03:
as stated before in other such related issues, this burden should not be put on each and every developer using this lib. It should be solved internally by the lib and the developer uses the RN coordinate system. @mrousavy 

Since I have no clue how to fix that, I can't provide any PR to help out with this.
Maybe you got some general hints or can point me to issues providing (partial) solutions to this, I'll work on a PR.

eckler-matthew, 2024-10-08:
Does anyone have any suggestions here? I'm trying a bunch of stuff as well, and can't find a way to map things correctly.