margelo/react-native-vision-camera #2553: Using an image for scanning💭

parice02, 2024-02-13:
### Question

I use react-native-vision-camera to scan barcodes and OCR from the camera. I would like to know if it is also possible to scan or OCR from an image imported from the gallery.

Best,

### VisionCamera Version

3.8.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-14:
Hey - nope, not at the moment. We're planning something though.

closed 2024-02-14 by mrousavy

xulihang, 2024-02-20:
This is possible as long as the frame processor plugin also has an interface for decoding images.

You can check out this repo: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader