margelo/react-native-vision-camera #2554: 🐛 Androod 3.9.0-beta.3 codeScanner crash still

podger, 2024-02-14:
### What's happening?

I'm dealing with this error non stop since version 3.x.
On 3.8.1 I found a workaround putting IsActive false **immediately** in the the codeScanner callback and destroying the view with a setTimeout (showCamera = false). Without doing this the camera would crash with an ML kit error very often.

On 3.8.1 I had some black screen issuers reports, so I decided to try this 3.9.0-beta.3 version. Crash on code scanner is back on with both methods: deactivating first and destroying the view with a delay or destroy directly on receiving the code scanner. The error in the log is different tough. No more ML kit involved and it's also less frequent (at least on my device).

What I noticed is that the crashes happen while trying to dismiss the camera in the codeScanner callback, If I omit the closing camera code it never crashes (as the example app) neither dismissing it with a custom button. Basically it's the combo: Code Scanned Event + Dismiss camera that goes wrong.

### Reproduceable Code

```tsx
{showCamera && authorizationCamera == "granted" && !!currentCamera && (
                <View className='w-full h-full'>
                    <Camera
                        className='w-full h-full'
                        ref={camera}
                        device={currentCamera}
                        isActive={isAppForeground && isFocused}
                        enableZoomGesture={false}
                        codeScanner={codeScanner}
                    />
                    <View className='flex-row items-center justify-around absolute bottom-2 left-0 w-full py-14 z-50'>
                       ˙ <TouchableOpacity className="rounded-full bg-white w-14 h-14 p-2 items-center" onPress={() => onCurrentDismiss()}>
                            <FontIcon size={40} name="times" color={'black'} ></FontIcon>
                        </TouchableOpacity>
                    </View>

                </View>
```

### Relevant log output

```shell
8545-8677  Persistent...ureSession com.myapp                    I  Creating new device...
2024-02-14 08:58:57.751  8545-8677  CameraManager           com.myapp                    I  Camera #0: Opening...
2024-02-14 08:58:57.752  8545-9259  DynamiteModule          com.myapp                    I  Considering local module com.google.mlkit.dynamite.barcode:10000 and remote module com.google.mlkit.dynamite.barcode:0
2024-02-14 08:58:57.752  8545-9259  DynamiteModule          com.myapp                    I  Selected local version of com.google.mlkit.dynamite.barcode
2024-02-14 08:58:57.756  8545-8545  CameraSession           com.myapp                    I  PreviewView Surface created! Surface(name=null)/@0xf7c8df0
2024-02-14 08:58:57.756  8545-8545  CameraSession           com.myapp                    I  Setting Preview Output...
2024-02-14 08:58:57.756  8545-8545  CameraSession           com.myapp                    I  PreviewView Surface updated! Surface(name=null)/@0xf7c8df0 1240 x 2205
2024-02-14 08:58:57.802  8545-8680  CameraDevices           com.myapp                    I  Camera #0 is now unavailable.
2024-02-14 08:58:57.804  8545-8677  CameraSession           com.myapp                    I  configure { ... }: Waiting for lock...
2024-02-14 08:58:57.804  8545-8678  CameraManager           com.myapp                    I  Camera #0: Opened!
2024-02-14 08:58:57.804  8545-8677  CameraSession           com.myapp                    I  configure { ... }: Waiting for lock...
2024-02-14 08:58:57.804  8545-8677  Persistent...ureSession com.myapp                    I  Creating new session...
2024-02-14 08:58:57.807  8545-8677  CreateCaptureSession    com.myapp                    I  Camera #0: Creating Capture Session #7... (Hardware Level: 1 | Outputs: [VIDEO (1280 x 720 CodeScanner for [QR] (35))])
2024-02-14 08:58:57.807  8545-8677  SurfaceOutput           com.myapp                    I  Using optimized stream use case 3 for VIDEO output.
2024-02-14 08:58:57.807  8545-8677  CreateCaptureSession    com.myapp                    I  Using new API (>=28)
2024-02-14 08:58:57.827  8545-8678  CreateCaptureSession    com.myapp                    I  Camera #0: Successfully created CameraCaptureSession #7!
2024-02-14 08:58:57.828  8545-8677  Persistent...ureSession com.myapp                    D  Stopping repeating request...
2024-02-14 08:58:57.828  8545-8677  Persistent...ureSession com.myapp                    D  Configure() done! isActive: false, ID: 0, device: android.hardware.camera2.impl.CameraDeviceImpl@80c28f, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@bb4991c
2024-02-14 08:58:57.828  8545-8677  CameraSession           com.myapp                    I  configure { ... }: Completed CameraSession Configuration! (isActive: false, isRunning: false)
2024-02-14 08:58:57.828  8545-8677  CameraView              com.myapp                    I  invokeOnInitialized()
2024-02-14 08:58:57.830  8545-8677  CameraSession           com.myapp                    I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=false, isActiveChanged=true)
2024-02-14 08:58:57.830  8545-8677  Persistent...ureSession com.myapp                    D  --> setIsActive(false)
2024-02-14 08:58:57.831  8545-8677  Persistent...ureSession com.myapp                    D  Configure() with isActive: false, ID: 0, device: android.hardware.camera2.impl.CameraDeviceImpl@80c28f, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@bb4991c
2024-02-14 08:58:57.832  8545-8677  Persistent...ureSession com.myapp                    D  Stopping repeating request...
2024-02-14 08:58:57.832  8545-8677  Persistent...ureSession com.myapp                    D  Configure() done! isActive: false, ID: 0, device: android.hardware.camera2.impl.CameraDeviceImpl@80c28f, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@bb4991c
2024-02-14 08:58:57.832  8545-8677  CameraSession           com.myapp                    I  configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: false)
2024-02-14 08:58:57.832  8545-8677  CameraSession           com.myapp                    I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-02-14 08:58:57.832  8545-8677  CameraSession           com.myapp                    I  Destroying previous outputs...
2024-02-14 08:58:57.832  8545-8677  SurfaceOutput           com.myapp                    I  Closing BarcodeScanner..
2024-02-14 08:58:57.834  8545-8677  CameraSession           com.myapp                    I  Creating outputs for Camera #0...
2024-02-14 08:58:57.841  8545-8677  CameraSession           com.myapp                    I  Adding 1920x1080 Preview Output...
2024-02-14 08:58:57.841  8545-8545  PreviewView             com.myapp                    I  Setting PreviewView Surface Size to 1920x1080...
2024-02-14 08:58:57.842  8545-8545  SurfaceHolder           com.myapp                    I  Resizing SurfaceHolder to 1920 x 1080...
2024-02-14 08:58:57.853  8545-8545  CameraSession           com.myapp                    I  PreviewView Surface updated! Surface(name=null)/@0xf7c8df0 1920 x 1080
2024-02-14 08:58:57.853  8545-8545  SurfaceHolder           com.myapp                    I  Resized SurfaceHolder to 1920 x 1080!
2024-02-14 08:58:57.859  8545-8677  CameraSession           com.myapp                    I  Adding 1280x720 CodeScanner Output in YUV_420_888...
[... 93 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 8

### VisionCamera Version

3.9.0-beta.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

dFelinger, 2024-02-15:
Something wrong on Android emulator with api level 28 while using codeScanner.
Plugin version is 3.9.0-beta.4

https://github.com/mrousavy/react-native-vision-camera/assets/5339870/12dfb876-a805-40d5-8ad1-ef0d70368f1b

Code of the app:

```javascript
function App() {
  const { hasPermission, requestPermission } = useCameraPermission();
  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  }, [hasPermission, requestPermission]);

  const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      console.log(codes);
    },
  });

  if (device == null) {
    return <Text>No device</Text>;
  }

  return (
    <View style={{ flex: 1, alignItems: 'center', justifyContent: 'center' }}>
      <Camera
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        orientation="portrait"
        audio={false}
        photo={false}
        video={false}
        codeScanner={codeScanner}
        onError={e => {
          console.log('Error', e);
        }}
      />
    </View>
  );
}
```

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

mrousavy, 2024-03-25:
If latest V4 fixes this bug, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

closed 2024-03-25 by mrousavy