margelo/react-native-vision-camera #2559: 💭 [3.9.0-beta.3]: Android reboots every time I open the Camera

svjatoslavtt, 2024-02-14:
### Question

Hi, I updated my library with the latest version `3.9.0-beta.3`, and some problems were resolved. But now I have another problem only with `Android`. It reboots when I open the camera.

In my application, we use `react-native-vision-camera` combined with `@ismaelmoreiraa/vision-camera-ocr `to read text in real-time.

Reading `Bar-Code` and `QR-code` works very well, but when I try to use `frameProcessor` (and `pixelFormat` tells me that the default format now is 'yuv') my phone reboots every time I open the camera.

Could you please give me possible solutions to fix this? Maybe I need to add some additional properties?

Here's what the component looks like:
`<Camera
      torch={torch}
      device={device}
      isActive={hasPermission}
      frameProcessor={frameProcessor}
      style={styles.camera}
/>`

### What I tried

I built and tested my application with different build versions. And every time I had some problems with building my app. 
And `[3.9.0-beta.3]` build and works very well.

Interesting fact: when I run the android simulator from Android Studio, it works fine. OCR gives me some text values and the simulator does not restart.

### VisionCamera Version

3.9.0-beta.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-14:
Hey - I just released [VisionCamera 3.9.0-beta.4](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.4) which probably fixes this issue - can you try that latest beta and let me know? :)

svjatoslavtt, 2024-02-15:
@mrousavy yes, with `3.9.0-beta.4` works pretty well, no errors at all, thanks a lot :)

closed 2024-02-15 by svjatoslavtt

mrousavy, 2024-02-15:
Awesome, quite important to fix such a critical bug 💪