margelo/react-native-vision-camera #2560: Fatal exception: CameraDevice was already closed 🐛

fsrajner, 2024-02-14:
### What's happening?

Hi! I'm trying to use the camera simply to take and save pictures, but after using it for more than some test photos, the app crashes.

Behavior:
Opening and closing the camera repeatedly, and focusing randomly causes the application to crash within a minute.

Expected Behavior:
Not crashing, or at worst restarting the camera device.

### Reproduceable Code

```tsx
import * as React from 'react'
import { useRef, useState, useCallback, useMemo, useEffect } from 'react'
import { AppState, AppStateStatus, Pressable, StyleSheet, View } from 'react-native'
import { Gesture, GestureDetector } from 'react-native-gesture-handler'
import {
    CameraDevice, CameraRuntimeError, Code, Orientation, PhotoFile, Point, useCameraDevice,
    useCameraDevices, useCameraFormat, useCameraPermission, useCodeScanner, VideoFile, Camera as VisionCamera,

    CameraProps as VisionCameraProps
} from 'react-native-vision-camera'
import Reanimated, { Easing, Extrapolation, interpolate, runOnJS, useAnimatedProps, useAnimatedStyle, useSharedValue, withRepeat, withSpring, withTiming } from 'react-native-reanimated'
import { useIsFocused } from '@react-navigation/core'
import { Icon } from '../Icon'
import { colors, layout, spacing } from 'app/theme'
import { Button } from '../Button'
import { alert, orientationCalculation } from 'app/utils/utils'
import { translate } from 'app/i18n'
import { DeviceMotion } from 'expo-sensors';

Reanimated.addWhitelistedNativeProps({
    zoom: true,
})
const ReanimatedCamera = Reanimated.createAnimatedComponent(VisionCamera)

const useIsForeground = (): boolean => {
    const [isForeground, setIsForeground] = useState(true)

    useEffect(() => {
        const onChange = (state: AppStateStatus): void => {
            setIsForeground(state === 'active')
        }
        const listener = AppState.addEventListener('change', onChange)
        return () => listener.remove()
    }, [setIsForeground])

    return isForeground
}

function usePreferredCameraDevice(_preferredDeviceId: string): [CameraDevice | undefined, (device: CameraDevice) => void] {
    const [preferredDeviceId, setPreferredDeviceId] = useState(_preferredDeviceId)

    const set = useCallback(
        (device: CameraDevice) => {
            setPreferredDeviceId(device.id)
        },
        [setPreferredDeviceId],
    )

    const devices = useCameraDevices()
    const device = useMemo(() => devices.find((d) => d.id === preferredDeviceId), [devices, preferredDeviceId])

    return [device, set]
}

export interface CameraProps {
    /**
     * An optional style override useful for padding & margin.
[... 483 lines cut ...]
    "hasTorch": true,
    "position": "back",
    "id": "0"
}
```

### Device

Samsung Galaxy S20 FE

### VisionCamera Version

3.9.0-beta.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-14:
This is in your logs:

```
02-14 16:19:39.183  3207  3248 E AndroidRuntime: 	at com.mrousavy.camera.core.PersistentCameraCaptureSession$focus$2$4.invokeSuspend(PersistentCameraCaptureSession.kt:243)
```

I'm guessing you are calling `focus()`, right?

mrousavy, 2024-02-14:
...if yes, then I already fixed this in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2550

Will release a new version later

closed 2024-02-14 by mrousavy

fsrajner, 2024-02-14:
Yes, I'm just copying the example code. Thanks for the response. will update as soon as it releases :)

mrousavy, 2024-02-14:
Hey - I just released [VisionCamera 3.9.0-beta.4](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v3.9.0-beta.4) which probably fixes this issue - can you try that latest beta and let me know? :)

fsrajner, 2024-02-14:
Thanks, it works!