margelo/react-native-vision-camera #2562: 🐛 Cannot read property 'getConstants' of null

ginxx009, 2024-02-15:
### What's happening?

Actually there's 2 error currently I am facing . 

Cannot read property 'getConstants' of null
Cannot read property 'CameraDevices' of null

So I have upgraded to version 3

"react-native-vision-camera": "^3.8.2",
"react-native-worklets-core": "^0.3.0",
"vision-camera-code-scanner": "^0.2.0",
"react-native-reanimated": "^3.5.4",
 

### Reproduceable Code

```tsx
import {
  Camera,
  useFrameProcessor,
  useCameraDevice,
  useCameraPermission,
} from 'react-native-vision-camera';
import {BarcodeFormat, scanBarcodes, Barcode} from 'vision-camera-code-scanner';

const [hasPermission, setHasPermission] = useState(false);
const { requestPermission } = useCameraPermission()
const device = useCameraDevice('back');

  useEffect(() => {
    (async () => {
      // const status = await Camera.requestCameraPermission();
      // setHasPermission(status === 'authorized');
      const status = await requestPermission();
      console.log(`status : ${status}`);
      setHasPermission(status);
    })();
  }, []);

{device && hasPermission && (
            <>
              <Camera
                style={StyleSheet.absoluteFillObject}
                device={device}
                isActive={scannerActive}
                frameProcessor={frameProcessor}
                // frameProcessorFps={1}
              />
              {/* Custom Marker */}
              <View style={styles.maskOutter}>
                <View style={[styles.maskRow, styles.maskBkg]} />
                <View style={styles.maskMid}>
                  <View style={[styles.maskCol, styles.maskBkg]} />
                  <View style={styles.maskInner}>
                    <View style={[styles.maskCorners, styles.maskBkg]} />
                    <View style={[styles.maskCorners, styles.maskBkg]} />
                    <View
                      style={[
                        styles.maskCorners,
                        styles.maskCornersBottom,
                        styles.maskBkg,
                      ]}
                    />
                    <View
                      style={[
                        styles.maskCorners,
                        styles.maskCornersBottom,
                        styles.maskBkg,
                      ]}
[... 173 lines cut ...]
### Camera Device

```json
n/a
```

### Device

Iphone 12

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-02-16 by ginxx009

haiffy420, 2024-03-20:
have you find a way to fix this

pdavid0, 2024-03-27:
@ginxx009 I have the same issue, did you manage to fix it ?

ginxx009, 2024-03-28:
Yes I did . So found out that the camera wasn't initialize properly .

simon-assistiq, 2024-04-08:
> Yes I did . So found out that the camera wasn't initialize properly .

Could you be more specific, please?