margelo/react-native-vision-camera #2572: Support retrieving raw binary data from QR codes ✨

ltrel, 2024-02-16:
### What feature or enhancement are you suggesting?

 Currently, if the data inside of the code cannot be decoded into text then no value can be retrieved at all. It would be nice to have something similar to the API provided in https://github.com/cozmo/jsQR which provides both the decoded text and raw binary representation.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

For the time being this can be achieved using frame processors, however out of all the native plugins listed at https://react-native-vision-camera.com/docs/guides/frame-processor-plugins, the two which seem like they might support this are either deprecated (https://github.com/rodgomesc/vision-camera-code-scanner/tree/master) or rely on a proprietary library (https://github.com/xulihang/vision-camera-dynamsoft-barcode-reader)

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

ltrel, 2024-02-17:
My use case is using QR codes as a sort of 'cute and tangible' offline way of exchanging data between app users. Since they're obviously pretty limited in the amount of data they can store, I've made up a binary format to store the data more compactly.

However, after a lot more digging through Stack Overflow threads, and various project documentations. it seems like support for using QR codes in this way is a little limited across the board. The ISO standard makes no mention of it and is entirely focused on various text encodings. Even the so-called "Byte Mode" really just means "a text encoding which happens to have 8-bit wide characters".

So it seems like it's really up to individual tools to decide whether they want to support a way of short-circuiting the whole decoding process like this. For example, ZBar only added an option for this in 2019 in response to [this issue](https://github.com/mchehab/zbar/issues/55)

ltrel, 2024-02-27:
Closing this for now since I've been able to put together a basic frame processor plugin for this on Android for what I need.

closed 2024-02-27 by ltrel

matheusmoreira, 2024-03-04:
> but why would you need the raw data?

The original use case for the feature was encoding 4096 bit RSA secret keys as QR codes so they could be printed out, stored in a safe and then quickly restored via laptop camera.

Others have since then found rather creative uses for it:

 - [Can you fit a whole game into a QR code?][1]

[1]: https://youtu.be/ExwqNreocpg

mrousavy, 2024-04-02:
It seems like @gev2002's [react-native-vision-camera-v3-barcode-scanner](https://github.com/gev2002/react-native-vision-camera-v3-barcode-scanner) supports getting raw values from barcodes, and uses MLKit on both iOS and Android :)

mrousavy, 2024-04-02:
another feature request created here btw: https://github.com/mrousavy/react-native-vision-camera/issues/2701