margelo/react-native-vision-camera #2577: Error on taking picture - Precapture timed out after 5 seconds!🐛

IvanByRool, 2024-02-16:
### What's happening?

Hey,

trying to take a picture with latest 3.9.0

Looks like the camera is trying to focus, but can't before timeout.

Therefore the capture crashes.

### Reproduceable Code

```tsx
<Camera
        ref={(_ref: Camera) => {
          setCameraRef(_ref);
        }}
        device={device}
        zoom={device?.neutralZoom}
        photo={true}
        format={format}
        isActive={true}
        enableDepthData={false}
        enableHighQualityPhotos={true}
        enableZoomGesture={false}
        onInitialized={() => {
          setIsCameraReady(true);
        }}
        onError={err => {
          Sentry.captureMessage('camera error on pictures' + err, 'info');
        }}
      />
```

### Relevant log output

```shell
Taking photo... Options: {qualityPrioritization=quality, enableAutoStabilization=true, flash=off, enableShutterSound=false}
2024-02-16 17:49:35.249 13778-16387 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4032x3024 image...
2024-02-16 17:49:35.249 13778-16387 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
2024-02-16 17:49:35.249 13778-16387 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
2024-02-16 17:49:35.251 13778-16387 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=OFF, pointsOfInterest=[], skipIfPassivelyFocused=true))
2024-02-16 17:49:35.557 13778-16387 Precapture              com.carooltruck.dev                  I  Auto-Flash: Flash is not required for photo capture.
2024-02-16 17:49:35.557 13778-16387 Precapture              com.carooltruck.dev                  I  AF is already focused, skipping...
2024-02-16 17:49:35.913 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:35.913 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:35.952 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:35.952 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:35.994 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:35.994 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.034 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.034 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.079 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.079 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.119 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.119 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.162 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.163 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.203 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.203 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.244 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.244 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.287 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.288 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.329 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.329 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.370 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.370 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Searching (isCompleted: false)
2024-02-16 17:49:36.413 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AE State: Precapture (isCompleted: false)
2024-02-16 17:49:36.413 13778-16387 CameraCaptureSession    com.carooltruck.dev                  I  AWB State: Converged (isCompleted: true)
[... 231 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung SM-G960U1 / Android 10

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-16:
Does this always happen? Is this phone normally fast with capture in the native Camera app?

It looks like AE is just always on Precapture, not sure if it is adjusting or if we can already shoot a picture there..

IvanByRool, 2024-02-16:
Same behaviour on samsung SM-T545 / Android 11.0
the only devices I own ;)

dibyendu, 2024-02-18:
Happens in Pixel 6 (Android 14). After switching from 3.8.2 to 3.9.0.

Dingenis, 2024-02-18:
Same here, only with the front camera. Tested on the Samsung SM-A536B / Android 14

fsteinhanseskb, 2024-02-19:
Same problem with the back camera. Works fine on beta-2.

mrousavy, 2024-02-19:
Created a PR, can you try if this fixes the issue for you guys? https://github.com/mrousavy/react-native-vision-camera/pull/2586

mrousavy, 2024-02-19:
According to the [Android Developer Documentation on the AE State values](https://developer.android.com/reference/android/hardware/camera2/CaptureResult#CONTROL_AE_STATE), is `Precapture` only a temporary state and should resolve to either `Locked` or `Converged` once it finished finding a good AE value.

The way I understood this, is it should never remain in this state, so it is only a temporary state. This makes me wonder, how can a fairly new device (Samsung Galaxy S9, Level: FULL Camera) not find a good exposure value in 5 seconds?

[image]

iliapnmrv, 2024-02-19:
hey! for me https://github.com/mrousavy/react-native-vision-camera/pull/2586 did not resolve the issue. Xiaomi Redmi 9T

Dingenis, 2024-02-19:
hey mrousavy, thanks for taking a look at this so quickly! 🙏 On my Samsung SM-A536B (Android 14) I can no longer reproduce the issue (capture/timed-out), with #2586.

iliapnmrv, 2024-02-19:
@mrousavy 
i set branch as a package
```json
"react-native-vision-camera": "https://gitpkg.now.sh/mrousavy/react-native-vision-camera/package?fix/precapture-timeout"
```
then i simply use camera im my app (scan barcodes, open and close modals) and crash happens
here is my sentry issue
https://sentry.vibeapp.ru/share/issue/9aa315f0d8114f13bad62ee086b7ee85/

do i do something wrong?

IvanByRool, 2024-02-19:
I'm in a meeting today, I'll test this on both devices tonight.

IvanByRool, 2024-02-19:
Hey, 

I've tested, it's "working" but it's too slow when using the flash

In the log below you'll see that it's taking the picture (when using the flash) after the 5s timeout.
The first picture on the other hand (without flash) is working fine directly.

Let me know if you need more.

Camera Device
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
Device
Samsung SM-G960U1 / Android 10

`Taking photo... Options: {qualityPrioritization=quality, enableAutoStabilization=true, flash=off, enableShutterSound=false}
2024-02-19 20:35:58.120 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4032x3024 image...
2024-02-19 20:35:58.120 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
2024-02-19 20:35:58.120 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
2024-02-19 20:35:58.121 20753-28926 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=OFF, pointsOfInterest=[], skipIfPassivelyFocused=true, timeoutMs=5000))
2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  Precapture current states: AF: PassiveFocused, AE: Converged, AWB: Converged
2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  AF is already focused, skipping...
2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  AE is already focused, skipping...
2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  AWB is already focused, skipping...
2024-02-19 20:35:58.600 20753-28926 CameraCaptureSession    com.carooltruck.dev                  I  Precapture state: AF: PassiveFocused, AE: Converged, AWB: Converged
2024-02-19 20:35:58.605 20753-28926 Precapture              com.carooltruck.dev                  I  AF/AE/AWB successfully locked!
2024-02-19 20:35:58.605 20753-28926 PhotoCaptureRequest     com.carooltruck.dev                  I  Using CaptureRequest Template PHOTO...
2024-02-19 20:35:58.938 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo Captured!
2024-02-19 20:35:58.938 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo captured! 4032 x 3024
2024-02-19 20:35:58.968 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 2/3 - waiting for image with timestamp 641711024794574 now...
2024-02-19 20:35:58.968 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 3/3 - received 4032 x 3024 image, preparing result...
2024-02-19 20:35:58.972 20753-28926 CameraView.takePhoto    com.carooltruck.dev                  I  Successfully captured 4032 x 3024 photo!
2024-02-19 20:35:59.002 20753-28926 CameraView.takePhoto    com.carooltruck.dev                  I  Successfully saved photo to file! /data/user/0/com.carooltruck.dev/cache/mrousavy5278860813447148588.jpg
2024-02-19 20:35:59.321 20753-28724 System                  com.carooltruck.dev                  W  A resource failed to call close. 
2024-02-19 20:35:59.338 20753-28927 CreateCaptureSession    com.carooltruck.dev                  I  Camera #0: CameraCaptureSession #6 has been closed.
2024-02-19 20:35:59.339 20753-28927 Persistent...ureSession com.carooltruck.dev                  I  Session android.hardware.camera2.impl.CameraCaptureSessionImpl@b2c1223 closed!
2024-02-19 20:36:01.120 20753-20753 ViewRootIm...nActivity] com.carooltruck.dev                  I  ViewPostIme pointer 0
2024-02-19 20:36:01.235 20753-20753 ViewRootIm...nActivity] com.carooltruck.dev                  I  ViewPostIme pointer 1
2024-02-19 20:36:01.249 20753-20753 CameraView              com.carooltruck.dev                  D  Finding view 1355...
2024-02-19 20:36:01.250 20753-20753 CameraView              com.carooltruck.dev                  D  Found view 1355!
2024-02-19 20:36:01.250 20753-28926 CameraView.takePhoto    com.carooltruck.dev                  I  Taking photo... Options: {qualityPrioritization=quality, enableAutoStabilization=true, flash=on, enableShutterSound=false}
2024-02-19 20:36:01.251 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4032x3024 image...
2024-02-19 20:36:01.251 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
2024-02-19 20:36:01.251 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
2024-02-19 20:36:01.253 20753-28926 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=ON, pointsOfInterest=[], skipIfPassivelyFocused=false, timeoutMs=5000))
2024-02-19 20:36:01.593 20753-28926 CameraCaptureSession    com.carooltruck.dev                  I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
[... cut ...]
2024-02-19 20:36:09.187 20753-20753 SurfaceOutput           com.carooltruck.dev                  I  Closing 4032x3024 PHOTO ImageReader..
2024-02-19 20:36:09.188 20753-28732 CameraManagerGlobal     com.carooltruck.dev                  I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.carooltruck.dev API Level 2
2024-02-19 20:36:09.188 20753-20753 CameraSession           com.carooltruck.dev                  I  CameraSession closed!
2024-02-19 20:36:09.188 20753-28929 CameraDevices           com.carooltruck.dev                  I  Camera #0 is now available.`

mrousavy, 2024-02-20:
why tf is that phone not leaving the Precapture state???

IvanByRool, 2024-02-20:
If it helps, on SM-T545 / Android 11.0

Same behavior

```shell
Taking photo... Options: {qualityPrioritization=quality, enableAutoStabilization=true, flash=off, enableShutterSound=false}
2024-02-20 09:56:16.177 22127-25818 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4128x3096 image...
2024-02-20 09:56:16.178 22127-25818 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
2024-02-20 09:56:16.178 22127-25818 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
2024-02-20 09:56:16.180 22127-25818 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=OFF, pointsOfInterest=[], skipIfPassivelyFocused=true, timeoutMs=5000))
2024-02-20 09:56:16.464 22127-25818 Precapture              com.carooltruck.dev                  I  Precapture current states: AF: PassiveFocused, AE: Converged, AWB: Converged
2024-02-20 09:56:16.464 22127-25818 Precapture              com.carooltruck.dev                  I  AF is already focused, skipping...
2024-02-20 09:56:16.464 22127-25818 Precapture              com.carooltruck.dev                  I  AE is already focused, skipping...
2024-02-20 09:56:16.464 22127-25818 Precapture              com.carooltruck.dev                  I  AWB is already focused, skipping...
2024-02-20 09:56:16.784 22127-25818 CameraCaptureSession    com.carooltruck.dev                  I  Precapture state: AF: PassiveFocused, AE: Converged, AWB: Converged
2024-02-20 09:56:16.786 22127-25818 Precapture              com.carooltruck.dev                  I  AF/AE/AWB successfully locked!
2024-02-20 09:56:16.786 22127-25818 PhotoCaptureRequest     com.carooltruck.dev                  I  Using CaptureRequest Template PHOTO...
2024-02-20 09:56:17.192 22127-25818 CameraSession           com.carooltruck.dev                  I  Photo Captured!
2024-02-20 09:56:17.192 22127-25818 CameraSession           com.carooltruck.dev                  I  Photo captured! 4128 x 3096
2024-02-20 09:56:17.203 22127-25818 CameraSession           com.carooltruck.dev                  I  Photo capture 2/3 - waiting for image with timestamp 2075721113417426 now...
2024-02-20 09:56:17.204 22127-25818 CameraSession           com.carooltruck.dev                  I  Photo capture 3/3 - received 4128 x 3096 image, preparing result...
2024-02-20 09:56:17.213 22127-25818 CameraView.takePhoto    com.carooltruck.dev                  I  Successfully captured 4128 x 3096 photo!
2024-02-20 09:56:17.239 22127-25818 CameraView.takePhoto    com.carooltruck.dev                  I  Successfully saved photo to file! /data/user/0/com.carooltruck.dev/cache/mrousavy8816217373587776219.jpg
2024-02-20 09:56:17.314 22127-25822 ReactNativeJS           com.carooltruck.dev                  I  { url: '/configuration/customers/carool/configurations',
                                                                                                      method: 'GET',
                                                                                                      headers: { 'Content-Type': 'application/json' } }
2024-02-20 09:56:17.696 22127-25264 System                  com.carooltruck.dev                  W  A resource failed to call close. 
2024-02-20 09:56:18.233 22127-25262 carooltruck.de          com.carooltruck.dev                  I  NativeAlloc concurrent copying GC freed 12397(781KB) AllocSpace objects, 2(40KB) LOS objects, 49% free, 12MB/25MB, paused 125us total 107.089ms
2024-02-20 09:56:18.251 22127-25820 CreateCaptureSession    com.carooltruck.dev                  I  Camera #0: CameraCaptureSession #6 has been closed.
2024-02-20 09:56:18.251 22127-25820 Persistent...ureSession com.carooltruck.dev                  I  Session android.hardware.camera2.impl.CameraCaptureSessionImpl@4ecb1b0 closed!
2024-02-20 09:56:20.780 22127-22127 Choreographer           com.carooltruck.dev                  W  Frame time is 0.040035 ms in the future!  Check that graphics HAL is generating vsync timestamps using the correct timebase.
2024-02-20 09:56:21.080 22127-22127 ViewRootIm...nActivity] com.carooltruck.dev                  I  ViewPostIme pointer 0
2024-02-20 09:56:21.207 22127-22127 ViewRootIm...nActivity] com.carooltruck.dev                  I  ViewPostIme pointer 1
2024-02-20 09:56:21.231 22127-22127 CameraView              com.carooltruck.dev                  D  Finding view 1575...
2024-02-20 09:56:21.231 22127-22127 CameraView              com.carooltruck.dev                  D  Found view 1575!
2024-02-20 09:56:21.232 22127-25818 CameraView.takePhoto    com.carooltruck.dev                  I  Taking photo... Options: {qualityPrioritization=quality, enableAutoStabilization=true, flash=on, enableShutterSound=false}
2024-02-20 09:56:21.232 22127-25818 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4128x3096 image...
2024-02-20 09:56:21.232 22127-25818 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
2024-02-20 09:56:21.232 22127-25818 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
2024-02-20 09:56:21.233 22127-25818 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=ON, pointsOfInterest=[], skipIfPassivelyFocused=false, timeoutMs=5000))
[... cut ...]
2024-02-20 09:56:30.534 22127-22127 CameraSession           com.carooltruck.dev                  I  configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: false)
2024-02-20 09:56:30.534 22127-22127 CameraSession           com.carooltruck.dev                  I  Preview Output destroyed!
2024-02-20 09:56:30.534 22127-22127 CameraView              com.carooltruck.dev                  I  Updating CameraSession...
2024-02-20 09:56:30.535 22127-22127 CameraSession           com.carooltruck.dev                  I  Closing CameraSession...
2024-02-20 09:56:30.535 22127-22127 CameraSession           com.carooltruck.dev                  I  Destroying session..

fsteinhanseskb, 2024-02-20:
> Hey,
> 
> I've tested, it's "working" but it's too slow when using the flash
> 
> In the log below you'll see that it's taking the picture (when using the flash) after the 5s timeout. The first picture on the other hand (without flash) is working fine directly.
> 
> Let me know if you need more.
> 
> Camera Device
> 
> ```
> {
>   "formats": [],
>   "sensorOrientation": "landscape-left",
>   "hardwareLevel": "full",
>   "maxZoom": 8,
>   "minZoom": 1,
>   "maxExposure": 20,
>   "supportsLowLightBoost": false,
>   "neutralZoom": 1,
>   "physicalDevices": [
>     "wide-angle-camera"
>   ],
>   "supportsFocus": true,
>   "supportsRawCapture": true,
>   "isMultiCam": false,
>   "minFocusDistance": 10,
>   "minExposure": -20,
>   "name": "BACK (0)",
>   "hasFlash": true,
>   "hasTorch": true,
>   "position": "back",
>   "id": "0"
> }
> ```
> 
> Device Samsung SM-G960U1 / Android 10
> 
> ```
> 2024-02-19 20:35:58.120 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4032x3024 image...
> 2024-02-19 20:35:58.120 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
> 2024-02-19 20:35:58.120 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
> 2024-02-19 20:35:58.121 20753-28926 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=OFF, pointsOfInterest=[], skipIfPassivelyFocused=true, timeoutMs=5000))
> 2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  Precapture current states: AF: PassiveFocused, AE: Converged, AWB: Converged
> 2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  AF is already focused, skipping...
> 2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  AE is already focused, skipping...
> 2024-02-19 20:35:58.328 20753-28926 Precapture              com.carooltruck.dev                  I  AWB is already focused, skipping...
> 2024-02-19 20:35:58.600 20753-28926 CameraCaptureSession    com.carooltruck.dev                  I  Precapture state: AF: PassiveFocused, AE: Converged, AWB: Converged
> 2024-02-19 20:35:58.605 20753-28926 Precapture              com.carooltruck.dev                  I  AF/AE/AWB successfully locked!
> 2024-02-19 20:35:58.605 20753-28926 PhotoCaptureRequest     com.carooltruck.dev                  I  Using CaptureRequest Template PHOTO...
> 2024-02-19 20:35:58.938 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo Captured!
> 2024-02-19 20:35:58.938 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo captured! 4032 x 3024
> 2024-02-19 20:35:58.968 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 2/3 - waiting for image with timestamp 641711024794574 now...
> 2024-02-19 20:35:58.968 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 3/3 - received 4032 x 3024 image, preparing result...
> 2024-02-19 20:35:58.972 20753-28926 CameraView.takePhoto    com.carooltruck.dev                  I  Successfully captured 4032 x 3024 photo!
> 2024-02-19 20:35:59.002 20753-28926 CameraView.takePhoto    com.carooltruck.dev                  I  Successfully saved photo to file! /data/user/0/com.carooltruck.dev/cache/mrousavy5278860813447148588.jpg
> 2024-02-19 20:35:59.321 20753-28724 System                  com.carooltruck.dev                  W  A resource failed to call close. 
> 2024-02-19 20:35:59.338 20753-28927 CreateCaptureSession    com.carooltruck.dev                  I  Camera #0: CameraCaptureSession #6 has been closed.
> 2024-02-19 20:35:59.339 20753-28927 Persistent...ureSession com.carooltruck.dev                  I  Session android.hardware.camera2.impl.CameraCaptureSessionImpl@b2c1223 closed!
> 2024-02-19 20:36:01.120 20753-20753 ViewRootIm...nActivity] com.carooltruck.dev                  I  ViewPostIme pointer 0
> 2024-02-19 20:36:01.235 20753-20753 ViewRootIm...nActivity] com.carooltruck.dev                  I  ViewPostIme pointer 1
> 2024-02-19 20:36:01.249 20753-20753 CameraView              com.carooltruck.dev                  D  Finding view 1355...
> 2024-02-19 20:36:01.250 20753-20753 CameraView              com.carooltruck.dev                  D  Found view 1355!
> 2024-02-19 20:36:01.250 20753-28926 CameraView.takePhoto    com.carooltruck.dev                  I  Taking photo... Options: {qualityPrioritization=quality, enableAutoStabilization=true, flash=on, enableShutterSound=false}
> 2024-02-19 20:36:01.251 20753-28926 CameraSession           com.carooltruck.dev                  I  Photo capture 1/3 - capturing 4032x3024 image...
> 2024-02-19 20:36:01.251 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Capturing photo...
> 2024-02-19 20:36:01.251 20753-28926 Persistent...ureSession com.carooltruck.dev                  I  Locking AF/AE/AWB...
> 2024-02-19 20:36:01.253 20753-28926 Precapture              com.carooltruck.dev                  I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE, AWB], flash=ON, pointsOfInterest=[], skipIfPassivelyFocused=false, timeoutMs=5000))
[... cut ...]
> 2024-02-19 20:36:09.188 20753-28732 CameraManagerGlobal     com.carooltruck.dev                  I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.carooltruck.dev API Level 2
> 2024-02-19 20:36:09.188 20753-20753 CameraSession           com.carooltruck.dev                  I  CameraSession closed!
> 2024-02-19 20:36:09.188 20753-28929 CameraDevices           com.carooltruck.dev                  I  Camera #0 is now available.
> ```

Just integrated the fix branch. I'm experiencing the same, also without flash. Logs are looking the same. Device: Samsung Galaxy S21 FE 5G. Thank you for your support.

theopomies, 2024-02-20:
Hi, I get a similar issue since upgrading to 3.9.0 to fix a preview stretch (not resolved, might open an issue for this).

The logs are very similar
```
[capture/timed-out: [capture/timed-out] The image capture was aborted because it timed out.]
```
02-20 11:49:38.141 10988 11328 I CameraSession: configure { ... }: Waiting for lock...
02-20 11:49:38.141 10988 11329 I CameraManager: Camera #1: Opened!
02-20 11:49:38.141  1370  1587 W BatteryUsageReport: Warning: when report...
02-20 11:49:38.143 10988 11328 I CameraSession: configure { ... }: Waiting for lock...
02-20 11:49:38.144 10988 11328 I PersistentCameraCaptureSession: Creating new session...
02-20 11:49:38.147 10988 11328 I CreateCaptureSession: Camera #1: Creating Capture Session #1... (Hardware Level: 0 | Outputs: [PHOTO (2592x1944 in JPEG)])
02-20 11:49:38.148 10988 11328 I CreateCaptureSession: Using new API (>=28)
02-20 11:49:38.155   638   693 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [4, 0]
02-20 11:49:38.155   638   693 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [4, 1]
02-20 11:49:38.156   617 11506 I chatty  : uid=1047(cameraserver) provider@2.4-se expire 108 lines
02-20 11:49:38.157   617 11534 I chatty  : uid=1047(cameraserver) provider@2.4-se expire 6 lines
02-20 11:49:38.165 10988 11329 I CreateCaptureSession: Camera #1: Successfully created CameraCaptureSession #1!
02-20 11:49:38.166   634 22647 E WifiHAL : Invalid data received from driver
02-20 11:49:38.166 10988 11328 D PersistentCameraCaptureSession: Stopping repeating request...
02-20 11:49:38.167 10988 11328 D PersistentCameraCaptureSession: Configure() done! isActive: false, ID: 1, device: android.hardware.camera2.impl.CameraDeviceImpl@62c2cf9, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@819343e
02-20 11:49:38.167 10988 11328 I CameraSession: configure { ... }: Completed CameraSession Configuration! (isActive: false, isRunning: false)
02-20 11:49:38.167 10988 11328 I CameraView: invokeOnInitialized()
02-20 11:49:38.169 10988 11328 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=false, isActiveChanged=true)
02-20 11:49:38.169 10988 11328 D PersistentCameraCaptureSession: --> setIsActive(false)
02-20 11:49:38.170 10988 11328 D PersistentCameraCaptureSession: Configure() with isActive: false, ID: 1, device: android.hardware.camera2.impl.CameraDeviceImpl@62c2cf9, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@819343e
02-20 11:49:38.171 10988 11328 D PersistentCameraCaptureSession: Stopping repeating request...
02-20 11:49:38.171 10988 11328 D PersistentCameraCaptureSession: Configure() done! isActive: false, ID: 1, device: android.hardware.camera2.impl.CameraDeviceImpl@62c2cf9, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@819343e
02-20 11:49:38.172 10988 11328 I CameraSession: configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: false)
02-20 11:49:38.172 10988 11328 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
02-20 11:49:38.173 10988 11328 I CameraSession: Destroying previous outputs...
02-20 11:49:38.173 10988 11328 I SurfaceOutput: Closing 2592x1944 PHOTO ImageReader..
02-20 11:49:38.174 10988 11328 I CameraSession: Creating outputs for Camera #1...
02-20 11:49:38.181 10988 11328 I CameraSession: Adding 2592x1944 Photo Output in JPEG...
02-20 11:49:38.186 10988 11328 I CameraSession: Adding 1024x768 Preview Output...
02-20 11:49:38.187 10988 10988 I PreviewView: Input Orientation changed: LANDSCAPE_LEFT -> LANDSCAPE_RIGHT
02-20 11:49:38.188 10988 10988 I SurfaceHolder: Resizing SurfaceHolder to 1024 x 768...
02-20 11:49:38.191 10988 10988 I PreviewView: Surface Size changed: 800x1216 -> 1024x768
02-20 11:49:38.192 10988 10988 I CameraSession: PreviewView Surface updated! Surface(name=null)/@0x2dca543 1024 x 768
02-20 11:49:38.192 10988 10988 I SurfaceHolder: Resized SurfaceHolder to 1024 x 768!
02-20 11:49:38.195 10988 11328 D PersistentCameraCaptureSession: --> setOutputs([PHOTO (2592x1944 in JPEG), PREVIEW (1024 x 768)])
02-20 11:49:38.195 10988 11328 I CameraSession: Successfully configured Session with 2 outputs for Camera #1!
02-20 11:49:38.196 10988 11328 D PersistentCameraCaptureSession: --> setRepeatingRequest(...)
02-20 11:49:38.196 10988 11328 D PersistentCameraCaptureSession: --> setIsActive(true)
02-20 11:49:38.196 10988 11328 D PersistentCameraCaptureSession: Configure() with isActive: true, ID: 1, device: android.hardware.camera2.impl.CameraDeviceImpl@62c2cf9, session: null
02-20 11:49:38.197 10988 11328 I PersistentCameraCaptureSession: Creating new session...
02-20 11:49:38.199 10988 11328 I CreateCaptureSession: Camera #1: Creating Capture Session #2... (Hardware Level: 0 | Outputs: [PHOTO (2592x1944 in JPEG), PREVIEW (1024 x 768)])
02-20 11:49:38.200 10988 11328 I CreateCaptureSession: Using new API (>=28)
02-20 11:49:38.200 10988 11329 I CreateCaptureSession: Camera #1: CameraCaptureSession #1 has been closed.
02-20 11:49:38.200 10988 11329 I PersistentCameraCaptureSession: Session android.hardware.camera2.impl.CameraCaptureSessionImpl@819343e closed!
02-20 11:49:38.204   638   693 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [4, 0]
[... cut ...]
02-20 11:49:39.301 10988 11328 I CameraCaptureSession: AE State: Converged (isCompleted: true)
02-20 11:49:39.301 10988 11328 I CameraCaptureSession: AWB State: Converged (isCompleted: true)
02-20 11:49:39.333 10988 11328 I CameraCaptureSession: AF State: Inactive (isCompleted: false)
02-20 11:49:39.333 10988 11328 I CameraCaptureSession: AE State: Converged (isCompleted: true)
02-20 11:49:39.334 10988 11328 I CameraCaptureSession: AWB State: Converged (isCompleted: true)
```

Device is a CrossCall Core-T4 running android 10

iliapnmrv, 2024-02-20:
Hey! In case you still need logs. I focus and then tried to leave the screen a few seconds later.
Samsung S9 Android 10

```log
02-20 18:36:14.884  8666  8741 I CameraSession: configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: true)
02-20 18:36:14.884  8666  8741 I CameraView: invokeOnStarted()
02-20 18:36:15.105  8666  8743 I CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client com.ticketscloud.scanner API Level 2
02-20 18:36:15.154  8666  8893 W tscloud.scanne: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
02-20 18:36:15.156  8666  8893 W tscloud.scanne: Accessing hidden method Lsun/misc/Unsafe;->getInt(Ljava/lang/Object;J)I (greylist, linking, allowed)
02-20 18:36:15.161  8666  8893 I tflite  : Replacing 43 out of 43 node(s) with delegate (TfLiteXNNPackDelegate) node, yielding 1 partitions for the whole graph.
02-20 18:36:15.163  8666  8893 I tflite  : Replacing 42 out of 47 node(s) with delegate (TfLiteXNNPackDelegate) node, yielding 11 partitions for the whole graph.
02-20 18:36:15.165  8666  8893 I native  : I0000 00:00:1708443375.165279    8893 oned_decoder_client.cc:695] barhopper::deep_learning::OnedDecoderClient is created successfully.
02-20 18:36:15.176  8666  8893 W tscloud.scanne: Accessing hidden method Lsun/misc/Unsafe;->putInt(Ljava/lang/Object;JI)V (greylist, linking, allowed)
02-20 18:36:16.419  8666  8914 W tscloud.scanne: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
02-20 18:36:17.217  8666  8893 W tscloud.scanne: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
02-20 18:36:18.887  8666  8666 I ViewRootImpl@f65fc1c[MainActivity]: ViewPostIme pointer 0
02-20 18:36:18.919  8666  8666 I ViewRootImpl@f65fc1c[MainActivity]: ViewPostIme pointer 1
02-20 18:36:18.920  8666  8666 W unknown:ReactNative: Can't cancel already finished gesture. Is a child View trying to start a gesture from an UP/CANCEL event?
02-20 18:36:18.927  8666  8666 D CameraView: Finding view 649...
02-20 18:36:18.927  8666  8666 D CameraView: Found view 649!
02-20 18:36:18.928  8666  8741 I ROTATE  : Point(129, 270) -> PointF(0.35833332, 0.421875) -> LANDSCAPE_LEFT -> 1701.0, 1940.3999
02-20 18:36:18.928  8666  8741 I PreviewView: Converted layer point Point(129, 270) to camera point Point(1701, 1940)! (LANDSCAPE_LEFT, 4032x3024 -> 360x640)
02-20 18:36:18.928  8666  8741 I PersistentCameraCaptureSession: Focusing to Point(1701, 1940)...
02-20 18:36:18.931  8666  8741 I Precapture: Running precapture sequence... (PrecaptureOptions(modes=[AF, AE], flash=OFF, pointsOfInterest=[Point(1701, 1940)], skipIfPassivelyFocused=false, timeoutMs=3000))
02-20 18:36:19.539  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:19.599  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:19.669  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:19.739  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:19.802  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
02-20 18:36:19.874  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
02-20 18:36:19.941  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.005  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.073  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
02-20 18:36:20.143  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
02-20 18:36:20.208  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.276  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.344  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
02-20 18:36:20.410  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
02-20 18:36:20.478  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.500  8666  8898 W tscloud.scanne: Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
02-20 18:36:20.543  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.605  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
02-20 18:36:20.607  8666  8666 I ViewRootImpl@f65fc1c[MainActivity]: ViewPostIme key 0
02-20 18:36:20.658  8666  8666 I ViewRootImpl@f65fc1c[MainActivity]: ViewPostIme key 1
02-20 18:36:20.669  8666  8741 I CameraCaptureSession: Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
[... cut ...]
02-20 18:36:22.559  8666  8741 E AndroidRuntime:        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
02-20 18:36:22.559  8666  8741 E AndroidRuntime:        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
02-20 18:36:22.559  8666  8741 E AndroidRuntime:        Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@9ca8947, mrousavy/VisionCamera.main]
```

vishpatil1999, 2024-02-28:
> hey mrousavy, thanks for taking a look at this so quickly! 🙏 On my Samsung SM-A536B (Android 14) I can no longer reproduce the issue (capture/timed-out), with #2586.

which version of react-native-vision-camera you used

Dingenis, 2024-02-28:
hey @vishpatil1999, I didn't use a specific version, I used the fix/precapture-timeout branch to test if it would solve the problem. This branch is merged into main but there is no version yet that incoperates this fix.

vishpatil1999, 2024-02-28:
> hey @vishpatil1999, I didn't use a specific version, I used the fix/precapture-timeout branch to test if it would solve the problem. This branch is merged into main but there is no version yet that incoperates this fix.

can you please explain how to use this branch in react native project

iliapnmrv, 2024-02-28:
@vishpatil1999 this comment may help you https://github.com/mrousavy/react-native-vision-camera/issues/2577#issuecomment-1952390114

GodwinChibyk, 2024-02-29:
this is what solved it for me 
`photoData = await cameraRef.current?.takePhoto({
        enableShutterSound: true,
        flash: 'off',
        qualityPrioritization: Platform.OS === 'android' ? 'speed' : 'quality',
      });`

mrousavy, 2024-02-29:
Yea good point - if you set `qualityPrioritization` to `speed`, I will skip the entire AF/AE/AWB precapture sequence and instead just capture as fast as possible. This avoids all the precapture/timeout errors.

serybva, 2024-02-29:
Also happens on OnePlus Nord, Android 12.
Happens only with flash enabled, `qualityPrioritization` option doesn't solve the issue

IvanByRool, 2024-02-29:
> Yea good point - if you set `qualityPrioritization` to `speed`, I will skip the entire AF/AE/AWB precapture sequence and instead just capture as fast as possible. This avoids all the precapture/timeout errors.

If we do so then we skip focus right ? I'm not sure about picture quality then. Thx

serybva, 2024-02-29:
> @serybva yes if flash is enabled it will still require the precapture process, so this only works without flash.
Thanks for the explanation
I did test your PR tho and it fixes the issue, guess I just have to wait for the next release

arslanafzal077, 2024-03-07:
i'm still facing this issue if flash is on but it works fine when the flash is off

Fei-au, 2024-03-07:
same here when have qualityPrioritization: 'speed' on, comment it out will work.

viniciusyoshioka, 2024-03-08:
I still having this issue on 3.9.1, but not when trying to take a picture, it happens when trying to focus.

The logs are too similar:
```
2024-03-08 11:50:59.990 10860-10953 ROTATE                  com.docscan                          I  Point(313, 53) -> PointF(0.86944443, 0.0828125) -> LANDSCAPE_LEFT -> 333.90002, 394.80005
2024-03-08 11:50:59.990 10860-10953 PreviewView             com.docscan                          I  Converted layer point Point(313, 53) to camera point Point(333, 394)! (LANDSCAPE_LEFT, 4032x3024 -> 360x640)
2024-03-08 11:50:59.990 10860-10953 Persistent...ureSession com.docscan                          I  Focusing to Point(333, 394)...
2024-03-08 11:50:59.994 10860-10953 Precapture              com.docscan                          I  Running precapture sequence... (PrecaptureOptions(modes=[AF, AE], flash=OFF, pointsOfInterest=[Point(333, 394)], skipIfPassivelyFocused=false, timeoutMs=3000))
2024-03-08 11:51:00.265 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.299 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.332 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.366 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.399 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:00.431 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:00.465 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:00.498 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Focused, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.530 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.564 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.601 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.635 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.670 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.704 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.735 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.770 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Focused, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.802 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:00.836 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:00.869 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.903 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.937 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Searching
2024-03-08 11:51:00.970 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:01.004 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:01.037 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Focused, AE: Precapture, AWB: Converged
2024-03-08 11:51:01.069 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:01.103 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:01.138 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
2024-03-08 11:51:01.180 10860-10953 CameraCaptureSession    com.docscan                          I  Precapture state: AF: Scanning, AE: Precapture, AWB: Converged
[... cut ...] owed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.ActivityRecord.destroySurfaces:6539 com.android.server.wm.ActivityRecord.destroySurfaces:6520 com.android.server.wm.WindowState.onExitAnimationDone:5995 com.android.server.wm.ActivityRecord$$ExternalSyntheticLambda10.accept:2 java.util.ArrayList.forEach:1528 com.android.server.wm.ActivityRecord.onAnimationFinished:8607 com.android.server.wm.ActivityRecord.postApplyAnimation:6252 
```

Device: SM-G780G
Android: 13

mrousavy, 2024-03-19:
Hey everybody! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

sauravjmedhi, 2024-03-20:
> Hey everybody! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test [...]

We were facing a bunch of issues in multiple apps like image orientation, capture issues, etc. Those have been fixed and working in the beta update.

When can we expect a full rollout?

mrousavy, 2024-03-21:
> Those have been fixed and working in the beta update.

Awesome, great to hear!

arslanafzal077, 2024-03-21:
"react-native-vision-camera": "^4.0.0-beta.10",
black screen appears when open the camera

mrousavy, 2024-03-25:
Great to hear that V4 now works perfectly! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

ChristopherGabba, 2024-03-30:
+1 On this issue. I'm trying to take a photo while recording with a Google Pixel 3 and it throws this error. 
`"[capture/timed-out] The image capture was aborted because it timed out."` 

Using v4.0.0beta11

mrousavy, 2024-04-02:
@ChristopherGabba this is a different bug the one reported here was happening in my custom precapture sequence on V3, which is no longer present on V4.

But it is quite interesting - this is definitely a Google CameraX bug that should be reported in their issue tracker.

ChristopherGabba, 2024-04-02:
@mrousavy Hey Marc, thanks for the clarification. So just to clarify, taking a photo mid-video is currently not supported on Google Pixels? This seems out there as they are one of the top phone manufacturers out there and that would seem like a relatively basic feature. Do you have a link to a repo I could submit a new bug for (I'm guessing RNVC is not the place for this bug?)

mrousavy, 2024-04-02:
Please [create an issue in the CameraX issue tracker here](https://issuetracker.google.com/issues/new?component=618491&template=1257717), this is definitely a bug. On iOS this works, and on other Android phones this shoulda also work.

CameraX version is 1.4.0-alpha04.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

mostafaarafa-valeo, 2024-05-09:
I was having the same issue, upgraded to 4.0.3
The app stopped crashing but the focus doesn't work on android

PR #2586 (2024-02-20): fix: Fix `Precapture timed out after 5 seconds` error

## What

Fixes the "Precapture timed out after 5 seconds" error by changing two things:

1. If AE, AF, or AWB is not supported on the device, remove the unsupported value from the precapture trigger.
2. If the precapture timed out after 5 seconds, we still take a picture. Might be unfocused, but who cares. (not sure if this is a good idea tbh)

## Changes

## Tested on

## Related issues

- Fixes #2577