margelo/react-native-vision-camera #2579: 🐛 Frame Processor stops running when performing calculations

mfkrause, 2024-02-17:
### What's happening?

I'm trying to update some Shared Values from Reanimated (using `createRunInJsFn`) based on the results of a frame processor plugin and the size of the frame itself. Specifically, I'm detecting faces using a plugin (which uses MLKit) and then want to normalize the detected bounding box based on the frame and screen size.

However, the frame processor suddenly stops running after only a few frames. Console logs stop appearing and, if the result is drawn onto a Skia Canvas, that stops updating, too. The weird part: When I remove the normalization / multiplication from the code below and simply use the raw result bounding box from the plugin, the frame processor happily runs for seemingly forever. Only once I begin to multiply the bounding box with the frame size (and window size), the frame processor only runs for a few frames.

The problem is not specific to the frame processor plugin used in the reproducable code below. I could also reproduce it when using a custom model and react-native-fast-tflite.

### Reproduceable Code

```tsx
import { useEffect } from "react";
import { StyleSheet, View, useWindowDimensions } from "react-native";
import {
  Camera,
  runAsync,
  useCameraDevice,
  useCameraPermission,
  useFrameProcessor,
} from "react-native-vision-camera";
import { Worklets } from "react-native-worklets-core";
import {
  DetectionResult,
  detectFaces,
  Bounds as FaceBounds,
} from "react-native-vision-camera-face-detector";
import { useSharedValue } from "react-native-reanimated";

export default function App() {
  const { hasPermission, requestPermission } = useCameraPermission();
  const dimensions = useWindowDimensions();
  const cameraDevice = useCameraDevice("front");
  const faceTop = useSharedValue(0);

  const updateReanimated = Worklets.createRunInJsFn(
    (bounds: Pick<FaceBounds, "top">) => {
      faceTop.value = bounds.top;
      console.log("detection result", bounds);
    }
  );

  const frameProcessor = useFrameProcessor((frame) => {
    "worklet";
    console.log(1);
    runAsync(frame, () => {
      "worklet";
      detectFaces(
        frame,
        (result: DetectionResult) => {
          "worklet";
          console.log(2);
          const faces = Object.values(result.faces);
          if (faces.length) {
            updateReanimated({
              top: faces[0].bounds.top * (dimensions.height / frame.height), // remove the multiplication and the frame processor keeps running
            });
          } else {
            updateReanimated({ top: 0 });
          }
        },
        {
          // contourMode: "all",
        }
      );
    });
  }, []);

  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  }, [hasPermission]);

  return (
    <View style={styles.container}>
      {cameraDevice && (
        <Camera
          style={{ width: "100%", height: "100%" }}
          frameProcessor={frameProcessor}
          device={cameraDevice}
          isActive={true}
          pixelFormat="yuv"
        />
      )}
    </View>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: "#000",
    alignItems: "center",
    justifyContent: "center",
  },
});
```

### Relevant log output

```shell
N/A (no warnings or errors reported, both natively and through JS)
```

### Camera Device

```json
{
  "isMultiCam": false,
  "hasTorch": false,
  "sensorOrientation": "landscape-right",
  "position": "front",
  "supportsRawCapture": false,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "supportsFocus": true,
  "formats": [],
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "maxZoom": 128.875,
  "minFocusDistance": 20,
  "hasFlash": true,
  "name": "Front Camera",
  "minExposure": -8,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "hardwareLevel": "full"
}
```

### Device

iPhone 15 Pro (iOS 17.2)

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mfkrause, 2024-02-17:
This only seems to happen when using `runAsync`. If I use `runAtTargetFps` (tested with 15 FPS for now), it works seamlessly.

nonam4, 2024-02-19:
Hey, you're using my face detection package, glad it's helping you 😁 

I had the same problem and as a workaround I added frame data inside detection result data, so instead of using `frame.height` try using `result.frame.height`.

For me this works.

The most strange thing is that on androi it works, you can use frame data directly without any issue but on IOS it freezes after some frames

bglgwyng, 2024-02-23:
I make a minimal reproducer.
```ts
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'

  runAsync(frame, () => {
    'worklet'
    foo(frame)

    console.info('run frame processor')
  })
}, [])
```

And the definition of `foo` should be

```ts
const foo = (frame: Frame) => {
  'worklet'
  frame.height
}
```

When `foo` doesn't include any access to `frame`'s property(`height` here), the frame processor doesn't stop.

frodriguez-hu, 2024-04-09:
Check this, get the result and operate on the current processor and not on the callback

https://github.com/nonam4/react-native-vision-camera-face-detector/issues/28

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

mrousavy, 2024-04-22:
more accurately; react-native-worklets-core 1.x.x fixed this issue I think.

bglgwyng, 2024-04-27:
I think we need to reopen this issue.
I still observe the same behavior with react-native-vision-camera@4.0.1 and react-native-worklets-core@1.2.0.

mrousavy, 2024-04-29:
Ah shit. Well can we create a new issue with logs from latest react-native-vision-camera and rnwc then?

nonam4, 2024-04-29:
@mrousavy I created a new issue on worklets-core repo with all logs from latest versions

mrousavy, 2024-04-29:
Great, thanks

nonam4, 2024-04-29:
@mrousavy I messed up with issues I'm subscribed 😅 this don't have anything to do with worklets-core. Sorry.

But anyway, what solved this "freeze" issue for me was incrementing frame ref count before doing anything else with frame. Ex: 
```
const faces = detectFaces( frame )
const internal = frame as FrameInternal
// incremend frame ref before doing any task with frame
internal.incrementRefCount()
runOnJs(
  faces,
  frame
).finally( () => {
  'worklet'
  // decrement frame ref after all tasks
  internal.decrementRefCount()
} )
```

@bglgwyng Hope this helps

bglgwyng, 2024-05-02:
Thanks! But isn't it a workaround? `runAsync` internally increases/decreases the ref count. Should we perform this redundant operation?

mrousavy, 2024-05-02:
I fixed this issue in my latest PR to main.
Ref counting now works properly on iOS as well