margelo/react-native-vision-camera #2582: ✨ isRecording() prop for cameraRef

ChristopherGabba, 2024-02-18:
### What feature or enhancement are you suggesting?

Right now, you can use:

-` cameraRef.current?.startRecording()`
-` cameraRef.current?.stopRecording()`
-` cameraRef.current?.pauseRecording()`
-` cameraRef.current?.resumeRecording()`

If you call stopRecording() twice, you get a console.warn that you called it twice. 

Would be awesome to be able to control this logic within code like so:

```typescript
if(cameraRef.current?.isRecording()) { // could also be cameraRef.current?.isRecordingAsync() possibly
    cameraRef.current?.stopRecording() // prevent the warning and also get the status of the camera.
}
```

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Right now, you just set a separate state variable when you call:
```typescript
    cameraRef.current.startRecording({
        flash,
        fileType: "mp4",
        onRecordingError: (error) => {
          console.error("Recording failed!", error)
        },
        onRecordingFinished: (video) => {
          console.log(`Recording successfully finished! ${video.path}`)
          onMediaCaptured(video, "video")
          setCameraRecording(false) // track current state of camera in separate variable
        },
    })
    setCameraRecording(true) // track current state of camera in separate variable
```

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

ChristopherGabba, 2024-02-19:
@mrousavy that would work great too! Thanks again

mrousavy, 2024-02-19:
Yea but I'm not gonna work on this for a while, maybe some point in the future. If you need this we can of course build this feature for you through my consultancy company, otherwise I'll maybe take a look at this when VisionCamera is a TurboModule :)

mrousavy, 2026-04-21:
..via exactly the API I talked about 2 years ago - manifested it