margelo/react-native-vision-camera #2583: 🐛 Preview looks stretched on Android

grevtsovna, 2024-02-19:
### What's happening?

The preview looks stretched on Android.  
[image]

I figured out that it becomes ok when I change the height of the container in which the Camera renders. Here are a screenshot and logs after changing the height in debug mode.
[image]

```
2024-02-19 09:35:57.796  9065-9100  Parcel                  com.rncameratest                     W  Expecting binder but got null!
2024-02-19 09:35:57.922  9065-9065  PreviewView             com.rncameratest                     I  PreviewView is 1426x1740, rendering 960x1280 content (LANDSCAPE_LEFT). Resizing to: 1305x1740 (CONTAIN)
2024-02-19 09:35:57.967  9065-9100  EGL_emulation           com.rncameratest                     D  app_time_stats: avg=18224.98ms min=1.65ms max=90444.32ms count=5
```

It reproduces in the example app as well. To reproduce that, I needed to tap the QR-icon. Here is a screenshot.
[image]

Investigating the problem I found out that `onMeasure` method doesn't call after changing the surface size (you can see this in logs). Maybe this is a problem but I'm not sure as I don't have great experience in android development.

### Reproduceable Code

```tsx
function App() {
  const isDarkMode = useColorScheme() === 'dark';

  const backgroundStyle = {
    backgroundColor: isDarkMode ? Colors.darker : Colors.lighter,
  };

  const device = useCameraDevice('back');
  const [isMounted, setIsMounted] = React.useState(true);
  const { hasPermission, requestPermission } = useCameraPermission();

  if (!hasPermission) {
    requestPermission();
  }

  if (!device || !hasPermission) {
    return null;
  }

  return (
    <SafeAreaView style={backgroundStyle}>
      <Button
        title={isMounted ? 'Unmount' : 'Mount'}
        onPress={() => setIsMounted(!isMounted)}
      />
      <View style={{ height: 500, borderWidth: 2, borderColor: 'blue' }}>
        {isMounted && (
          <Camera
            device={device}
            isActive
            style={StyleSheet.absoluteFill}
            resizeMode="contain"
          />
        )}
      </View>
    </SafeAreaView>
  );
}
```

### Relevant log output

```shell
---------------------------- PROCESS STARTED (9065) for package com.rncameratest ----------------------------
2024-02-19 09:34:27.006  9065-9118  CameraManagerGlobal     com.rncameratest                     I  Connecting to camera service
2024-02-19 09:34:27.014  9065-9065  unknown:ReactNative     com.rncameratest                     W  Packager connection already open, nooping.
2024-02-19 09:34:27.021  9065-9123  unknown:Vi...rtyUpdater com.rncameratest                     W  Could not find generated setter for class com.facebook.react.views.traceupdateoverlay.TraceUpdateOverlayManager
2024-02-19 09:34:27.023  9065-9123  unknown:Vi...rtyUpdater com.rncameratest                     W  Could not find generated setter for class com.facebook.react.uimanager.LayoutShadowNode
[... 164 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Android Emulator (Pixel 7 Pro API 33), OnePlus 10R (CPH2411 Android 13))

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Georg7, 2024-02-19:
Same here. I have a stretched and zoomed preview on Android that makes it unusable. 

```
"react-native-vision-camera": "3.9.0-beta.2",
"expo": "~50.0.6",
``` 

```
const device = useCameraDevice("back");

const format = useCameraFormat(device, [{ photoResolution: "max" }]);

const focus = useCallback((point) => {
	const c = cameraRef.current;
	if (c == null) return;
	c.focus(point);
}, []);

const gesture = Gesture.Tap().onEnd(({ x, y }) => {
	runOnJS(setFocusPoint)({ x, y });
	runOnJS(focus)({ x, y });
});

const { props: cameraProps } = useBarcodeScanner({
	barcodeTypes: ["ean-13", "ean-8"],
	onBarcodeScanned: (barcodes) => {
		"worklet";
		updateBarcodesStateWorklet(barcodes);
	},
});

<GestureDetector gesture={gesture}>
	<Camera
		ref={cameraRef}
		photo={true}
		style={StyleSheet.absoluteFill}
		device={device}
		isActive={cameraIsActive}
		format={format}
		fps={30}
		{... cameraProps }
	/>
</GestureDetector>
```

On Android:
[image]

On iOS
[image]

boiboif, 2024-02-19:
My situation is that there is no stretching in the simulator, but it will stretch in the actual machine (landscape mode).

PelayoGooApps, 2024-02-19:
I'm experiencing the same issue. I upgraded to version 3.9.0 hoping to address this problem: https://github.com/mrousavy/react-native-vision-camera/pull/2519

Previously, the stretching was vertical, but now it is horizontal.

Pelasua, 2024-02-19:
I'm experiencing the same issue. I upgraded to version 3.9.0 hoping to address this problem: https://github.com/mrousavy/react-native-vision-camera/pull/2519

Previously, the stretching was vertical, but now it is horizontal.

mrousavy, 2024-02-19:
1. This is weird, on all my test devices (Samsung, Huawei, Google Pixel) the preview looks good. I can take another look but if the issue cannot be reproduced on my devices then I don't know what to do. If anyone of you guys wants to play around with the code in `PreviewView.kt` that would of course be a huge help.
2. As mentioned in the first pinned issue here in this repo, orientation is not yet supported. It's quite a large effort to implement it, so I'm crowdfunding the effort. https://github.com/mrousavy/react-native-vision-camera/issues/1891

Pelasua, 2024-02-19:
> 1. This is weird, on all my test devices (Samsung, Huawei, Google Pixel) the preview looks good. I can take another look but if the issue cannot be reproduced on my devices then I don't know what to do. If anyone of you guys wants to play around with the code in `PreviewView.kt` that would of course be a huge help.
> 2. As mentioned in the first pinned issue here in this repo, orientation is not yet supported. It's quite a large effort to implement it, so I'm crowdfunding the effort. [✨ Implement Orientation ($8,000) #1891](https://github.com/mrousavy/react-native-vision-camera/issues/1891)
Sorry, I would like to help more with Kotlin, but I don't have any idea.

I've tried changing the resizeMode prop from 'cover' to 'contain,' and this is the result (picture). I don't know why I have these measurements (the container has 600 height x 300 width). Are these the default preview measurements? Anyway, the preview still shows a stretched capture.

[image]

mrousavy, 2024-02-19:
@bruno-centanaro can you submit a PR? I can take a look and others can test this.

mrousavy, 2024-02-19:
> One thing I do not understand is why the `CameraDeviceDetails.getMaximumPreviewSize()` is used for the aspect ratio in `getSize`.

This is just the default value. It will be updated later by the `surfaceChanged` event where the actual `Surface` size gets assigned to `PreviewView.size`, causing `onMeasure` to re-calculate the aspect ratio (at least that's the plan).

bruno-centanaro, 2024-02-19:
Sorry, I messed up, as for my first message, it was containerAspectRatio instead of contentAspectRatio and not vice-versa
As for the surfaceChanged, you are correct it modifies the size variable but this is not causing the onMeasure to run again, at least on my phone.
As for the PR I can try to do it after work today, I'll link it here

mrousavy, 2024-02-19:
> As for the surfaceChanged, you are correct it modifies the size variable but this is not causing the onMeasure to run again, at least on my phone.

Interesting - I think this PR fixes that: https://github.com/mrousavy/react-native-vision-camera/pull/2588

mrousavy, 2024-02-19:
I _believe_ that the aspect ratio calculation is correct, are you sure that swapping the aspect ratios is right? Can you just quickly post your diff or what exactly you changed so I can take a look now?

Otherwise a PR after work is also much appreciated. Before and after screenshots are all I need

bruno-centanaro, 2024-02-19:
That PR does cause the onMeasure to re-run, but it is strange because the preview is still stretched. This are the logs

> 2024-02-19 10:19:08.709 26373-26373 PreviewView             BUNDLE_ID              I  Creating PreviewView...
2024-02-19 10:19:08.710 26373-26373 PreviewView             BUNDLE_ID              I  PreviewView is 0x0, rendering 1080x1920 content (LANDSCAPE_LEFT). Resizing to: 0x0 (COVER)
2024-02-19 10:19:08.878 26373-26373 SurfaceView@8057685     BUNDLE_ID              I  onWindowVisibilityChanged(0) true com.mrousavy.camera.core.PreviewView{8057685 V.E...... ......I. 0,0-0,0} of ViewRootImpl@a077172[MainActivity]
2024-02-19 10:19:08.883 26373-26373 PreviewView             BUNDLE_ID              I  PreviewView is 1080x1440, rendering 1080x1920 content (LANDSCAPE_LEFT). Resizing to: 1080x1920 (COVER)
2024-02-19 10:19:08.931 26373-26373 SurfaceView@8057685     BUNDLE_ID              I  surfaceCreated 2 #8 com.mrousavy.camera.core.PreviewView{8057685 V.E...... ......ID 0,-240-1080,1680}
2024-02-19 10:19:08.931 26373-26373 CameraSession           BUNDLE_ID             I  PreviewView Surface created! Surface(name=null)/@0xf56b2aa
2024-02-19 10:19:08.932 26373-26373 SurfaceView@8057685     BUNDLE_ID              I  surfaceChanged (1920,1080) 2 #8 com.mrousavy.camera.core.PreviewView{8057685 V.E...... ......ID 0,-240-1080,1680}
2024-02-19 10:19:08.932 26373-26373 PreviewView             BUNDLE_ID             I  Surface changed: 1920 x 1080
2024-02-19 10:19:08.932 26373-26373 CameraSession           BUNDLE_ID              I  PreviewView Surface updated! Surface(name=null)/@0xf56b2aa 1920 x 1080
2024-02-19 10:19:09.101 26373-26373 PreviewView             BUNDLE_ID              I  Input Orientation changed: LANDSCAPE_LEFT -> LANDSCAPE_RIGHT
2024-02-19 10:19:09.120 26373-26373 PreviewView             BUNDLE_ID              I  PreviewView is 1080x1920, rendering 1080x1920 content (LANDSCAPE_RIGHT). Resizing to: 1080x1920 (COVER)
2024-02-19 10:19:09.311 26373-26373 PreviewView             BUNDLE_ID              I  PreviewView is 1080x1920, rendering 1080x1920 content (LANDSCAPE_RIGHT). Resizing to: 1080x1920 (COVER)
2024-02-19 10:19:09.311 26373-26373 BLASTBufferQueue_Java   BUNDLE_ID              I  update, w= 640 h= 480 mName = null mNativeObject= 0xb400007d19803210 sc.mNativeObject= 0xb400007c4962f9d0 format= 4 caller= android.view.SurfaceView.setBufferSize:1432 android.view.SurfaceView.performSurfaceTransaction:988 android.view.SurfaceView.updateSurface:1204 android.view.SurfaceView.setFrame:559 android.view.View.layout:25731 com.mrousavy.camera.core.PreviewView.requestLayout$lambda$1:100 
2024-02-19 10:19:09.313 26373-26373 SurfaceView@8057685     BUNDLE_ID              I  surfaceChanged (640,480) 3 #5 com.mrousavy.camera.core.PreviewView{8057685 V.E...... ......I. 0,-240-1080,1680}
2024-02-19 10:19:09.313 26373-26373 PreviewView             BUNDLE_ID             I  Surface changed: 640 x 480
2024-02-19 10:19:09.313 26373-26373 PreviewView            BUNDLE_ID              I  Surface Size changed: 1920x1080 -> 640x480
2024-02-19 10:19:09.313 26373-26373 CameraSession           BUNDLE_ID              I  PreviewView Surface updated! Surface(name=null)/@0xf56b2aa 640 x 480
2024-02-19 10:19:09.324 26373-26373 PreviewView            BUNDLE_ID              I  PreviewView is 1080x1920, rendering 480x640 content (LANDSCAPE_RIGHT). Resizing to: 1440x1920 (COVER)

I'm not sure if swapping the aspect ratio's is the correct way to do it, it works on my phone. This is the code

` 
private fun getSize(contentSize: Size, containerSize: Size, resizeMode: ResizeMode): Size {
    val contentAspectRatio = contentSize.width.toDouble() / contentSize.height
    val containerAspectRatio = containerSize.width.toDouble() / containerSize.height

    val widthOverHeight = when (resizeMode) {
      ResizeMode.COVER -> contentAspectRatio > containerAspectRatio
      ResizeMode.CONTAIN -> contentAspectRatio < containerAspectRatio
    }

    Log.i(TAG, "Content Aspect Ratio: $contentAspectRatio, Container Aspect Ratio: $containerAspectRatio widthOverHeight: $widthOverHeight ($resizeMode)")

    return if (widthOverHeight) {
      // Scale by width to cover height
      val scaledWidth = containerSize.height * containerAspectRatio
      Size(scaledWidth.roundToInt(), containerSize.height)
    } else {
      val scaledHeight = if (containerAspectRatio.isNaN()) containerSize.width / contentAspectRatio else containerSize.width / containerAspectRatio
      Log.i(TAG, "Scaled Height: $scaledHeight, containerSize: $containerSize, contentAspectRatio: $contentAspectRatio")
      Size(containerSize.width, scaledHeight.roundToInt())
    }
  }
`

mrousavy, 2024-02-19:
Uhm well this code:

```kt
val scaledWidth = containerSize.height * contentAspectRatio
val scaledWidth = containerSize.height * containerAspectRatio
Size(scaledWidth.roundToInt(), containerSize.height)
```

Effectively does nothing, it's the same as 

```kt
containerSize
```

(1080 / 1920 * 1080 = 1920, so both width and height always stay the same values as before)

And then the second part:

```kt
val scaledHeight = if (containerAspectRatio.isNaN()) containerSize.width / contentAspectRatio else containerSize.width / containerAspectRatio
```

`containerAspectRatio` could be NaN on the first render, true, in that case I think an early return makes more sense though.

bruno-centanaro, 2024-02-19:
Yes, to be honest the widthOverHeight I did not test it as its false for me

bglgwyng, 2024-02-20:
I tried this PR https://github.com/mrousavy/react-native-vision-camera/pull/2588, but the preview is still stretched.

swamywiz, 2024-02-20:
For me, we were using two aspect ratio, 4:3 and 16:9, 16:9 doesn't show any stretching issue, but I do notice issue in case of 4:3. Tested on oneplus Nord.

robertyulisman, 2024-02-20:
[image]

there is update for this, i am still get this issue, sometime when i reload after save, the preview is normally, when go back to previous page and open camera again, this happen again

robertyulisman, 2024-02-20:
[image]
this is normal preview

seavan, 2024-02-20:
@bruno-centanaro @mrousavy 

I have the same issue with landscape stretching, and this is what helped me (since 3.7.1). The issue still persists on 3.9.0.

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt b/node_modules/react-      native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt                                                            index ba3c94b..f2b140b 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
@@ -105,7 +105,7 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
       Size(scaledWidth.roundToInt(), containerSize.height)
     } else {
       // Scale by height to cover width
-      val scaledHeight = containerSize.width / contentAspectRatio
+      val scaledHeight = containerSize.width * contentAspectRatio
       Size(containerSize.width, scaledHeight.roundToInt())
     }
   }
```

I think division operator here is not correct, it should be multiplication.

bruno-centanaro, 2024-02-20:
The operators there are correct I think. For anyone who has issues and is sure the aspect ratio of the camera container is the same as the resolution's one this works for me

```diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
index ba3c94b..cc19356 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
@@ -94,6 +94,10 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
     val contentAspectRatio = contentSize.width.toDouble() / contentSize.height
     val containerAspectRatio = containerSize.width.toDouble() / containerSize.height
 
+    if (!(contentAspectRatio > 0 && containerAspectRatio > 0)) {
+      return contentSize
+    }
+
     val widthOverHeight = when (resizeMode) {
       ResizeMode.COVER -> contentAspectRatio > containerAspectRatio
       ResizeMode.CONTAIN -> contentAspectRatio < containerAspectRatio
@@ -101,11 +105,11 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
 
     return if (widthOverHeight) {
       // Scale by width to cover height
-      val scaledWidth = containerSize.height * contentAspectRatio
+      val scaledWidth = containerSize.height * containerAspectRatio
       Size(scaledWidth.roundToInt(), containerSize.height)
     } else {
       // Scale by height to cover width
-      val scaledHeight = containerSize.width / contentAspectRatio
+      val scaledHeight = containerSize.width / containerAspectRatio
       Size(containerSize.width, scaledHeight.roundToInt())
     }
   }
```

Please make sure the aspect ratio's are the same otherwise I don't think that will work

swamywiz, 2024-02-20:
> The operators there are correct I think. For anyone who has issues and is sure the aspect ratio of the camera container is the same as the resolution's one this works for me
> 
> ```diff
> index ba3c94b..cc19356 100644
> --- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
> +++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
> @@ -94,6 +94,10 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
>      val contentAspectRatio = contentSize.width.toDouble() / contentSize.height
>      val containerAspectRatio = containerSize.width.toDouble() / containerSize.height
>  
> +    if (!(contentAspectRatio > 0 && containerAspectRatio > 0)) {
> +      return contentSize
> +    }
> +
>      val widthOverHeight = when (resizeMode) {
>        ResizeMode.COVER -> contentAspectRatio > containerAspectRatio
>        ResizeMode.CONTAIN -> contentAspectRatio < containerAspectRatio
> @@ -101,11 +105,11 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
>  
>      return if (widthOverHeight) {
>        // Scale by width to cover height
> -      val scaledWidth = containerSize.height * contentAspectRatio
> +      val scaledWidth = containerSize.height * containerAspectRatio
>        Size(scaledWidth.roundToInt(), containerSize.height)
>      } else {
>        // Scale by height to cover width
> -      val scaledHeight = containerSize.width / contentAspectRatio
> +      val scaledHeight = containerSize.width / containerAspectRatio
>        Size(containerSize.width, scaledHeight.roundToInt())
>      }
>    }
> ```
> 
> Please make sure the aspect ratio's are the same otherwise I don't think that will work

Tested this patch, this resolves the issue we were having with 4:3 aspect ratio.

Georg7, 2024-02-20:
> The operators there are correct I think. For anyone who has issues and is sure the aspect ratio of the camera container is the same as the resolution's one this works for me
>
> ```diff
> index [...]

Tried this on a Samsung Galaxy (A04e) but the preview remains stretched and zoomed. 

Samsung Galaxy (A04e)
[image]

iPhone 14 Pro
[image]

bruno-centanaro, 2024-02-21:
It seems as your container aspect ratio might not be the same as the video preview (set on the format prop)

Georg7, 2024-02-21:
I believe it is the same. Here is how I set it up: 

```
const device = useCameraDevice("back");

const screen = Dimensions.get("screen");

const format = useCameraFormat(device, [
	{ targetPhotoAspectRatio: screen.height / screen.width },
	{ autoFocusSystem: "phase-detection" },
	{ photoResolution: "max" },
	{ fps: 30 },
]);
```

```
<Camera
	ref={cameraRef}
	photo={true}
	style={StyleSheet.absoluteFill}
	device={device}
	isActive={cameraIsActive}
	format={format}
/>
```

bruno-centanaro, 2024-02-21:
The aspect ratio of the preview is the same as video, not the picture one, also the fact that you use an aspect ratio in the format does not necessarily mean it is supported by the camera, maybe try console logging the format to check

vanhoutk, 2024-02-21:
> ```diff
> containerAspectRatio
> ```

This seems to have sorted the issue for me, was occurring on a Fairphone 4 when my camera preview was not the full size of my screen

Georg7, 2024-02-22:
> The aspect ratio of the preview is the same as video, not the picture one, also the fact that you use an aspect ratio in the format does not necessarily mean it is supported by the camera, maybe try console logging the format to check

Here is what I've tried: 
```
const screen = Dimensions.get("screen");
const device = useCameraDevice("back");

const format = useCameraFormat(device, [
	{ photoWidth: screen.width },
	{ photoHeight: screen.height },
	{ videoHeight: screen.height },
	{ videoWidth: screen.width },
	{ autoFocusSystem: "phase-detection" },
	{ photoResolution: "max" },
	{ fps: 30 },
]);
```
Is this the correct/ideal way to set the format wrt to width and height?

Logging the format to the console : 
```
{
	autoFocusSystem: "contrast-detection",
	fieldOfView: 81.19526233755766,
	maxFps: 30,
	maxISO: 1600,
	maxZoom: 10,
	minFps: 1,
	minISO: 100,
	photoHeight: 3120,
	photoWidth: 4160,
	pixelFormats: ["yuv", "native"],
	supportsDepthCapture: false,
	supportsPhotoHdr: false,
	supportsVideoHdr: false,
	videoHeight: 1440,
	videoStabilizationModes: ["off", "standard", "off"],
	videoWidth: 1920,
}
```
No idea why the photo and video dimensions are not the same. 

When I manually reset the width or height on the camera once it's mounted, the preview renders correctly:
```
<Camera
	ref={cameraRef}
	photo={true}
	style={StyleSheet.absoluteFill} // <-- commenting this out to manually reset
	// style={{ width: screen.width, height: screen.height }} <-- uncommenting this to manually reset
	device={device}
	isActive={cameraIsActive}
	format={format}
/>
```

But it never renders correctly on the initial mount. Tried it with @swamywiz's batch and without (same behaviour).

stevenmathers, 2024-02-22:
Hi @mrousavy 

I have an issue where the preview image is slightly more zoomed in than the captured image, in one axis only.
This seems consistent across versions that I have tried  from 3.6.4  ->  3.8.2

The mode is set to 'contain' and I have tried various styles such as absoluteFill, and an absolute top, left, width, height to ensure there is a large margin around the preview area to ensure the view is not running off the edge of the screen somehow

This is consistent over a number of devices, but not all.

Is there a trick I am missing?

stevenmathers, 2024-02-22:
@jhau are you sure its both axis?  thats weird if so

stevenmathers, 2024-02-23:
update, 
this appears to be android only, and I really think its only in one axis - horizontal, regardless of what orientation you are holding the device.  And it does not affect the android emulator, presuambly some implemntation issue of the emulator as opposed to a physical camera.
the 1st image of the pair shows the preview window, and the second is the actual image returned from the camera.
NOTE: as I said, the image is not being drawn off the screen.  even if I reduce the window size so that it is less than the screen dimenstions, the same thing occurs
[image]
[image]
[image]
[image]

Pelasua, 2024-02-26:
Well, after conducting multiple tests, this is the trick I propose:

Play with re-renders. I mean, if you set a negligible size initially for the camera component, and after 1 second, you update it to the full size of the container, it seems that the re-render corrects the issue.

```
 const [sizeWithDelay, setSizeWithDelay] = useState<ViewStyle>({
    width: 1,
    height: 1,
  });

  useLayoutEffect(() => {
    !photo &&
      setTimeout(() => {
        setSizeWithDelay({
          width: '100%',
          height: '100%',
        });
      }, 1000);

    return () => {
      photo &&
        setSizeWithDelay({
          width: 1,
          height: 1,
        });
    };
  }, [photo]);
```

And in the component:
```
<Camera
            ref={camera}
            photo={true}
            style={[
              styles.camera,
              { height: sizeWithDelay.height, width: sizeWithDelay.width },
            ]}
            .../>
```

Note that if you leave the sizes at 0, it doesn't work well, which is why I had to set it to 1.

What I do is, if there's no photo, I use the preview and give it a 100% width and height, but it reverts to 1 if we already have the photo.

Depending on the delay set by the developer, it might be better to include some type of activity indicator to enhance the user experience or not.

I'm sure there's a more correct way to implement it, and I'm open to improvements until the error is fixed. I hope it also helps identify why the problem might be occurring.

Samsung Galaxy J6+:
[image]
[image]

mlecoq, 2024-02-26:
the sizeWithDelay helps me for stretch issue on redmi but I had also to modify the native code to have a correct preview (on samsung device)

see patch below

```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt
index a589c26..91f5b8c 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt
@@ -47,6 +47,13 @@ class CameraDeviceDetails(private val cameraManager: CameraManager, val cameraId
 
       return if (isHighResScreen) display1080p else displaySize
     }
+
+    fun getActualPreviewSize(): Size {
+      return Size(
+        Resources.getSystem().displayMetrics.widthPixels,
+        Resources.getSystem().displayMetrics.heightPixels
+      )
+    }
   }
 
   val characteristics by lazy { cameraManager.getCameraCharacteristics(cameraId) }
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
index ba3c94b..c2b7335 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt
@@ -20,7 +20,7 @@ import kotlinx.coroutines.withContext
 class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
   SurfaceView(context),
   SurfaceHolder.Callback {
-  var size: Size = CameraDeviceDetails.getMaximumPreviewSize()
+  var size: Size = CameraDeviceDetails.getActualPreviewSize()
     set(value) {
       if (field != value) {
         Log.i(TAG, "Surface Size changed: $field -> $value")
@@ -92,11 +92,11 @@ class PreviewView(context: Context, callback: SurfaceHolder.Callback) :
 
   private fun getSize(contentSize: Size, containerSize: Size, resizeMode: ResizeMode): Size {
     val contentAspectRatio = contentSize.width.toDouble() / contentSize.height
-    val containerAspectRatio = containerSize.width.toDouble() / containerSize.height
+    val containerAspectRatio = if(containerSize.height > 0) containerSize.width.toDouble() / containerSize.height else 0
 
     val widthOverHeight = when (resizeMode) {
-      ResizeMode.COVER -> contentAspectRatio > containerAspectRatio
-      ResizeMode.CONTAIN -> contentAspectRatio < containerAspectRatio
+      ResizeMode.COVER -> contentAspectRatio > containerAspectRatio.toDouble()
+      ResizeMode.CONTAIN -> contentAspectRatio < containerAspectRatio.toDouble()
     }
 
     return if (widthOverHeight) {

```

stefoid, 2024-02-28:
> the sizeWithDelay helps me for stretch issue on redmi but I had also to modify the native code to have a correct preview (on samsung device)
>
> see patch below
>
> ```
> diff --git [...]

I am using 3.8.2 because 3.9 seems buggy in different ways (timing out trying to take photos on android device)
I refactored this patch for 3.8.2  and it doesnt work, but some weird new behaviour occurs ->  for a moment, the preview looks like it is the proper width (in terms of field of view) but possibly it is stretched (wrong aspect ratio?), then it snaps to the same aspect ratio/field of view which is the same as my original issue -> the field of view of the preview is less horizontally than the captured image.  any thoughts?

mlecoq, 2024-02-28:
You're right, I thought it was fixing my issue (did you set photoAspectRatio in useFormat hook too) but it was worse in fact

xHeinrich, 2024-02-28:
Getting a similar issue to whats being described above on a moto e22i. I tried the patch in  https://github.com/mrousavy/react-native-vision-camera/pull/2588 which didnt resolve the issue. The preview is stretched but only in landscape, portrait seems fine. Replicatable on the latest ShadowLense app version. Actual image capture is fine. In 3.8.2 this isn't an issue.

[image]
[image]

AfanasievN, 2024-03-01:
Same issue persists

frenbergFNX, 2024-03-05:
```
16:52:55.788  I  Creating PreviewView...
16:52:55.788  I  PreviewView is 0x0, rendering 1415x720 content (LANDSCAPE_LEFT). Resizing to: 1415x720 (COVER)
16:52:55.790  I  Updating CameraSession...
16:52:55.791  I  configure { ... }: Waiting for lock...
16:52:55.792  I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
16:52:55.792  I  Configuring inputs for CameraSession...
16:52:55.792  D  --> setInput(0)
16:52:55.792  I  Destroying previous outputs...
16:52:55.793  I  Creating outputs for Camera #0...
16:52:55.797  I  Adding 4032x1908 Photo Output in JPEG...
16:52:55.799  I  Adding 1920x912 Video Output in YUV_420_888...
16:52:55.799  I  Initializing 1920 x 912 Video Pipeline (format: YUV)
16:52:55.799  I  Using ImageReader round-trip (format: #35)
16:52:55.799  I  Creating ImageReader with default usage flags...
16:52:55.799  I  Creating ImageWriter with format #35...
16:52:55.800  D  --> setOutputs([PHOTO (4032x1908 in JPEG), VIDEO (1920x912 in YUV)])
16:52:55.800  I  Successfully configured Session with 2 outputs for Camera #0!
16:52:55.800  I  Updating Video Outputs...
16:52:55.800  I  Removing RecordingSession Output...
16:52:55.800  D  --> setRepeatingRequest(...)
16:52:55.800  D  --> setIsActive(false)
16:52:55.800  D  Configure() with isActive: false, ID: 0, device: null, session: null
16:52:55.800  I  Creating new device...
16:52:55.800  I  Camera #0: Opening...
16:52:55.812  I  postSingleUpdate device: camera id 0 status STATUS_NOT_AVAILABLE
16:52:55.813  I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_OPEN for client se.fnx.powerup.acce API Level 2
16:52:55.813  I  Camera #0 is now unavailable.
16:52:55.826  I  Camera #0: Opened!
16:52:55.826  I  Updating CameraSession...
16:52:55.826  I  Creating new session...
16:52:55.827  I  onWindowVisibilityChanged(0) true com.mrousavy.camera.core.PreviewView{f168b99 V.E...... ......ID -707,-360-708,360} of ViewRootImpl@543e7a3[MainActivity]
16:52:55.827  I  Camera #0: Creating Capture Session #35... (Hardware Level: 1 | Outputs: [PHOTO (4032x1908 in JPEG), VIDEO (1920x912 in YUV)])
16:52:55.828  I  Using new API (>=28)
16:52:55.833  I  PreviewView is 720x1490, rendering 1415x720 content (LANDSCAPE_LEFT). Resizing to: 2928x1490 (COVER)
16:52:55.844  I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=8 res=0x1 s={true 479177400320} ch=false
16:52:55.855  I  surfaceCreated 2 #8 com.mrousavy.camera.core.PreviewView{f168b99 V.E...... ......ID -1104,0-1824,1490}
16:52:55.855  I  PreviewView Surface created! Surface(name=null)/@0x8643be
16:52:55.855  I  Setting Preview Output...
16:52:55.855  I  surfaceChanged (720,1415) 2 #8 com.mrousavy.camera.core.PreviewView{f168b99 V.E...... ......ID -1104,0-1824,1490}
16:52:55.855  I  PreviewView Surface updated! Surface(name=null)/@0x8643be 720 x 1415
16:52:55.860  I  doUpdatePositionAsync is called and callVoidMethod
16:52:55.892  I  doUpdatePositionAsync is called and callVoidMethod
16:52:55.924  D  Sending lifecycle 2 to service
16:52:55.925  D  Activity backgrounding at 12606912
16:52:55.930  D  unregisterListener ::   
16:52:55.937  W  A resource failed to call release. 
16:52:55.938  I  configure { ... }: Waiting for lock...
16:52:55.938  I  Camera #0: Successfully created CameraCaptureSession #35!
16:52:55.938  I  configure { ... }: Waiting for lock...
16:52:55.939  D  Stopping repeating request...
16:52:55.939  D  Configure() done! isActive: false, ID: 0, device: android.hardware.camera2.impl.CameraDeviceImpl@5851b0f, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@485479c
16:52:55.939  I  configure { ... }: Completed CameraSession Configuration! (isActive: false, isRunning: false)
16:52:55.939  I  invokeOnInitialized()
16:52:55.939  D  Finding view 7553...
16:52:55.940  D  Found view 7553!
16:52:55.940  I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=false, isActiveChanged=true)
16:52:55.940  D  --> setIsActive(false)
16:52:55.940  D  Configure() with isActive: false, ID: 0, device: android.hardware.camera2.impl.CameraDeviceImpl@5851b0f, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@485479c
16:52:55.942  I  doUpdatePositionAsync is called and callVoidMethod
16:52:55.942  D  Stopping repeating request...
16:52:55.942  D  Configure() done! isActive: false, ID: 0, device: android.hardware.camera2.impl.CameraDeviceImpl@5851b0f, session: android.hardware.camera2.impl.CameraCaptureSessionImpl@485479c
16:52:55.942  I  configure { ... }: Completed CameraSession Configuration! (isActive: true, isRunning: false)
16:52:55.943  I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
16:52:55.943  I  Destroying previous outputs...
16:52:55.943  I  Closing 4032x1908 PHOTO ImageReader..
16:52:55.943  I  Closing 1920x912 Video Pipeline..
16:52:55.947  I  Creating outputs for Camera #0...
16:52:55.951  I  Adding 4032x1908 Photo Output in JPEG...
16:52:55.953  I  Adding 1920x912 Video Output in YUV_420_888...
[... 35 lines cut ...]
16:52:56.061  I  Camera #0: Successfully created CameraCaptureSession #36!
16:52:56.061  I  configure { ... }: Waiting for lock...
16:52:56.061  D  Updating repeating request...
[... cut ...]
16:52:56.172  I  Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=3 res=0x1 s={true 479177400320} ch=false
16:52:56.372  I  Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client se.fnx.powerup.acce API Level 2
16:52:56.391  I  ImageReader::onImageAvailable!
16:52:56.426  I  ImageReader::onImageAvailable!
```

On 3.9.1 still have the issue on some android devices, [tried the hack](https://github.com/mrousavy/react-native-vision-camera/issues/2631#issuecomment-1979036415) with adjusting the style using the onInitialized hook and that is a usable work around.

xHeinrich, 2024-03-06:
Example of the code in react-native-camera, it takes into account the current screen orientation. When the screen is in landscape it flips the divide and multiply calculation. In RN Vision Camera its static no matter the rotation passed via the prop. Flipping the calculation if your camera is in landscape fixes the issue, at least on my test device.

https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L231-L247

https://github.com/mrousavy/react-native-vision-camera/blob/main/package/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt#L116-L124

```
    return if (widthOverHeight) {
      // Scale by width to cover height
      val scaledWidth = containerSize.height / contentAspectRatio
      Size(scaledWidth.roundToInt(), containerSize.height)
    } else {
      // Scale by height to cover width
      val scaledHeight = containerSize.width * contentAspectRatio
      Size(containerSize.width, scaledHeight.roundToInt())
    }

```

vincenzoiacovone, 2024-03-07:
I'm using class component
Tired of trying and trying workarounds, version changes, debugging..
Sure that after the first rendering all works well..
Switch on caveman mode:

added this style in the Camera component:
```
    style={{
        ...StyleSheet.absoluteFill,
        width: Dimensions.get('screen').width,
        height: Dimensions.get('screen').height + this.state.update,
      }}
```

and this in componentDidMount: 
```
    let context = this;
    setTimeout(function () {
      context.setState({update: 1});
    }, 500);
```

nica0012, 2024-03-08:
Hey folks, I have been keeping up with the past few updates but i cant seem to get the preview to be the same as the capture result (works fine on IOS, testing with Samsung A14 device)

My preview is either 1:1 Square or 4:3 so it is dynamic but no matter what the zoom is always slightly zoomed in compared to the result. I am using v 3.9.1 currently to test.

boiboif, 2024-03-11:
I downgraded the version to 2.16.8, which solved all the stretch issues and also supported landscape mode.

lucksp, 2024-03-12:
I am seeing this since updating from `version "3.6.4"`

WesleyMaciel2510, 2024-03-15:
Guys, I found out something insteresting. You can test if you want.

You simply use Dimensions.get instead of 'absoluteFill':
  const {width, height} = Dimensions.get('screen');

<Camera
              //style={StyleSheet.absoluteFill}
              style={{width: width, height: height }}
              device={device}
              isActive={true}
              ref={camera}
              photo={true}
            />
            
Now I reload the screen and the error persist. But when I change some value  for 'width' or 'height'.
For example, I change to style={{width: width, height: height-1 }} and save.
Now it fix the problem:

Before:
[image]
Later:
[image]

Does someone knows the solution?
Also, how can I diff the node modules as said before?

diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt
index a589c26..91f5b8c 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt

WesleyMaciel2510, 2024-03-15:
> 2.16.8

I tried, got the error:
FAILURE: Build failed with an exception.

* Where:
Build file 'D:\Personal Data\Code\Mobile\\node_modules\react-native-vision-camera\android\build.gradle' line: 354

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> Cannot invoke method replace() on null object

WesleyMaciel2510, 2024-03-15:
After hours testing, the version that worked for me is:     "react-native-vision-camera": "3.8",
If you are stuck, please downgrade the version to 3.8 before anything more complex.

Rc85, 2024-03-16:
The only workaround I am able to accomplish right now is store the started state and apply the width and height once it has started.
```
const CameraContainer = () => {
  const { width, height } = useWindowDimensions();
  const device = useCameraDevice('back');
  const [started, setStarted] = useState(fales);

  return <Camera
    isActive
    device={device}
    style={{ width: started ? width : 0, height: started ? height : 0 }}
    onStarted={() => setStarted(true)}
  />
}
```

mrousavy, 2024-03-19:
Hey everybody! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

mlecoq, 2024-03-19:
On my samsung tab I have still some stretching / zoom issues 

It depends on the ratio of the preview:
- with a ratio 1, the preview is stretched with front camera (back camera looks fine)
- with a ratio 0.625, no stretching issue but the taken picture is zoomed compared to the preview

I am forcing aspect ratio for camera format 

```
  const format = useCameraFormat(device, [
    {
      photoAspectRatio: aspectRatio,
      videoAspectRatio: aspectRatio,
    },
  ]);
```

But I have no more rotated pictures with front camera

mlecoq, 2024-03-19:
On my redmi I have no issue with ration 1, only 0.625 returns me a zoomed-in captured image

programminPete, 2024-03-19:
> Hey everybody! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test [...]

Hi @mrousavy,  On the good news side, the android preview stretching that I saw on 3.9.1 looks seems resolved for me on the new beta build.
But on the bad news side, I am using it for barcode scanning capability (code-39.  I have upgraded to the beta version (4.0.0-beta.8). It resolves the stretched view, but seems to not scan at all now for android.  
As a side-note, I tested the both versions on iOS and both 3.9.1 and 4.0.0-beta.8 work as expected for both scanning and the general camera view behavior. 

As 4.0.0 is a major version update, is there anything I may be missing as far as an api change from 3.9.1 that I might need to know?  I'm assuming not, if it's working properly on iOS

I can provide more code, but the basic usage I have is the following: 

`
  const codeScanner = useCodeScanner({
    codeTypes: ['code-39'],
    onCodeScanned: codes => {
    // additional code, doesn't seem to enter
 }
 ...
          <Camera
            ref={camera}
            style={stlyes.cameraPreview}
            device={device}
            codeScanner={codeScanner}
            isActive={true}
            photo={true}
            onError={handleError}
            torch={torchEnabled ? 'on' : 'off'}
          />
          `

maxxcs, 2024-03-20:
> Hey everybody! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test [...]

I was about to test in this version, but i got this error if I use `<Camera orientation="landscape-right" />`.
```
androidx.camera.core.impl.utils.CameraOrientationUtil.surfaceRotationToDegrees
CameraOrientationUtil.java, line 97
java.lang.IllegalArgumentException: Unsupported surface rotation: 270

androidx.camera.core.impl.utils.CameraOrientationUtil.surfaceRotationToDegrees CameraOrientationUtil.java:97
androidx.camera.camera2.internal.Camera2CameraInfoImpl.getSensorRotationDegrees Camera2CameraInfoImpl.java:210
androidx.camera.core.UseCase.getRelativeRotation UseCase.java:483
androidx.camera.core.UseCase.getRelativeRotation UseCase.java:472
androidx.camera.core.ImageCapture.takePictureInternal ImageCapture.java:1302
androidx.camera.core.ImageCapture.takePicture ImageCapture.java:863
androidx.camera.core.ImageCapture.lambda$takePicture$2$androidx-camera-core-ImageCapture ImageCapture.java:860
androidx.camera.core.ImageCapture$$ExternalSyntheticLambda1.run
android.os.Handler.handleCallback Handler.java:942
android.os.Handler.dispatchMessage Handler.java:99
android.os.Looper.loopOnce Looper.java:201
android.os.Looper.loop Looper.java:288
android.app.ActivityThread.main ActivityThread.java:7872
java.lang.reflect.Method.invoke Method.java
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run RuntimeInit.java:548
com.android.internal.os.ZygoteInit.main ZygoteInit.java:936
```

lucksp, 2024-03-20:
Anyone here using Expo with Android SDK (34)?

iagoandrei, 2024-03-24:
> > Hey everybody! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅
> > ```
> > yarn add react-native-vision-camera@beta
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.
> 
> I was about to test it in this version, but i got this error if I use `<Camera orientation="landscape-right" />`.
> 
> ```
> androidx.camera.core.impl.utils.CameraOrientationUtil.surfaceRotationToDegrees
> CameraOrientationUtil.java, line 97
> java.lang.IllegalArgumentException: Unsupported surface rotation: 270
> 
> androidx.camera.core.impl.utils.CameraOrientationUtil.surfaceRotationToDegrees CameraOrientationUtil.java:97
> androidx.camera.camera2.internal.Camera2CameraInfoImpl.getSensorRotationDegrees Camera2CameraInfoImpl.java:210
> androidx.camera.core.UseCase.getRelativeRotation UseCase.java:483
> androidx.camera.core.UseCase.getRelativeRotation UseCase.java:472
> androidx.camera.core.ImageCapture.takePictureInternal ImageCapture.java:1302
> androidx.camera.core.ImageCapture.takePicture ImageCapture.java:863
> androidx.camera.core.ImageCapture.lambda$takePicture$2$androidx-camera-core-ImageCapture ImageCapture.java:860
> androidx.camera.core.ImageCapture$$ExternalSyntheticLambda1.run
> android.os.Handler.handleCallback Handler.java:942
> android.os.Handler.dispatchMessage Handler.java:99
> android.os.Looper.loopOnce Looper.java:201
> android.os.Looper.loop Looper.java:288
> android.app.ActivityThread.main ActivityThread.java:7872
> java.lang.reflect.Method.invoke Method.java
> com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run RuntimeInit.java:548
> com.android.internal.os.ZygoteInit.main ZygoteInit.java:936
> ```

Same here

rodgomesc, 2024-03-25:
@iagoandrei  @maxxcs sent a fix for this can you guys check if it works for you?

https://github.com/mrousavy/react-native-vision-camera/pull/2683

one thing i'm really confuse is that this thread is about a "Preview stretched", the issue you guys reported is only reproduced  on my side when i take a picture and orientation is other than the default value.

you should be able to validate the ***stretching thing*** without take a picture or any other action, what i'm missing here? is the preview ok for you?

mrousavy, 2024-03-25:
Hey @programminPete - I just fixed that code scanning bug in latest V4 beta! :)

Anyways - great to hear that V4 now works perfectly! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

closed 2024-03-25 by mrousavy

stefoid, 2024-03-25:
Sorry, late to this - Im completely buried in other issues right now.

Can I ask are you aware of / have fixed, the issue that I brought up in this thread above, where the field of view of the Preview screen is more narrow than the field of view of the resulting image?  Its not stretching of the preview image, but more like clipping 10% of the width of either side

mlecoq, 2024-03-26:
@stefoid I have created a separated issue: https://github.com/mrousavy/react-native-vision-camera/issues/2690

Let me know if you are talking about the same issue

programminPete, 2024-03-26:
Morning @mrousavy .  Seems to be working now, thanks!
I added sponsor payment :)

mrousavy, 2024-03-26:
Thanks @programminPete! much appreciated! :)

n-ii-ma, 2024-04-15:
> I'm using class component Tired of trying and trying workarounds, version changes, debugging.. Sure that after the first rendering all works well.. Switch on caveman mode:
> 
> added this style in the Camera component:
> 
> ```
>     style={{
>         ...StyleSheet.absoluteFill,
>         width: Dimensions.get('screen').width,
>         height: Dimensions.get('screen').height + this.state.update,
>       }}
> ```
> 
> and this in componentDidMount:
> 
> ```
>     let context = this;
>     setTimeout(function () {
>       context.setState({update: 1});
>     }, 500);
> ```
> 
> In a functional component:
> 
> > ```
> > const [update, doUpdate] = useState(0);
> > 
> > return (
> >   <Camera
> >       ref={cameraRef}
> >       onInitialized={() => doUpdate(1)}
> >       style={[StyleSheet.absoluteFill, { width: width + update }]}
> > ```

This is the best way to solve this issue at the moment.

[5 more replies]