margelo/react-native-vision-camera #2587: 🐛 system/camera-module-not-found: Failed to initialize VisionCamera: The native Camera Module (`NativeModules.CameraView`) could not be found

ginxx009, 2024-02-19:
### What's happening?

This is a fresh react native project with version 0.71.14 and followed the instruction  [here](https://react-native-vision-camera.com/docs/guides) . 

### Reproduceable Code

```tsx
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 *
 * @format
 */
import React, { useEffect } from 'react';
import type {PropsWithChildren} from 'react';
import {
  ActivityIndicator,
  SafeAreaView,
  ScrollView,
  StatusBar,
  StyleSheet,
  Text,
  useColorScheme,
  View,
} from 'react-native';

import {
  Colors,
  DebugInstructions,
  Header,
  LearnMoreLinks,
  ReloadInstructions,
} from 'react-native/Libraries/NewAppScreen';

import { Camera, useCameraDevice, useCameraPermission } from 'react-native-vision-camera';

function App(): JSX.Element {
  const isDarkMode = useColorScheme() === 'dark';
  const {hasPermission, requestPermission} = useCameraPermission();
  const device = useCameraDevice('back');
  const backgroundStyle = {
    backgroundColor: isDarkMode ? Colors.darker : Colors.lighter,
  };

  useEffect(() => {
    if(!hasPermission) {
      requestPermission();
    }
  }, [hasPermission]);

 const CameraAvailable = ({devicee}: {devicee: any}) => {
  if(!device) {
    console.log('No camera device available');
    return <Text>No camera device available</Text>;
  }

  return <Camera style={{width: 500, height: 500}} device={devicee} isActive={true}/>;
 }

  return (
    <SafeAreaView style={backgroundStyle}>
      <StatusBar
        barStyle={isDarkMode ? 'light-content' : 'dark-content'}
        backgroundColor={backgroundStyle.backgroundColor}
      />
      {device && 
      <CameraAvailable devicee={device}/>
      }
    </SafeAreaView>
  );
}

const styles = StyleSheet.create({
  sectionContainer: {
    marginTop: 32,
    paddingHorizontal: 24,
  },
  sectionTitle: {
    fontSize: 24,
    fontWeight: '600',
  },
  sectionDescription: {
    marginTop: 8,
    fontSize: 18,
    fontWeight: '400',
  },
  highlight: {
    fontWeight: '700',
  },
});

export default App;
```

### Relevant log output

```shell
system/camera-module-not-found: Failed to initialize VisionCamera: The native Camera Module (`NativeModules.CameraView`) could not be found.
* Make sure react-native-vision-camera is correctly autolinked (run `npx react-native config` to verify)
* Make sure you ran `pod install` in the ios/ directory.
* Make sure you rebuilt the app., js engine: hermes 
    at App (http://localhost:8081/index.bundle//&platform=ios&dev=true&minify=false&inlineSourceMap=false&modulesOnly=false&runModule=true&app=org.reactjs.native.example.VisionCameraProject:94695:54)
    at RCTView
    at View (http://localhost:8081/index.bundle//&platform=ios&dev=true&minify=false&inlineSourceMap=false&modulesOnly=false&runModule=true&app=org.reactjs.native.example.VisionCameraProject:45212:43)
    at RCTView
    at View (http://localhost:8081/index.bundle//&platform=ios&dev=true&minify=false&inlineSourceMap=false&modulesOnly=false&runModule=true&app=org.reactjs.native.example.VisionCameraProject:45212:43)
    at AppContainer (http://localhost:8081/index.bundle//&platform=ios&dev=true&minify=false&inlineSourceMap=false&modulesOnly=false&runModule=true&app=org.reactjs.native.example.VisionCameraProject:45077:36)
    at VisionCameraProject(RootComponent) (http://localhost:8081/index.bundle//&platform=ios&dev=true&minify=false&inlineSourceMap=false&modulesOnly=false&runModule=true&app=org.reactjs.native.example.VisionCameraProject:83249:28)
 ERROR  TypeError: Cannot read property 'CameraModule' of undefined

This error is located at:
    in App
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in VisionCameraProject(RootComponent), js engine: hermes
 ERROR  TypeError: Cannot read property 'CameraModule' of undefined

This error is located at:
    in App
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in VisionCameraProject(RootComponent), js engine: hermes
```

### Camera Device

```json
n/a
```

### Device

iPhone 12(iOS17)

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ginxx009, 2024-02-19:
here's the npx react-native config

```
npx react-native config
warn Multiple Podfiles were found: ios/Podfile,vendor/bundle/ruby/2.7.0/gems/cocoapods-core-1.15.2/lib/cocoapods-core/Podfile. Choosing ios/Podfile automatically. If you would like to select a different one, you can configure it via "project.ios.sourceDir". You can learn more about it here: https://github.com/react-native-community/cli/blob/master/docs/configuration.md
{
  "root": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject",
  "reactNativePath": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/node_modules/react-native",
  "dependencies": {
    "react-native-vision-camera": {
      "root": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/node_modules/react-native-vision-camera",
      "name": "react-native-vision-camera",
      "platforms": {
        "ios": {
          "podspecPath": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/node_modules/react-native-vision-camera/VisionCamera.podspec",
          "configurations": [],
          "scriptPhases": []
        },
        "android": {
          "sourceDir": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/node_modules/react-native-vision-camera/android",
          "packageImportPath": "import com.mrousavy.camera.CameraPackage;",
          "packageInstance": "new CameraPackage()",
          "buildTypes": [],
          "componentDescriptors": [],
          "androidMkPath": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/node_modules/react-native-vision-camera/android/build/generated/source/codegen/jni/Android.mk",
          "cmakeListsPath": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/node_modules/react-native-vision-camera/android/build/generated/source/codegen/jni/CMakeLists.txt"
        }
      }
    }
  },
  "commands": [
    {
      "name": "log-ios",
      "description": "starts iOS device syslog tail"
    },
    {
      "name": "run-ios",
      "description": "builds your app and starts it on iOS simulator",
      "examples": [
        {
          "desc": "Run on a different simulator, e.g. iPhone SE (2nd generation)",
          "cmd": "react-native run-ios --simulator \"iPhone SE (2nd generation)\""
        },
        {
          "desc": "Run on a connected device, e.g. Max's iPhone",
          "cmd": "react-native run-ios --device \"Max's iPhone\""
        },
        {
          "desc": "Run on the AppleTV simulator",
          "cmd": "react-native run-ios --simulator \"Apple TV\"  --scheme \"helloworld-tvOS\""
        }
      ],
      "options": [
        {
          "name": "--simulator <string>",
          "description": "Explicitly set simulator to use. Optionally include iOS version between parenthesis at the end to match an exact version: \"iPhone 6 (10.0)\""
        },
        {
          "name": "--mode <string>",
          "description": "Explicitly set the scheme configuration to use"
        },
        {
          "name": "--configuration <string>",
          "description": "[Deprecated] Explicitly set the scheme configuration to use"
        },
        {
          "name": "--scheme <string>",
          "description": "Explicitly set Xcode scheme to use"
        },
        {
[... 217 lines cut ...]
      ]
    }
  ],
  "healthChecks": [],
  "platforms": {
    "ios": {},
    "android": {}
  },
  "project": {
    "ios": {
      "sourceDir": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/ios",
      "xcodeProject": {
        "name": "VisionCameraProject.xcworkspace",
        "isWorkspace": true
      }
    },
    "android": {
      "sourceDir": "/Users/ginxx009/Documents/VisionCameraProject/VisionCameraProject/android",
      "appName": "app",
      "packageName": "com.visioncameraproject"
    }
  }
}

```

mrousavy, 2024-02-19:
Hm it did resolve properly - can you try clearing everything/cache to see if that fixes it?

ginxx009, 2024-02-19:
tried clearing it actually and it did not resolve anything . By the way on android it works but on iOS it does not .

juninhoall, 2024-02-19:
same issue =[

k-saparia, 2024-02-21:
Same issue on my side while running `yarn test`

ginxx009, 2024-02-22:
What I am trying to do right now is upgrade the react native versions to see where it is compatible .

ginxx009, 2024-02-27:
Hi everyone . If you are still encountering this problem . What I did was 

Make sure that the swift version you are using on your xcode is 5.2 . Mine was default 5.0 so I change it on the build setting of the xcode then rebuild it . And it was working for both iOS and Android.

closed 2024-02-27 by ginxx009

leonardorib, 2024-08-06:
> Same issue on my side while running `yarn test`

Has anyone with this problem when running tests managed to solve it?

sky-kryst, 2024-09-25:
> Has anyone with this problem when running tests managed to solve it?

I was having this problem with tests. We're using this library in two places: in our app and a library within. This error occurs when `[react-native-vision-camera](https://unpkg.com/browse/react-native-vision-camera@4.5.2/)/[src](https://unpkg.com/browse/react-native-vision-camera@4.5.2/src/)/NativeCameraModule.ts` is not found in the `node_modules`. The versions of packages in both apps were not the same. Updating to the same version fixed the problem.

MaxPleaner, 2024-12-01:
I'm having this same issue on fresh-out the box Expo install. I just followed the official React instructions to get Expo (https://reactnative.dev/docs/environment-setup) and then the instructions here (https://react-native-vision-camera.com/docs/guides). I ran `npx expo prebuild` and then `eas build` and it showed this error in the iOS simulator.

AshutoshKuinkel, 2025-10-25:
I was facing this exact same issue too. I just rebuilt the app again using eas build --platform android --profile development and it worked for me!