margelo/react-native-vision-camera #2593: 🐛 Incorrect Recognition of Code-39 Barcodes on Android

melhotiby1, 2024-02-20:
### What's happening?

We're experiencing an issue with barcode scanning on Android devices. Specifically, the Code-39 barcodes are not being recognized correctly. For instance, the character 'JJ' is frequently misread as '66'. This issue is not limited to these characters; we've noticed other inaccuracies with different badges as well.

Here's an example of a barcode we're trying to scan, and the incorrect results we're getting:

The incorrect scans are random, but frequent. Here's an example of the output we're getting:

```bash
{"B-Y80KZW668019":3,"B-Y80K0W668019":1,"B-Y80KZWJJ8019":1}
```

Here is the badge you can print
[image]

Interestingly, this issue is only present on Android. The iOS version is scanning correctly, and we didn't need to implement a 'best of 5 scans' workaround.

Here are the relevant versions
```json
"react": "18.2.0",
"react-native": "0.73.4",
"react-native-reanimated": "^3.6.2",
"react-native-screens": "3.29.0",
"react-native-vision-camera": "^3.8.2"
```

Any assistance with this issue would be greatly appreciated.

### Reproduceable Code

```tsx
import React, { useEffect, useMemo, useRef, useState } from 'react'
import { Dimensions, StyleSheet } from 'react-native'
import {
  Camera,
  useCameraDevice,
  useCameraDevices,
  useCameraPermission,
  useCodeScanner,
} from 'react-native-vision-camera'
import { findBestDevice, isBarcodeInScanArea } from './helpers'

const { width, height } = Dimensions.get('window')
const SCAN_COUNT_THRESHOLD = 5

export default function CameraAndroidVision({
  onScan,
  isScanEnabledRef,
  dynamicScanArea,
}) {
  console.log(' \n \n ---- CameraAndroidVision - top ---- \n \n ')
  const scanCount = useRef({})
  const [torchEnabled, setTorchEnabled] = useState(false)
  const { requestPermission, hasPermission } = useCameraPermission()
  const camera = useRef(null)
  const devices = useCameraDevices()
  // console.log('1 - devices: ', JSON.stringify(devices, null, 2))
  const bestDevice = useMemo(() => findBestDevice(devices), [devices])
  // console.log('\n \n bestDevice: ', bestDevice)
  const device = useCameraDevice(
    'back',
    bestDevice ? { physicalDevices: bestDevice } : undefined,
  )
  let bestFormatWidth = null
  let bestFormatHeight = null
  let maxResolution = 0
  if (device && device.formats) {
    device.formats.forEach(format => {
      // console.log('in foreach - format: ', format)
      const currentResolution = format.photoWidth * format.photoHeight
      if (currentResolution > maxResolution) {
        // console.log('in the IF ')
        maxResolution = currentResolution
        bestFormatWidth = format.photoWidth
        bestFormatHeight = format.photoHeight
      }
    })
  }

  const cameraResolution = {
    width: bestFormatWidth || width,
    height: bestFormatHeight || height,
  }
  useEffect(() => {
    if (!hasPermission) {
      requestPermission()
    }
  }, [hasPermission, requestPermission])
  const codeScanner = useCodeScanner({
    codeTypes: ['code-39'],
    onCodeScanned: codes => {
      console.log({ enabled: isScanEnabledRef.current })
      if (!isScanEnabledRef.current) {
        return
      }
      if (!codes?.[0] || !codes?.[0].value) {
        return
      }

      if (!codes[0].value.startsWith('B-')) {
        console.log(codes[0].value)
        return
      }

      if (codes[0].value.length !== 14) {
        console.log(codes[0].value)
        return
      }

      if (isBarcodeInScanArea(codes[0], cameraResolution, dynamicScanArea)) {
        if (scanCount.current[codes[0].value] >= SCAN_COUNT_THRESHOLD) {
          console.log('accepting scan: ', codes[0].value)
          console.log('full object: ', JSON.stringify(scanCount.current))
          onScan(codes[0].value)
          scanCount.current = {}
          return
        }
        scanCount.current[codes[0].value] =
          scanCount.current[codes[0].value] || 0
        scanCount.current[codes[0].value]++
        console.log('adding : ', codes[0].value)
      } else {
        console.log(' NOT IN SCAN AREA')
      }
    },
  })

  return (
    <>
      {device && hasPermission && (
        <Camera
          ref={camera}
          style={styles.cameraPreview}
          device={device}
          codeScanner={codeScanner}
          isActive={true}
          photo={true}
          preset="medium"
          torch={torchEnabled ? 'on' : 'off'}
        />
      )}
    </>
  )
}

const styles = StyleSheet.create({
  cameraPreview: {
    flex: 1,
  },
})
```

### Relevant log output

```shell
^[[A02-19 22:29:30.551  1726  1971 D ActivityManager: sync unfroze 31013 com.android.settings for 3
02-19 22:29:30.553 31013 31013 D BatteryUsageBroadcastReceiver: onReceive:android.intent.action.BATTERY_LEVEL_CHANGED
02-19 22:29:30.555  1726  1971 W ProcessStats: Tracking association SourceState{9378660 com.google.android.euicc/10135 Fgs #217192} whose proc state 3 is better than process ProcessState{cb7d470 com.google.android.gms.persistent/10161 pkg=com.google.android.gms} proc state 9 (113 skipped)
^R
^[[02-19 22:29:31.803  1051  1051 D vendor.qti.bluetooth@1.0-uart_transport: SocRxDWakeup: Flow off->Change UART baudrate to 38.4kbs->send 0x00->Change UART baudrate to max->Flow on
02-19 22:29:31.803  1051  1051 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_ioctl: UART Flow Off
02-19 22:29:31.807 15170 27691 I AiAiScreenProcessor: (REDACTED) Evicting the bitmap at evictionTimeStampMs: %d
02-19 22:29:31.815  1051  1051 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_set_baud: 17
02-19 22:29:31.826  1051  1051 I vendor.qti.bluetooth@1.0-uart_transport: SetBaudRate: in BOTHER
02-19 22:29:31.826  1051  1051 I vendor.qti.bluetooth@1.0-uart_transport: ## userial_vendor_ioctl: UART Flow On
02-19 22:29:31.827  1051  3418 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
02-19 22:29:31.827  1051  3418 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
02-19 22:29:31.828  1051  3418 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
02-19 22:29:31.828  1051  3418 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Signal wack_cond_
02-19 22:29:31.828  1051  1051 D vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Unblocked from waiting for FC
02-19 22:29:33.427  1066  1264 I CHRE    : @ 196131.514: [AR] inconsistent: 41.176471
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 7,
  "minZoom": 0.6150000095367432,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera",
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "minFocusDistance": 9.800000093460085,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Iphone 14 pro

### VisionCamera Version

3.8.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

xulihang, 2024-02-22:
Android uses ML Kit white iOS uses AVCapture for reading barcodes. This may cause the difference. You can switch to [Dynamsoft Barcode Reader](https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/) for higher precision and consistent results across Android and iOS.

SMJ93, 2024-02-23:
@melhotiby1 We were getting a similar issue with Android barcodes and added a secondary check to make sure it scanned the same barcode twice:

```
  const [previousBarcode, setPreviousBarcode] = useState<string | undefined>(
    undefined,
  );
  
  const onScanBarcode = useCallback(
    async (barcodes: Code[]) => {
       // Make sure it scans the same barcode twice to prevent partial barcode scans
      setPreviousBarcode(barcodeNumber);
      if (previousBarcode !== barcodeNumber) {
        return;
      }
      
      ...
  }
```

mrousavy, 2024-04-22:
Exactly as @xulihang said - this is the MLKit library. If they do something wrong, there's nothing I can do. You can switch to alternative libraries/QR models instead using VisionCamera's powerful and extensible Frame Processors API :)

closed 2024-04-22 by mrousavy

omars94, 2024-07-04:
Thank you for the suggestion, @melhotiby1 . I tried your solution, but it took a considerable amount of time to get the correct result. To enhance it, I modified the implementation to store the previous values into an array of size 20. This way, I can check for duplicates and return the duplicate as the correct value. Here's the updated code:

 ```
if (previousCodeValue.includes(value) ) {
        setPrevCodeValue([]);
        setPrevCodeValue(value)
      } else {
        setPrevCodeValue(prev => {
          const updatedCodes = [value, ...prev].slice(0, 20); // Keep only the last 20 codes
          return updatedCodes;
        });
      }
```