margelo/react-native-vision-camera #2594: 💭 How to extend the flash duration

Z-LianYan, 2024-02-20:
### Question

How to extend the flash duration？

### VisionCamera Version

3.6.16

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-02-20:
The flash is enabled until the Camera finds a good Auto-Exposure value. Once that AE value is found, it fires a photo.

If you want to enable the flash longer, just enable `torch` yourself. Why do you even want to extend the Flash duration?

Z-LianYan, 2024-02-22:
I don't know if it's a problem with my phone, but when I turned on the flash to take photos, I didn't capture the moment of the flash

ziga-hvalec, 2024-02-26:
photo gets take too quickly. flash turns on, then turns off, then photo is taken. Tested on v3.9.0

PauloGrachetelli, 2024-05-07:
Same at xiaomi/samsung

rsainiWin, 2024-08-23:
Face similar on One Plus 11

PauloGrachetelli, 2024-08-23:
Try 3.6.4