margelo/react-native-vision-camera #2596: 🐛 codeScanner causes black screen on first camera start

dFelinger, 2024-02-20:
### What's happening?

After upgrading from 3.9.0-beta.5 to 3.9.0-beta.6 there is black screen on first app start with codeScanner enabled. Without codeScanner in 3.9.0-beta.6 or with codeScanner but in 3.9.0-beta.5 there is no this issue.

https://github.com/mrousavy/react-native-vision-camera/assets/5339870/e7f05021-aab0-4620-ba53-96d73543b692

### Reproduceable Code

```tsx
function App() {
  const { hasPermission, requestPermission } = useCameraPermission();
  useEffect(() => {
    if (!hasPermission) {
      requestPermission();
    }
  }, [hasPermission, requestPermission]);

  const device = useCameraDevice('back');

  const [active, setActive] = useState(true);
  const toggle = () => {
    setActive(!active);
  };

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      console.log(codes);
    },
  });

  if (device == null) {
    return <Text>No device</Text>;
  }

  if (!hasPermission) {
    return <Text>No hasPermission</Text>;
  }

  return (
    <View style={{ flex: 1 }}>
      <View style={{ alignItems: 'center', justifyContent: 'center' }}>
        <Text onPress={toggle}>TOGGLE ACTIVE {JSON.stringify(active)}</Text>
      </View>
      <View style={{ flex: 1, alignItems: 'center', justifyContent: 'center', backgroundColor: 'red' }}>
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={active}
          audio={false}
          photo={false}
          video={false}
          codeScanner={codeScanner}
          onError={e => {
            console.log('Error', e);
          }}
        />
      </View>
    </View>
  );
}
```

### Relevant log output

```shell
02-21 00:18:26.219 17346 17399 I ReactNativeJS: Running "testcamera" with {"rootTag":11}
02-21 00:18:26.233 17346 17412 D TrafficStats: tagSocket(134) with statsTag=0xffffffff, statsUid=-1
02-21 00:18:26.357 17346 17399 D CompatibilityChangeReporter: Compat change id reported: 250678880; UID 10192; state: ENABLED
02-21 00:18:26.365 17346 17399 D CompatibilityChangeReporter: Compat change id reported: 206033068; UID 10192; state: ENABLED
[... 115 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

Android emulator Pixel_3a_API_34_extension_level_7_arm64-v8a

### VisionCamera Version

3.9.0-beta.6

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Saidnazar08, 2024-03-01:
I have the same issue, but in my case when I use the format prop with the useCameraFormat.

nhatduy51, 2024-03-11:
same issue

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

dFelinger, 2024-03-19:
No black screen at start! But codeScanner triggering only once and if code is in preview on camera start. Tested in example app v4.0.0-beta.8

https://github.com/mrousavy/react-native-vision-camera/assets/5339870/ce5f91a3-982b-4db8-b588-2ed5dec19d51

thib-info, 2024-03-21:
Hello @dFelinger, I'm also using the QR code scanner feature of the package and faced the black screen error also.
I've only tested on real hardware and not on emulator. I'm facing this issue on some phones, but not all of them. 

Have you also tried on real devices or not ?

dFelinger, 2024-03-21:
I'm using 3.9.0-beta.5 version. There is no black screen on emulator and real devices. Tested on 2 real devices.

mrousavy, 2024-03-21:
QR Code issue is fixed in latest V4 beta! :)

thib-info, 2024-03-23:
Hello back, I can confirm you @mrousavy that's on the new beta, the scanner and the access of the camera is now working !!! However, I still have a black screen after minimizing the app and opening it again. 
Do you have any ideas of where that could come from ?

mrousavy, 2024-03-25:
Thanks guys, great to hear that V4 now works pretty well! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

> !!! However, I still have a black screen after minimizing the app and opening it again.

Hm, not sure. I'd have to spend time investigating that.

closed 2024-03-25 by mrousavy