margelo/react-native-vision-camera #2597: 🐛 Camera.onError(device/no-device)

xulihang, 2024-02-21:
### What's happening?

I got an error about the camera device although the camera can open.

### Reproduceable Code

```tsx
export default function App() {
  const [hasPermission, setHasPermission] = useState(false);
  const [isActive,setIsActive] = useState(true);
  const device = useCameraDevice("back");

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'granted');
      setIsActive(true);
    })();
  }, []);

  return (
    <SafeAreaView style={styles.container}>
      {device != null &&
      hasPermission && (
      <>
          <Camera
            style={StyleSheet.absoluteFill}
            isActive={isActive}
            device={device}
            pixelFormat='yuv'
          />
      </>)}
    </SafeAreaView>
  );
}
```

### Relevant log output

```shell
LOG  Running "VisionCameraCropperExample" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 ERROR  Camera.onError(device/no-device): [device/no-device] No device was set! Use `useCameraDevice(..)` or `Camera.getAvailableCameraDevices()` to select a suitable Camera device. [device/no-device: [device/no-device] No device was set! Use `useCameraDevice(..)` or `Camera.getAvailableCameraDevices()` to select a suitable Camera device.]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 18,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 5,
  "minExposure": -18,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

OPPO A72 5G Android 12

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

romeshyadav001, 2024-02-21:
I am also facing this issue

abhinandvk, 2024-02-22:
same issue

bglgwyng, 2024-02-23:
I got this error with the production build only, on SM-S901N.
Did you guys see this issue with the debug build?

abhinandvk, 2024-02-23:
> I got this error with the production build only, on SM-S901N. Did you guys see this issue with the debug build?

yes debug build also

nroV, 2024-02-27:
Testing on real device got this error as well

shiganaiman, 2024-02-27:
@xulihang
You try to turn on camera permission  in this app of  Android settings?
I fixed this issue.
But, I have other issue that 'userCameraPermission' where the dialog does not pop up after reinstalling app.

xulihang, 2024-02-28:
> @xulihang You try to turn on camera permission in this app of Android settings? I fixed this issue.

Permission is granted.

bglgwyng, 2024-02-28:
@xulihang Do you face this issue every time you open the app? or occasionally?

xulihang, 2024-02-28:
Everytime when directly using the camera in `App.tsx`. Never when using it in a second screen.

Jorge-Luis-Rangel-Peralta, 2024-02-28:
I hat to downgrade to version 3.8.2 and all is working without changing code

henryyohn, 2024-03-01:
I get the same issue on android phone

mattwoolford, 2024-03-01:
I have the same issue on Android, although it happens intermittently making it all the harder to debug.

Some useful information however, this happens in the debug build, even with react-native-worklets-core not installed (and frame processors disabled). Therefore I do not believe it is related to this

thib-info, 2024-03-04:
I have a same issue, but not on all my android devices. Do we have any update about this ? I can't go below the 3.9.0 release because this release fixed the torch and the QRScanner on android ...

thib-info, 2024-03-04:
If it helps a bit much here a snippet of the error message output: `E  'Camera.onError(unknown/unknown): [unknown/unknown] CAMERA_ERROR (3): endConfigure:512: Camera 0: Error configuring streams: Function not implemented (-38)', { [unknown/unknown: [unknown/unknown] CAMERA_ERROR (3): endConfigure:512: Camera 0: Error configuring streams: Function not implemented (-38)]`

mattwoolford, 2024-03-04:
> If it helps a bit much here a snippet of the error message output: `E 'Camera.onError(unknown/unknown): [unknown/unknown] CAMERA_ERROR (3): endConfigure:512: Camera 0: Error configuring streams: [...]

Hate to say it but I think this is a different issue. The error message is different to the one reported in this issue which is “device/no-device”. Maybe raise a new issue?

nhatphanims, 2024-03-08:
my solution is put an empty handler for onError props
`<Camera 
{...otherprops}
onError={()=>{}}
/>
`

Sakura-echos, 2024-03-08:
> my solution is set an empty handler for onError `<Camera {...otherprops} onError={()=>{}} /> `

but the problem still here, just no error log

nhatphanims, 2024-03-08:
@Sakura-echos Idk, after set onError handler, my release build not crash anymore

lucksp, 2024-03-10:
same issue here on `3.9.1`

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

thib-info, 2024-03-19:
My Android SDK is not the latest one, which is a pain in the ass to upgrade. But if you are advising using the last beta release, that might fix the screen access for android phone, then I'll try to upgrade it.

I'll keep you update with the results.

Sidibedev, 2024-03-23:
Im still having the issue even on the beta version 
`"react-native-vision-camera": "^4.0.0-beta.10",`

When i try to change the device position from front to back and back to front i have this error:
 
Camera.onError(unknown/unknown): Use case Preview:androidx.camera.core.Preview-16f8634e-5f04-4737-8405-40c8f1888af7 already bound to a different lifecycle. [unknown/unknown: Use case Preview:androidx.camera.core.Preview-16f8634e-5f04-4737-8405-40c8f1888af7 already bound to a different lifecycle.]

chengcheng1231, 2024-03-26:
I use version 3.9.1 and also got `Camera.onError(device/no-device)` error, but i added below to get permission within my camera screen and it works! Hope it could help someone.
```
...
  const [cameraPermission, setCameraPermission] = useState(false);

  const requestCameraPermission = async () => {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.CAMERA,
        {
          title: 'App Camera Permission',
          message: 'App needs access to your camera ',
          buttonNeutral: 'Ask Me Later',
          buttonNegative: 'Cancel',
          buttonPositive: 'OK',
        },
      );
      console.log('granted', granted);
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        setCameraPermission(true);
        console.log('You can use the camera');
      } else {
        setCameraPermission(false);
        console.log('Camera permission denied');
      }
    } catch (err) {
      console.warn(err);
    }
  };

  useLayoutEffect(() => {
    requestCameraPermission();
  }, []);
...
  ```
  and render the camera view
  ```
  {cameraPermission && device ? (
        <Camera
          style={styles.cameraContainer}
          codeScanner={codeScanner}
          device={device}
          isActive
          resizeMode="cover"
        />
      ) : (
        <Text style={styles.noCameraText}>No camera device available</Text>
      )}

iannngit, 2024-03-28:
I ran into this situation as well and got a solution. 

**Device**
Android 14, Pixel 8

**App structure**
My app has a screen for the barcode scanner (camera). For apps displaying <Camera/> conditionally, check @chengcheng1231 's solution.

**Previous logic**
I followed the doc to request camera permission on the screen of barcode scanner.

**Issues**
For the first time of using the barcode scanner, the app requests camera permission, I pressed Allow , then the app gave black/blank screen and no-device error.

**Solution/Workaround**
The core idea is to ensure both of the permission and device are ready for the <Camera/> component. My solution was moving camera permission request prior to going to the barcode scanner screen. So check and request camera permission first, then go to the barcode scanner screen only if the user grants the permission. The issue hasn't happened since.

amf-paulinho, 2024-04-05:
> yarn add react-native-vision-camera@beta

Any updates on this problem ? I am having same issue

AlixH, 2024-04-08:
Same issue here, also randomly seeing this one https://github.com/mrousavy/react-native-vision-camera/issues/2597#issuecomment-1977486076

mrousavy, 2024-04-08:
fixed it, will release a new beta in the coming days

ShamuXX, 2024-04-15:
same issue

EmanuilM, 2024-04-16:
Is there any progress with this issue? I am using version 3.9.2 and still get this error on Android device

mrousavy, 2024-04-16:
again

> fixed it, will release a new beta in the coming days

no new V4 beta yet. maybe later today

Rickson298, 2024-04-22:
the permission camera bug still happens here in SDK 33

Rickson298, 2024-04-22:
It still doesn't work on Android 13. I imagine it's because the camera permissions have changed in Android 13:

https://developer.android.com/about/versions/13/behavior-changes-13?hl=pt-br#granular-media-permissions

mrousavy, 2024-04-22:
What exactly is not working @Rickson298 ?

Rickson298, 2024-04-22:
> What exactly is not working @Rickson298 ?

```ts
  const { hasPermission, requestPermission } = useCameraPermission();
```

the function `requestPermission` does nothing

Rickson298, 2024-04-22:
> > What exactly is not working @Rickson298 ?
> 
> ```ts
>   const { hasPermission, requestPermission } = useCameraPermission();
> ```
> 
> the function `requestPermission` does nothing

and `hasPermission`always `false`

Rickson298, 2024-04-22:
@mrousavy works for me before I manually added permission to AndroidManifest.xml

```xml
<uses-permission android:name="android.permission.CAMERA"/>
```

Rickson298, 2024-04-22:
yes, I thought the other media permissions I had (or the expo plugin) would do it automatically. Sorry

mrousavy, 2024-04-22:
hm, normally it would actually add those permissions: https://github.com/mrousavy/react-native-vision-camera/blob/03d543ffe29978a6260c3899cc723310b23b25a5/package/src/expo-plugin/withVisionCamera.ts#L31

are you sure you re-ran expo prebuild?

Rickson298, 2024-04-22:
No, I just removed the Android folder and then re-ran yarn android, so the Android folder was recreated.

When running Expo prebuild, I encountered the error: Cannot find module './expo-plugin/withVisionCamera'. However, before I installed version 4, this error did not occur anymore.

Rickson298, 2024-04-22:
Anyway, I think it wasn't working because I'd forgotten to add the manifest permission myself

sergey-zakharov, 2024-05-09:
Got the same error after switching on `useSkiaFrameProcessor`
If I change it to useFrameProcessor with the same code around - no problems

"react-native-vision-camera": "version": "4.0.3"
"react-native-skia": "version": "1.2.3"
"react-native-reanimated": "version": "3.11.0"
ndkVersion = "23.1.7779620"
compileSdkVersion = 34

rajansomani12, 2024-07-12:
import React, { useEffect, useState,useRef } from 'react';
import { Text, View ,Button,Image} from 'react-native';
import { Camera, useCameraDevice,useCameraDevices } from 'react-native-vision-camera';

const App = () => {
  const [cameraPermission, setCameraPermission] = useState(null);
  const device = useCameraDevice('back'); // Set the initial camera device
  const camera = useRef<Camera>(null);
  const [capturedPhoto, setCapturedPhoto] = useState(null);
  const [showPreview, setShowPreview] = useState(false);

  const checkCameraPermission = async () => {
    const status = await Camera.getCameraPermissionStatus();
    console.log('status',status);
    
    if (status === 'granted') {
      setCameraPermission(true);
    } else if (status === 'notDetermined') {
      const permission = await Camera.requestCameraPermission();
      setCameraPermission(permission === 'authorized');
    } else {
      setCameraPermission(false);
    }
  };

  useEffect(() => {
    checkCameraPermission();
  }, []);

  if (cameraPermission === null) {
    return <Text>Checking camera permission...</Text>;
  } else if (!cameraPermission) {
    return <Text>Camera permission not granted</Text>;
  }

  if (!device) {
    return <Text>No camera device available</Text>;
  }

  // const camera = useRef<Camera>(null);
  // const camera = useRef(null);

  const takePhoto = async () => {
    try {
      if (!camera.current) {
        console.error('Camera reference not available.', camera);
        return;
      }

      const photo = await camera.current.takePhoto();
      console.log(photo);
      
      if (photo) {
        setCapturedPhoto(`file://${photo.path}`);
        setShowPreview(true);
      } else {
        console.error('Photo captured is undefined or empty.');
      }
    } catch (error) {
      console.error('Error capturing photo:', error);
    }
  };

  const confirmPhoto = () => {
    // User confirmed, further actions with the captured photo
    // For example, save the photo to storage, etc.
    console.log('Photo confirmed:', capturedPhoto);
    setShowPreview(false); // Hide the preview after confirmation
  };

  const retakePhoto = () => {
    // User wants to retake the photo
    setCapturedPhoto(null); // Clear the captured photo
    setShowPreview(false); // Hide the preview
  };

  const onCameraReady = (ref) => {
    // Camera component is ready, set the camera reference
    camera.current = ref;// Reference to the Camera component (e.g., obtained from ref prop)
  };

  return (
    <View style={{ flex: 1 }}>
      <Camera
        style={{ flex: 1 }}
        device={device}
        isActive={true}
        ref={(ref) => onCameraReady(ref)} 
        photo={true}
        video={true}
      />
       {showPreview && capturedPhoto ? (
        <View style={{ flex: 1, justifyContent: 'center', alignItems: 'center' }}>
          <Image
            source={{ uri: capturedPhoto }} // Assuming the photo is a valid URI
            style={{ width: 300, height: 300, marginBottom: 20 }}
          />
          <View style={{ flexDirection: 'row', justifyContent: 'space-between' }}>
            <Button title="Retake" onPress={retakePhoto} />
            <Button title="Confirm" onPress={confirmPhoto} />
          </View>
        </View>
      ) : (
        <View style={{ flexDirection: 'row', justifyContent: 'space-evenly' }}>
          <Button title="Take Photo" onPress={takePhoto} />
      </View>

      )}
      
    </View>
  );
};

export default App;

Try this code.

kanyo33, 2024-07-23:
For those of you still struggling with this. The problem lies in the documentation and the way the device information is returned when you are selecting a camera. 
`
import React, { useEffect, useState } from 'react';
import { StyleSheet, Text, View } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

const App = () => {
  const [hasPermission, setHasPermission] = useState(false);
  const [device, setDevice] = useState(null);
  const devices = useCameraDevices();

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  useEffect(() => {
    if (devices && devices.length > 0) {
     // find object manually based on devices returned not through useCameraDevices
      const frontCamera = devices.find(device => device.position === 'front');
      setDevice(frontCamera);
    }
  }, [devices]);

  if (!devices || !device) return <Text>Loading...</Text>;
  if (!hasPermission) return <Text>No access to camera</Text>;

  return (
    <View style={styles.container}>
      <Camera
        style={styles.camera}
        device={device}
        isActive={true}
      />
    </View>
  );
};

const styles = StyleSheet.create({
  container: {
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
    backgroundColor: '#fff',
  },
  camera: {
    width: '100%',
    height: '100%',
  },
});

export default App;
`

mrousavy, 2024-07-23:
> The problem lies in the documentation

Where exactly? @kanyo33

kanyo33, 2024-07-23:
@mrousavy I think here:
`const device = useCameraDevice('back')`
Adding the value directly into the function doesn't appear to grab the desired device. 
Instead:
```
const devices = useCameraDevice()
const frontCamera = devices.find(device => device.position === 'front');
```
Seems to work

mrousavy, 2024-07-23:
That's not true - this is exactly what `useCameraDevice('back')` is doing. It filters the devices with that position.

kanyo33, 2024-07-23:
That's what I thought too, but it returns the entire array containing all options, regardless if I add a property or leave it empty.

mrousavy, 2024-07-23:
Then you are using `useCameraDevices()`, not `useCameraDevice(..)`.

TopStar20, 2024-09-02:
useCameraDevice() is not working too.
Cannot find module './expo-plugin/withVisionCamera'.