margelo/react-native-vision-camera #2598: ✨ Is it possible to display the front and back cameras at the same time

Dablu77, 2024-02-21:
### What feature or enhancement are you suggesting?

Displaying both the front and back cameras simultaneously in a React Native application is technically feasible, but it's not straightforward using the react-native-camera library alone. The library is designed to handle a single camera instance at a time.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

Is it possible to display the front and back cameras at the same time

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

bglgwyng, 2024-02-25:
I didn't try it before, but could you try just rendering two `Camera` components at the same time?

mrousavy, 2024-03-04:
_Might_ work on Android, will not work on iOS.
This would require a lot of effort, but we actually built something like this before at Margelo. The Capture Session is a multi-cam session then.

closed 2024-03-04 by mrousavy

baatten, 2024-05-16:
Can we record from both? I mean without displaying it?