margelo/react-native-vision-camera #2603: 🔧 'react/bridging/CallbackWrapper.h' file not found

mikoprus, 2024-02-23:
### How were you trying to build the app?

I tried to build my app with XCode

I tried troubleshooting step no 5:
_Make sure you have created a Swift bridging header in your project_
However, I'm not prompted by the XCode like guide suggests:
_Press Create Bridging Header when promted._

### Full build logs

```tsx
/Users/user/Library/Developer/Xcode/DerivedData/mobile-apywzqgnyufposbigwaniblslult/Build/Products/Debug-iphonesimulator/ReactCommon/ReactCommon.framework/Headers/react/nativemodule/core/ReactCommon/CallbackWrapper.h:11:10 'react/bridging/CallbackWrapper.h' file not found
```

### Project dependencies

```json
"dependencies": {
    "@react-native-community/netinfo": "^11.2.1",
    "@react-native-firebase/analytics": "^18.7.2",
    "@react-native-firebase/app": "^18.7.2",
    "@react-navigation/native": "^6.1.9",
    "@react-navigation/native-stack": "^6.9.17",
    "@reduxjs/toolkit": "^2.2.0",
    "@sayem314/react-native-keep-awake": "^1.2.2",
    "axios": "^1.5.1",
    "events": "^3.3.0",
    "jwt-decode": "^3.1.2",
    "lottie-react-native": "^6.5.1",
    "ramda": "^0.29.1",
    "react": "18.2.0",
    "react-native": "0.72.3",
    "react-native-audio-recorder-player": "^3.6.1",
    "react-native-bouncy-checkbox": "^3.0.7",
    "react-native-circular-progress-indicator": "^4.4.2",
    "react-native-config": "^1.5.1",
    "react-native-device-info": "^10.12.0",
    "react-native-file-logger": "^0.4.1",
    "react-native-fs": "^2.20.0",
    "react-native-inset-shadow": "^1.0.3",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-reanimated": "^3.5.4",
    "react-native-render-html": "^6.3.4",
    "react-native-safe-area-context": "^4.7.4",
    "react-native-screens": "^3.27.0",
    "react-native-sound": "^0.11.2",
    "react-native-svg": "^13.14.0",
    "react-native-vision-camera": "^3.9.0",
    "react-native-zip-archive": "^6.1.0",
    "react-redux": "^9.1.0",
    "rn-fetch-blob": "^0.12.0",
    "react-native-vision-camera-face-detector": "^1.2.2",
    "react-native-worklets-core": "^0.2.4"
  }
```

### VisionCamera Version

3.9.0

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rohitbindal, 2024-02-29:
Hi @mikoprus have you checked #1121

mikoprus, 2024-02-29:
adding `s.dependency "ReactCommon"` to `VisionCamera.podspec` fixes the bug

mrousavy, 2024-03-04:
Great, maybe I need to add that again.

closed 2024-03-04 by mrousavy

mikoprus, 2024-03-04:
@mrousavy I've created a PR adding this: https://github.com/mrousavy/react-native-vision-camera/pull/2620