margelo/react-native-vision-camera #2604: 💭How to process the image captured for OCR scanning?

johnernest02-automanager, 2024-02-23:
### Question

I have already implemented the OCR scanning frame processor and it works well. But I would like to implement OCR as some sort of fallback from QR scanning when the user taps on the capture button, realizes there is no readable QR code, then checks to see the text on screen. Is this possible?

### VisionCamera Version

3.8.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

xulihang, 2024-02-26:
It seems that you need to process the same frame for QR code reading and OCR.

You can try this [frame cropping plugin](https://github.com/tony-xlh/vision-camera-cropper) to get the base64 and then use native module methods to process it.

You can add native module methods in a frame processor plugin like this one: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader

mrousavy, 2024-02-27:
Not sure if I understand, but if you want to process a local image which has already been captured, then no, that is not part of VisionCamera.

closed 2024-02-27 by mrousavy