margelo/react-native-vision-camera #2607: Running out of options | Unable to detect QR code similar to the built-in camera app in the phone

neilanthonyte, 2024-02-26:
### Question

[image]

The QR code above can be detected using the camera built-in in the phone; however, when I use the `react-native-vision-camera` . . nothing is being detected although it can detect other QR codes but I need library to work with the QR code above as well. . please help thanks

### What I tried

I already tried these configs but still not detecting the QR code above:

```
const device = useCameraDevice('back', {
    physicalDevices: [
      'ultra-wide-angle-camera',
      'wide-angle-camera',
      'telephoto-camera'
    ],
    supportsFocus: true,
    supportsLowLightBoost: true,
    supportsDepthCapture: true
  });
  const format = useCameraFormat(device, [
    {enableDepthData: true},
    {enableHighQualityPhotos: true},
    {enablePortraitEffectsMatteDelivery: true},
    {autoFocusSystem: true},
    {supportsFocus: true},
    {supportsLowLightBoost: true},
    {supportsDepthCapture: true},
    {videoStabilizationModes: 'cinematic-extended'},
    {supportsPhotoHdr: true}
  ]);
```

### VisionCamera Version

3.9.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Dingenis, 2024-02-26:
hey @neilanthonyte, did you follow the guide on the website regarding [QR scanning](https://react-native-vision-camera.com/docs/guides/code-scanning)? From your code I can't see if you pass in the codeScanner into the Camera component.

neilanthonyte, 2024-02-26:
hi @Dingenis thanks for the response . . yes I did . . here is the component

```
<Camera 
                  exposure={camera.current ? 1 : 0}
                  ref={camera}
                  focusable={true}
                  codeScanner={codeScanner} 
                  device={device} 
                  format={format}
                  enableDepthData={true}
                  enableHighQualityPhotos={true}
                  pixelFormat='native'
                  style={StyleSheet.absoluteFill}
                  isActive={show} />
```

Dingenis, 2024-02-26:
Mhh weird! Maybe you could use a different QR/barcode to scan? Could be that this specific code doesn't work too well.

Dingenis, 2024-02-26:
Mhhh I get that 🤔  Maybe you could check if the QR scanner in your app works at all by just trying to scan a bunch of codes from the internet. That way, you could narrow down the cause.

neilanthonyte, 2024-02-26:
hi @Dingenis yes upon testing several QR codes it works with the normal QR code design (white background) . . our current QR code design has black background
[image]

 . . could that be the reason why the scanner can't detect the QR code? if so do you have an idea as to why the built-in camera app can detect the QR code with black background?

Dingenis, 2024-02-27:
hey @neilanthonyte, I am not sure. This library uses Google MLKit for QR/barcodes, maybe it could be related to this: https://github.com/googlesamples/android-vision/issues/62

xulihang, 2024-02-27:
There should be a white quiet zone around the QR code or the library may not be able to detect it.

You can also use other advanced barcode reader to read it like this one: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/

neilanthonyte, 2024-02-27:
@Dingenis yep definitely related . . :) thanks so much for the reference.

neilanthonyte, 2024-02-27:
@xulihang thank you so much for the response . . I will definitely try your suggestion and post an update here on the result.

closed 2024-02-27 by mrousavy