margelo/react-native-vision-camera #2609: 🐛 some Samsung phone issue

Jin-seop, 2024-02-27:
### What's happening?

The app turns off when using the camera on the Galaxy A33 or the flash function on the S24.

### Reproduceable Code

```tsx
const [currentDevice, setCurrentDevice] = useState<CameraDevice>();

const setDeviceMode = useCallback(
    (type: 'front' | 'back') => {
      let findUltraDevice = false;
      let findWideDevice = false;
      let index = 0;
      let count = 0;
      for (let i = 0; i < devices.length; i++) {
        if ((devices[i].position === type && devices[i].physicalDevices[0]) === 'ultra-wide-angle-camera') {
          if (devices[i].physicalDevices.length > count) {
            index = i;
            count = devices[i].physicalDevices.length;
            findUltraDevice = true;
          }
        } else if (
          devices[i].position === type &&
          devices[i].physicalDevices[0] === 'wide-angle-camera' &&
          !findUltraDevice
        ) {
          if (devices[i].physicalDevices.length > count) {
            console.log('wide-angle-camera => ', i);
            index = i;
            count = devices[i].physicalDevices.length;
            findWideDevice = true;
          }
        } else if (devices[i].position === type && !findWideDevice) {
          if (devices[i].physicalDevices.length > count) {
            index = i;
            count = devices[i].physicalDevices.length;
          }
        }
      }
      setCurrentDevice(devices[index]);
    },
    [devices],
  );

useEffect(() => {
    if (devices) {
      setDeviceMode('back');
    }
  }, [devices, setDeviceMode]);

...
return (
...
  {currentDevice && (
              <Camera
                ref={(ref) => (camera.current = ref)}
                photo={true}
                style={styles.cameraViewLayout}
                device={currentDevice}
                isActive={isActive}
                zoom={currentDevice.minZoom}
                enableZoomGesture={true}
                focusable={true}
                enableHighQualityPhotos={true}
                orientation={cameraOrientation}
              />
            )}
...
)
```

### Relevant log output

```shell
//only Samsung A33 log

jdk.internal.misc.Unsafe in park
java.util.concurrent.locks.LockSupport in parkNanos at line 252
kotlinx.coroutines.BlockingCoroutine in joinBlocking at line 88
kotlinx.coroutines.BuildersKt__BuildersKt in runBlocking at line 59
kotlinx.coroutines.BuildersKt in runBlocking at line 1
kotlinx.coroutines.BuildersKt__BuildersKt in runBlocking$default at line 38
kotlinx.coroutines.BuildersKt in runBlocking$default at line 1
com.mrousavy.camera.core.CameraSession in close at line 92
com.mrousavy.camera.CameraView in destroy at line 135
com.mrousavy.camera.CameraViewManager in onDropViewInstance at line 38
com.mrousavy.camera.CameraViewManager in onDropViewInstance at line 16
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 635
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in dropView at line 646
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in dropView at line 399
Suspect Frame
com.facebook.react.uimanager.NativeViewHierarchyManager in manageChildren at line 499
Suspect Frame
com.swmansion.reanimated.layoutReanimation.ReanimatedNativeHierarchyManager in manageChildren at line 300
Suspect Frame
com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation in execute at line 217
com.facebook.react.uimanager.UIViewOperationQueue$1 in run at line 915
com.facebook.react.uimanager.UIViewOperationQueue in flushPendingBatches at line 1026
com.facebook.react.uimanager.UIViewOperationQueue in -$$Nest$mflushPendingBatches
com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback in doFrameGuarded at line 1086
com.facebook.react.uimanager.GuardedFrameCallback in doFrame at line 29
com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher in doFrame at line 175
Suspect Frame
com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1 in doFrame at line 85
android.view.Choreographer$CallbackRecord in run at line 1648
android.view.Choreographer$CallbackRecord in run at line 1659
android.view.Choreographer in doCallbacks at line 1129
android.view.Choreographer in doFrame at line 1045
android.view.Choreographer$FrameDisplayEventReceiver in run at line 1622
android.os.Handler in handleCallback at line 958
android.os.Handler in dispatchMessage at line 99
android.os.Looper in loopOnce at line 230
android.os.Looper in loop at line 319
android.app.ActivityThread in main at line 8893
java.lang.reflect.Method in invoke
com.android.internal.os.RuntimeInit$MethodAndArgsCaller in run at line 608
com.android.internal.os.ZygoteInit in main at line 1103
```

### Camera Device

```json
The app was turned off so I couldn't receive logs.
```

### Device

Samsung A33,S24

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jin-seop, 2024-02-27:
sentry log

[image]

Jin-seop, 2024-02-27:
It's a sentry AI solution, but I don't know if it will help.

Hello there! 👋

Problem Description
It seems like your application is experiencing an ANR (Application Not Responding) error. This means that your app has become unresponsive to user input for a significant amount of time, causing frustration and annoyance for your users. The error message simply says "ANR", which isn't the most informative, but it does convey the general idea.

Looking at the stack trace, the issue seems to be happening in the park function from the jdk.internal.misc.Unsafe module. Wow, that sounds like a really safe and reliable place for a problem to occur! From there, the error propagates through various functions in the Kotlin coroutines library (kotlinx.coroutines), until it reaches the dispatchMessage function in the android.os.Handler module. Finally, it ends up in the main function of the android.app.ActivityThread module.

Proposed Solution
To fix this ANR issue, you'll need to ensure that your app is not blocking the main thread for too long. Blocking the main thread can lead to unresponsiveness, as it prevents your app from handling user input and updating the UI.

One way to address this is by moving long-running tasks, such as network requests or heavy computations, to background threads or coroutines. This way, the main thread remains free to handle user interactions promptly. In this case, you'll need to review the code in the runBlocking function calls from the Kotlin coroutines library and make sure they are not used inappropriately or excessively.

Here's an example of how you can modify your code to use coroutines properly:

// Old code
runBlocking {
    // Long-running task
    // ...
}

// New code
GlobalScope.launch {
    // Long-running task
    // ...
}
By using GlobalScope.launch, you create a coroutine that runs on a background thread, allowing the main thread to stay responsive. Remember to handle any necessary synchronization and error handling within the coroutine.

What Else
Remember, Sentry can't fix this ANR issue for you, but it can help you identify where it's happening and provide insights into how it's affecting your users. With Sentry, you'll be able to catch and track ANR errors, allowing you to proactively address them and improve the overall user experience.

So don't let your app get stuck like a deer in the headlights 🦌, take the necessary steps to prevent ANRs and keep your users happy! 🎉

📜 Oh ANR, oh ANR, why do you stall?
You block the main thread, causing a brawl.
But fear not, for coroutines can save the day,
Move your work off the main thread, and let it sway. 🚀

mrousavy, 2024-02-27:
Hey - I need the native stacktrace or tombstone, otherwise I cannot do anything.

Jin-seop, 2024-02-28:
example is working,I'll solve the problem on my own

closed 2024-02-28 by Jin-seop

FaggioniHQ, 2024-05-03:
Hello There,

Any workaround for this?. We recently migrated our camera and we are getting some crashes because of this

Thanks in advance

mrousavy, 2024-05-03:
Yes this is fixed in V4.

FaggioniHQ, 2024-05-03:
Hello,

Perfect, thanks for the quick reply