margelo/react-native-vision-camera #2610: The codeType "codabar" is not supported by the Code Scanner! 🐛

baghel-rohan, 2024-02-27:
### What's happening?

using CodeScanner with codeType codabar stops the app from showing camera.

There are many other codetypes I have provided but, codabar just being in the list stops the app to show the camera.
codeTypes: ['codabar', 'code-128', 'code-39', 'code-93', 'ean-13', 'itf', 'upc-e', 'ean-8'],

### Reproduceable Code

```tsx
const codeScanner = useCodeScanner({
    codeTypes: ['codabar', 'code-128', 'code-39', 'code-93', 'ean-13', 'itf', 'upc-e', 'ean-8'],
    onCodeScanned: (codes) => {
      const detectedBarCodes = codes.map((code) => code.value)
      setBarCodes(detectedBarCodes)
    },
  })

{showCameraView ? (
        <>
          <Camera
            device={device}
            isActive={isActive}
            codeScanner={codeScanner}
            frameProcessorFps={5}
            style={[styles.visionCamera, cameraStyle]}
            torch={torch ? 'on' : 'off'}
            enableZoomGesture
            {...cameraProps}
          />
...}
```

### Relevant log output

```shell
2024-02-27 12:32:08.011200+0530 theftpatrol[373:18091] [native] VisionCamera.didSetProps(_:): Updating 16 props: [left, right, onViewReady, isActive, torch, codeScannerOptions, position, onError, enableZoomGesture, cameraId, onInitialized, enableBufferCompression, onCodeScanned, bottom, enableFrameProcessor, top]
2024-02-27 12:32:08.011448+0530 theftpatrol[373:18091] [native] VisionCamera.configure(_:): Updating Session Configuration...
2024-02-27 12:32:08.013671+0530 theftpatrol[373:18091] [native] VisionCamera.onError(_:): Invoking onError(): The codeType "codabar" is not supported by the Code Scanner!
2024-02-27 12:32:08.016869+0530 theftpatrol[373:18408] [native] VisionCamera.withSessionLock(_:): Beginning CameraSession configuration...
2024-02-27 12:32:08.016990+0530 theftpatrol[373:18405] [native] VisionCamera.configure(_:): Beginning AudioSession configuration...
2024-02-27 12:32:08.017142+0530 theftpatrol[373:18408] [native] VisionCamera.configureDevice(configuration:): Configuring Input Device...
2024-02-27 12:32:08.017163+0530 theftpatrol[373:18405] [native] VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
2024-02-27 12:32:08.017542+0530 theftpatrol[373:18408] [native] VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
2024-02-27 12:32:08.017799+0530 theftpatrol[373:18405] [native] VisionCamera.configure(_:): Committed AudioSession configuration!
2024-02-27 12:32:08.090646+0530 theftpatrol[373:18408] [native] VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
2024-02-27 12:32:08.090832+0530 theftpatrol[373:18408] [native] VisionCamera.configureOutputs(configuration:): Configuring Outputs...
2024-02-27 12:32:08.091338+0530 theftpatrol[373:18408] [native] VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
2024-02-27 12:32:08.093586+0530 theftpatrol[373:18408] [native] VisionCamera.withSessionLock(_:): Committed CameraSession configuration!
2024-02-27 12:32:08.093870+0530 theftpatrol[373:18408] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2024-02-27 12:32:08.094119+0530 theftpatrol[373:18408] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2024-02-27 12:32:08.094191+0530 theftpatrol[373:18408] [native] VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
2024-02-27 12:32:08.094608+0530 theftpatrol[373:18408] [native] VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
2024-02-27 12:32:08.095708+0530 theftpatrol[373:18408] [native] VisionCamera.onSessionInitialized(): Camera initialized!
```

### Camera Device

```json
{
  "maxZoom": 95.625,
  "position": "back",
  "isMultiCam": false,
  "name": "Back Camera",
  "hasFlash": true,
  "hasTorch": true,
  "sensorOrientation": "landscape-right",
  "supportsLowLightBoost": false,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "minZoom": 1,
  "supportsRawCapture": false,
  "supportsFocus": true,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "hardwareLevel": "full",
  "formats": []
}
```

### Device

iPhone 6 (iOS 12.5.7

### VisionCamera Version

3.6.4

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-27:
Hey - `codabar` is only available since iOS 15.4.

closed 2024-02-27 by mrousavy

iliapnmrv, 2024-03-15:
hey! this error disables camera for us. Every user with ios between 13 and 15.4 catches this error 

```ts
const onError = useCallback((error: CameraCaptureError) => {}, []);
  
  <ReanimatedCamera
         orientation="portrait"
         isActive={isCameraActive}
         style={styles.cameraStyle}
         onError={onError}
         codeScanner={codeScanner}
  />
```

mrousavy, 2024-03-15:
hey - again, I just said that `codabar` is only available since iOS 15.4. iOS did not have this before 15.4. There's nothing I can do.

wangzhi17, 2026-11-05:
Codabar can be used on the vision-camera-code-scanner library