margelo/react-native-vision-camera #2613: 🐛  Error while trying to take a photo or use frame processors and using React Native New Renderer Interop Layer

rayronvictor, 2024-02-27:
### What's happening?

The app crashes when trying to take a photo or use frame processors using the New Renderer Interop Layer.

### Reproduceable Code

```tsx
// react-native.config.js
module.exports = {
    project: {
        android: {
            unstable_reactLegacyComponentNames: ["CameraView"]
        },
        ios: {
            unstable_reactLegacyComponentNames: ["CameraView"]
        }
    },
};

// expo app.config.js
export default {
  // ...
  plugins: [
    [
      "expo-build-properties",
      {
        ios: {
          newArchEnabled: true
        },
        android: {
          newArchEnabled: true
        }
      }
    ],
    [
      "react-native-vision-camera",
      {
        "cameraPermissionText": "$(PRODUCT_NAME) needs access to your Camera."
      }
    ]
  ]
}

export default function CameraScreen() {
    const camera = useRef<Camera>(null)
    const [isCameraInitialized, setIsCameraInitialized] = useState(false)

    const device = useCameraDevice('back')

    const frameProcessor = useFrameProcessor((frame) => {
      'worklet'
      console.log(`Frame processed...`)
    }, [])

    const onInitialized = useCallback(() => {
        setIsCameraInitialized(true)
    }, [])

    const onMediaCaptured = useCallback(
        (media: PhotoFile | VideoFile, type: 'photo' | 'video') => {
            console.log(`Media captured! ${JSON.stringify(media)}`)
        }, []
    )

    if (device == null) return null;

    return (
        <View style={styles.container}>
            <Camera
                ref={camera}
                frameProcessor={frameProcessor}
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
                photo={true}
                enableZoomGesture={false}
                exposure={0}
                onInitialized={onInitialized}
            />

            <View style={styles.captureButton}>
                <Button
                    title="Capture"
                    disabled={!isCameraInitialized}
                    onPress={async () => {
                        try {
                            if (camera.current == null) throw new Error('Camera ref is null!')

                            console.log('Taking photo...')
                            const photo = await camera.current.takePhoto()
                            onMediaCaptured(photo, 'photo')
                        } catch (e) {
                            console.error('Failed to take photo!', e)
                        }
                    }}
                />
            </View>
        </View>
    );
}

const styles = StyleSheet.create({
    container: {
        flex: 1,
        alignItems: 'center',
        justifyContent: 'center',
    },
    captureButton: {
        position: 'absolute',
        alignSelf: 'center',
        bottom: 30,
    },
});
```

### Relevant log output

```shell
com.facebook.react.uimanager.IllegalViewOpera tionException: Trying to resolve view with tag 116
which doesn't exist
at com.facebook.react.uimanager.NativeViewHierarchyManager.resolveView(NativeViewHierarchyManager.java:102)
at com.facebook.react.uimanager.UlManagerModule.resolveView(UlManagerModule.java:870)
at com.mrousavy.camera.frameprocessor.VisionCameraProxy.findCameraViewByld(VisionCameraProxy.kt:43)
at com.mrousavy.camera.frameprocessor.VisionCameraProxy.setFrameProcessor$lambda$0(VisionCameraProxy.kt:52)
at com.mrousavy.camera.frameprocessor.VisionCameraProxy.$r8$lambdaSILGATtBMqfunZ4A7y30ByA5WOoQ(U
nknown Source:0)
at com.mrousavy.camera.frameprocessor.VisionCameraProxy$$ExternalSyntheticLambda0.run(Unknown Source:6)
at android.os.Handler.handleCallback(Handler.java:958)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loopOnce(Looper.java:222)
at android.os.Looper.loop(Looper.java:314)
at android.app.ActivityThread.main(ActivityThread.java)
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Xiaomi POCO X6 Pro

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-02-27:
Yea I'm using this code to find the Camera View:

```kotlin
UIManagerHelper.getUIManager(
  reactApplicationContext,
  viewId
)?.resolveView(viewId) as CameraView?
```

..where `viewId` is from JavaScript:

```ts
private get handle(): number {
  const nodeHandle = findNodeHandle(this.ref.current)
  if (nodeHandle == null || nodeHandle === -1) {
    throw new CameraRuntimeError(
      'system/view-not-found',
      "Could not get the Camera's native view tag! Does the Camera View exist in the native view-tree?",
    )
  }

  return nodeHandle
}
```

Apparently the new Renderer interop layer doesn't have support for that yet, so I'd need to manually migrate that to Fabric/TurboModules.

koktaildotcom, 2024-03-05:
I also have this issue when using ^3.9.1.

closed 2024-04-02 by mrousavy

ignaciotcrespo, 2025-08-19:
Same problem here, fixed using the following patch:

[react-native-vision-camera-npm-4.7.0-0f7ce208b2.patch](https://github.com/user-attachments/files/21845882/react-native-vision-camera-npm-4.7.0-0f7ce208b2.patch)

copied fix from https://github.com/mrousavy/react-native-vision-camera/pull/2702