margelo/react-native-vision-camera #2614: ✨ Migrate to new-arch (Fabric/TurboModules)

mrousavy, 2024-02-27:
### What feature or enhancement are you suggesting?

VisionCamera needs to migrate to the new architecture, and use Fabric for the View, and TurboModules + CodeGen for the native methods.

Since VisionCamera is a bit more complicated, there are a few blockers I have encountered:

1. ❌ I have both a View (`CameraView`) and two Modules (`CameraModule` + `DevicesModule`) in the codebase. Currently, the create-react-native-library template does not have a template for multiple CodeGen specs (views + modules).
    - [`CameraViewManager.swift`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/CameraViewManager.swift#L12-L13)
    - [`CameraDevicesManager.swift`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/CameraDevicesManager.swift#L12-L13)
    - [`CameraView.swift`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/CameraView.swift#L23)
2. ❌ For the Frame Processor runtime I need access to the `jsi::Value`/`jsi::Function` that the user passes to the `<Camera>` view directly, instead of converting it to a callback within the TurboModules/Native Modules system because the Frame Processor function is a Worklet. This is how this works currently:
    - JS: [`Camera.tsx`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/src/Camera.tsx#L449-L456)
    - Native (C++): [`VisionCameraProxy.mm`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/Frame%20Processor/VisionCameraProxy.mm#L109-L118)
3. ❌ For the Frame Processor runtime I need access to the `jsi::Runtime` as early as possible (**on demand**). Currently I get the `jsi::Runtime` from the `RCTCxxBridge`/`CatalystInstance`, which is a kinda private and unsafe API.
    - iOS: [`VisionCameraInstaller::install`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/Frame%20Processor/VisionCameraProxy.mm#L164-L177)
    - Android: [`VisionCameraProxy.kt`](https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/android/src/main/java/com/mrousavy/camera/frameprocessor/VisionCameraProxy.kt#L31-L36)

And then a few things I wanted to wait for (which aren't blockers) before switching to the new arch are:

1. I want to pass multiple callbacks to a native function. This was a limitation with the Bridge, but should now work with TurboModules I think? In my case `startRecording()` takes both `onRecordingFinished` and `onRecordingError` callbacks, **and also returns a `Promise`** which resolves once the recording has actually been started.
2. I couldn't find a clear script to trigger CodeGen (something like `npx react-native codegen`) to generate both iOS and Android specs.
3. New arch doesn't have first class Swift support as far as I know?
4. New arch doesn't support custom "hybrid" objects (so e.g. I could return an in-memory `UIImage`/`Image` instance in `takePhoto()`, instead of writing it to a file and returning a `string` as that's the only supported type)

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Currently only the renderer interop layer can be used, but there are some issues:

- https://github.com/mrousavy/react-native-vision-camera/issues/2613

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-02-27:
Thanks so much for your detailed answer @cipolleschi! This is really helpful :)

rayronvictor, 2024-02-27:
> We did a lot of work on interop layers for 0.74. The issue author is not mentioning which version of React Native they are using. It is likely that we fix them already in 0.74 (or they will be fixed soon).

I'm using RN 0.73.2

fabOnReact, 2024-02-28:
> For the Frame Processor runtime I need access to the jsi::Value/jsi::Function that the user passes to the <Camera> view directly, instead of converting it to a callback within the TurboModules/Native Modules system because the Frame Processor function is a Worklet.

https://react-native-vision-camera.com/docs/guides/frame-processors

If I try to solve the issue without using worklet (we can try to add it later).

```js
function App() {
  const userDefinedFunction = (frame) => {
    const objects = detectObjects(frame)
    console.log(`Detected ${objects.length} objects.`)
  }
  const frameProcessor = useFrameProcessor(userDefinedFunction)

  return (
    <Camera
      {...cameraProps}
      frameProcessor={frameProcessor}
    />
  )
}
```

1) Register userDefinedFunction using createFromHostFunction
2) Get the host function from cpp
```cpp
  /// Returns the underlying HostFunctionType iff isHostFunction returns true
  /// and asserts otherwise. You can use this to use std::function<>::target
  /// to get the object that was passed to create the HostFunctionType.
  ///
  /// Note: The reference returned is borrowed from the JS object underlying
  ///       \c this, and thus only lasts as long as the object underlying
  ///       \c this does.
```
https://github.com/facebook/react-native/blob/8317325fb2bf6563a9314431c42c90ff68fb35fb/packages/react-native/ReactCommon/jsi/jsi/jsi.h#L1073-L1079

3) Call the host function with parameter the current frame 
```cpp
  /// Calls the function with \c count \c args.  The \c this value of the JS
  /// function will not be set by the C++ caller, similar to calling
  /// Function.prototype.apply(undefined, args) in JS.
  /// \b Note: as with Function.prototype.apply, \c this may not always be
  /// \c undefined in the function itself.  If the function is non-strict,
  /// \c this will be set to the global object.
  Value call(Runtime& runtime, const Value* args, size_t count) const;
```
https://github.com/facebook/react-native/blob/8317325fb2bf6563a9314431c42c90ff68fb35fb/packages/react-native/ReactCommon/jsi/jsi/jsi.h#L1002-L1008

[20]: https://github.com/facebook/react-native/blob/8317325fb2bf6563a9314431c42c90ff68fb35fb/packages/react-native/ReactCommon/jsi/jsi/jsi.h#L100-L111
[21]: https://github.com/facebook/react-native/blob/8ff05b5a18db85ab699323d1745a5f17cdc8bf6c/packages/react-native/ReactCommon/react/renderer/uimanager/UIManagerBinding.cpp#L599

philIip, 2024-02-29:
> ❌ For the Frame Processor runtime I need access to the jsi::Value/jsi::Function that the user passes to the <Camera> view directly, instead of converting it to a callback within the [...]

hiiiii thanks so much for this post; reading this and the code i'm not totally understanding what the core issue is - is it that the UIManager APIs themselves are not supported / behaving as expected in the new architecture?

another question, i'm assuming that the problem you're implying with the worklet is that it needs to be run on a specific thread, but the callback conversion forces it to always be run on JS thread? the piece that i'm missing here is where the native module block conversion is happening? it seems like everything in your code pointers is pure JSI at the moment and doesn't involve the native module infra.

> ❌ For the Frame Processor runtime I need access to the jsi::Runtime as early as possible (on demand). Currently I get the jsi::Runtime from the RCTCxxBridge/CatalystInstance, which is a kinda private and unsafe API.

right now we made this backward compatible, but i will be writing some docs on what the future-looking APIs are for this!

mrousavy, 2024-02-29:
> hiiiii thanks so much for this post; reading this and the code i'm not totally understanding what the core issue is - is it that the UIManager APIs themselves are not supported / behaving as expected in the new architecture?

Hey @philIip thanks for tuning in! 👋  so the idea is that VisionCamera's View uses the RN architecture for almost everything:
- `device`
- `format`
- `isActive`
- `resizeMode`
- ...etc

But there is one prop which the user passes to the Camera that I cannot use the RN architecture for, and that is the `frameProcessor`. The reason is that I do some special transformations to that function, and need to get it as the pure `jsi::Function` on the native side to convert it to a callable Worklet on my native side. 
If I don't manually get it as a `jsi::Function` and instead use the bridge for this, it will convert it to a RCTCallback or something, which is not what I want.

This then runs on the Camera Thread and is synchronously invoked with a `jsi::HostObject` as a parameter.

But this is only one specific prop for the Camera (`frameProcessor`) the rest should use the RN architecture as normal.

The RN architecture simply doesn't have a mechanism for Worklets, or custom `jsi::HostObject`s currently, so I need to write that manually in JSI/C++.

> another question, i'm assuming that the problem you're implying with the worklet is that it needs to be run on a specific thread, but the callback conversion forces it to always be run on JS thread? [...]

Exactly, that and that I cannot pass a custom `jsi::HostObject` ("Frame") to the callback.

This is where I eventually convert the `jsi::Function` to a normal Objective-C/Swift method: 
- https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/Frame%20Processor/VisionCameraProxy.mm#L77
- https://github.com/mrousavy/react-native-vision-camera/blob/bca9472ab8d973d8079da89144f7b58a0321f5ec/package/ios/Frame%20Processor/FrameProcessor.mm#L33-L43

philIip, 2024-03-01:
ok i see, so bear with me just want to make sure i'm getting it right! this worklet scenario is not properly supported in the old or new architecture, but you've figured out a way to make it work doing all this custom JSI stuff.

is it a blocker because this is something you feel strongly should be baked into the framework, or is there actually a gap in the old / new architecture here? i think that's the last part i'm missing

mrousavy, 2024-03-01:
No worries! Yes this is something that does not work in old or new arch, and I don't think it even should be part of core. 
It's very custom stuff.

The reason I thought it was a blocker for me is because the workaround in old arch was using the `RCTCxxBridge`/`CatalystInstance` to get the Runtime, and this now no longer works in new arch - so I want to use a Cxx TurboModule to get access to the runtime directly, but I have my doubts that codegen will work in this "hybrid" mode (generate everything for C++, but also bridge 90% of the props over to Java/Swift, while the others remain in C++ JSI)

joacub, 2024-04-05:
hi @mrousavy thanks for this great project, im looking for this as we are using the scanner in a ver small application for cuba and the phones in there are very old ones, and i think this will be a really good for the performance of the overall scanner, do you ahve any time on when this will be migrated to the new arch ?, the react-native-workles is a blocker for us to use the new arch

mrousavy, 2024-04-08:
Not yet, I asked the Meta guys a question here: https://github.com/reactwg/react-native-new-architecture/discussions/167#discussioncomment-8938596 and I am waiting for a reply - that'll decide how easy it's gonna be to migrate over to new arch.

flexingCode, 2024-04-29:
hi guys, which RN version should i to use to use this lib? while is working on the migration :)

necmettindev, 2024-04-30:
> hi guys, which RN version should i to use to use this lib? while is working on the migration :)

0.73.6

mrousavy, 2024-05-06:
It's only me doing the migration - and no, only V4 will be migrated. If you are on V3, I'd recommend to switch over to V4 because there are tons of improvements

coder-xiaomo, 2024-05-07:
Thanks for the author's open source project, which has helped me a lot. I would like to know, is it a big workload to migrate v4 to new architecture, and does it take a lot of time and energy?

I am preparing to migrate to react native 0.74.1, but this project does not support the new architecture at present, I would like to know how long will it be appropriate to migrate, such as weeks, months, or even longer?

Thanks again for the author's open source project, which has helped me a lot!

MinskLeo, 2024-05-07:
Thanks for author's effort. Would be really good to have react-native-vision-camera on 0.74+.

angelo-hub, 2024-05-08:
@mrousavy this sounds like kinda an exciting low level problem to solve, I don't have full context on the C++ code, but i'd love to take a crack at it, if you have any design docs or info you can share on how the global memory address is used and set but is a reference to a C++ variable

mrousavy, 2024-05-08:
> Would be really good to have react-native-vision-camera on 0.74+

Yea, you can fund the development of this by sponsoring to the Polar pool linked here in the issue. Then we all get it faster.

> I would like to know, is it a big workload to migrate v4 to new architecture, and does it take a lot of time and energy?

Yes, because the Frame Processor runtime works a lot with the `jsi::Runtime`, and all of that needs to be migrated to a CxxTurboModule as the Bridge no longer exists (which is currently used for a lot of APIs). react-native-worklets-core also needs to be migrated, as that's a dependency of VisionCamera.

So I'm guessing if I really dedicate a full week to this, I could hack this out. Once the sponsor pool (Polar, linked here) is full, I can take a stab at this.

mrousavy, 2024-05-08:
> if you have any design docs or info you can share on how the global memory address is used and set but is a reference to a C++ variable

Thanks - not sure if I understand your question, but this is a thing I'd probably more rather tackle alone as it is quite a big refactor on the native side. Normally I'm happy about every contribution, but in this case I'd more rather build this myself as many things just require deep knowledge about the codebase of this.

Alper-Demir-Eng, 2024-05-10:
Just a question, for apps that use VC with the Frame Processor disabled, is it possible to somehow compile successfully under RN 0.74?

mrousavy, 2024-05-10:
I don't know. It's pretty losely coupled, but the only part that cannot be losely coupled is the `install()` function, which uses JSI code that doesn't work in bridgeless.
So either way it's probably the best to just migrate everything to new arch.

joacub, 2024-05-10:
vision camera currently can be working with react native 0.74.1 um using with that and no issues, the camera is even working super fast, ios buils with no issues and andoird you just need to include the error that android studio shows up and fix including the symbols that android studio suggest to you.

Alper-Demir-Eng, 2024-05-11:
@joacub Could you please create a tutorial, I've never built on Android Studio and not not sure how to go about it.

joacub, 2024-05-11:
> @joacub Could you please create a tutorial, I've never built on Android Studio and not sure how to go about it.

Are you using a bare react native or expo managed ?

Alper-Demir-Eng, 2024-05-11:
@joacub Bare RN.

joacub, 2024-05-11:
Im using a expo so dont know in bare, but if you have a android folder just open that in android studio and build, thats it, I guess in bare rn will be the same

idrisssakhi, 2024-05-11:
A small patch for those having a build issue. Not sure if the frame processor will continue working after this patch, waiting for your inputs. But the camera will work, and the build issues will be resolved.

`diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
index d697befe..8de418b0 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt
@@ -7,12 +7,14 @@ import com.facebook.jni.HybridData
 import com.facebook.proguard.annotations.DoNotStrip
 import com.facebook.react.bridge.ReactApplicationContext
 import com.facebook.react.bridge.UiThreadUtil
+import com.facebook.react.common.annotations.FrameworkAPI
 import com.facebook.react.turbomodule.core.CallInvokerHolderImpl
 import com.facebook.react.uimanager.UIManagerHelper
 import com.mrousavy.camera.core.ViewNotFoundError
 import com.mrousavy.camera.react.CameraView
 import java.lang.ref.WeakReference
 
+@OptIn(FrameworkAPI::class)
 @Suppress("KotlinJniMissingFunction") // we use fbjni.
 class VisionCameraProxy(private val reactContext: ReactApplicationContext) {
   companion object {`

tremblerz, 2024-05-12:
Thanks @idrisssakhi that worked for me!
Also had to add `implementation 'com.google.mlkit:barcode-scanning:17.2.0'` in `build.gradle` inside the `src` folder for my barcode use-case.

nikitapilgrim, 2024-05-12:
> Im using a expo so dont know in bare, but if you have a android folder just open that in android studio and build, thats it, I guess in bare rn will be the same

can you make patch pls?

joacub, 2024-05-13:
> > Im using a expo so dont know in bare, but if you have a android folder just open that in android studio and build, thats it, I guess in bare rn will be the same
> 
> can you make patch pls?

this is the patch:

https://github.com/mrousavy/react-native-vision-camera/issues/2614#issuecomment-2105995734

chawkip, 2024-05-14:
> A small patch for those having a build issue. Not sure if the frame processor will continue working after this patch, waiting for your inputs. But the camera will work, and the build issues will be [...]

Works like a charm! Thanks 🙏🏻

cjadhav, 2024-05-23:
Hi Guys,

What's the tentative plan to release this fix.?
4.0.5 is still missing this fix.

Thanks

mrousavy, 2024-05-27:
> Hi Guys,
> 
> What's the tentative plan to release this fix.? 4.0.5 is still missing this fix.
> 
> Thanks

This is not a _fix_, it's quite a big migration. As for status; I haven't worked on this yet.

Thanks for all sponsors so far!

mrousavy, 2024-06-13:
JFYI; I just released [react-native-vision-camera 4.3.2](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.3.2) which works on react-native 0.74 on both the old and the new architecture.

It is a temporary workaround, as it still uses the compatibility layer and is not a true TurboModule/Fabric view.

For now this works fine in my tests, but in the future VisionCamera will definitely need to be migrated to a TurboModule/Fabric view, which is what this issue here is about.

joacub, 2024-06-14:
> JFYI; I just released [react-native-vision-camera 4.3.2](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.3.2) which works on react-native 0.74 on both the old and the new [...]

new arch fails to build in ios:

```
The Swift pod `VisionCamera` depends upon `react-native-worklets-core`, which does not define modules. To opt into those targets generating module maps (which is necessary to import them from Swift when building as static libraries), you may set `use_modular_headers!` globally in your Podfile, or specify `:modular_headers => true` for particular dependencies.
```

mrousavy, 2024-07-15:
> If I understand correctly, supporting both old and new architectures is only temporary, right?

Well essentially every library out there only supports both new and old arch temporarily - at some point in the not so distant future, the old bridge/old arch will be completely gone from react native.
VisionCamera might just drop support for the old arch sooner than other libraries.

> May I ask what prevents keeping support for both?

Because I am maintaining VisionCamera (and a bunch of other libraries) alone and it is already a huge effort. Maintaining multiple architectures makes things twice as complicated, and I choose to keep things simple.
If the time comes when I drop old-arch support, you can always stick to an older version of VisionCamera, until you upgrade to new arch as well.

Also, unlike a few other simpler libraries, VisionCamera has a very complex JSI part (Frame Processors, with synchronous access to the Frame object and raw data), which I plan to simplify a lot by moving it to new arch. This makes the entire structure simpler and easier to maintain, but won't be backwards compatible.

And lastly, I am currently prototyping a different custom React Native Framework (basically an alternative to TurboModules), and I might migrate VisionCamera over to that.

The benefit is much better performance, simpler codebase (for the ObjC bridge and also especially for Frame Processors), and custom objects support in takePhoto() or prepareRecording().
More on that soon, maybe I'll post updates on Twitter. But this is really exciting :)

> It's quite painful that no react native Camera library seem to support both architectures

What? VisionCamera supports both old and new architecture(*1) in the latest stable release(s).

> if one would like to have a sample app for a library that supports both architectures cannot have even a quick qr code scanning in that sample without a library that supports both...

I'm not sure if I understand what you're saying. VisionCamera supports both old and new architecture(*1), and the example app has a QR code scanner embedded in it. Did you try it?

You can also simply enable the new arch flag in the example to build it with the new arch.

- *1: New architecture is currently just the compatibility layer built by Meta. It is not a true codegen'd TurboModule/Fabric view yet.

balazsgerlei, 2024-07-15:
> I'm not sure if I understand what you're saying. VisionCamera supports both old and new architecture(*1), and the example app has a QR code scanner embedded in it. Did you try it?

Sure, sorry for not being clear - I meant that _except the current release_ of react-native-vision-camera nothing seem to support both architectures (for QR code reading) and it seems like this will be temporary too.

I have not upgraded to 4.x with a library sample app that I based my use-case example on, but I surely will, but my comment was basically an inquiry on whether this state will remain or would no longer be the case (soon).

Thanks for shedding some light on your feature plans!

mrousavy, 2024-07-15:
Well yes I will eventually drop support for old-arch - probably a bit sooner than other libraries, but that is due to the complexity that will be required to maintain both archs. My plan for the new arch (maybe VisionCamera 5.x.x) will be to use the new foundation that I am writing, which is essentially just a bare C++ module that calls into Swift and Kotlin - so it will by design not support the old architecture unless I write a bridging layer for it - which I consider unnecessary if release this in e.g. 4 months. V4 will support old arch (and new arch with the compat layer), V5 then only new arch.

jeanraymonddaher, 2024-07-31:
will be awesome once we have this. Thank you !!

zzz08900, 2024-08-19:
Just noticed RN 0.75 came with an official way of accessing jsi::Runtime in TurboModules. Does that benefit RNVC by any means?

Here's the link
https://reactnative.dev/blog/2024/08/12/release-0.75

mrousavy, 2024-08-19:
Hey - yep they introduced this specifically for modules like react-native-vision-camera, but I likely won't need it anymore because I'm building Nitro Modules now.

mrousavy, 2024-10-30:
Update: I recently released [Nitro Modules](https://github.com/mrousavy/nitro), and I'm currently thinking about my implementation strategy here.

I'll likely use Nitro because VisionCamera has a bunch of native objects it passes around (like `frameProcessor={...}`, `device={...}`, or `format={...}`) as well as instances it creates imperatively (`prepareRecordingSession(): RecordingSession`, `takePhoto(): Image`).
This is not possible with Turbo/Fabric, but very well supported in Nitro - _at least as of right now_.

However Nitro does not have first-class view support, and I'd have to somehow bridge that over to a Turbo/Fabric view. (e.g. [like this](https://nitro.margelo.com/docs/view-components)).

davidburson, 2025-01-30:
Looks like quite a bit of work went into Nitro (and is still happening), looks great!  Just wondering where things are at with migrating RNVC to RN New Architecture?  Is there still more to be done on Nitro before RNVC will be migrated?  Or are they proceeding in parallel?

Or, have you settled on a different implementation strategy here?

mrousavy, 2025-01-30:
Hey - good question!

Yea **A LOT** of work went into Nitro. Funny how much work goes into this, I hope people can appreciate that. 

Nitro is production ready - but View Components are not yet directly supported by Nitro. I started working on that a month ago, and created a PR now that pretty much finishes that project: https://github.com/mrousavy/nitro/pull/512.

I am not merging this yet though because it will require react-native 0.78, which as of today is still in the RC-phase. react-native-screens and react-native-safe-area-context do not yet support RN 0.78, so they fail to build.

Once that's fixed, I will release Nitro Views. Then I will start to migrate VisionCamera over to Nitro Views.
Hopefully we can have a beta when react-native 0.78 is out.

MarianAlexandruAlecu, 2025-01-31:
@mrousavy Thank you for all the work! 🙏 

To my understanding, vision camera does not work yet with `newArchEnabled=true` in RN 0.76, right? 
And is expected to work with RN 0.76 only if `newArchEnabled=false`?

I am able to use vision camera with `newArchEnabled=true` only on Android, on iOS it crashes.

mrousavy, 2025-01-31:
You're welcome :) 

No - VisionCamera should work on both old and new arch right now.

In the future, with Nitro it will only work on new arch and RN 78+.
But there's many benefits from doing that!

MarianAlexandruAlecu, 2025-02-04:
@mrousavy `react-native-vision-camera: 4.6.3` is not compatible with `react-native: 0.76.6`, it crashes on iOS on both simulator and real device. I tried debugging the native code, but I didn't get much info. It crashes at `basic_string`. 

For a while I thought it's the bug here https://github.com/facebook/react-native/issues/48065 , but I get the same crash even when I apply myself the react-native fix in `node_modules`. Maybe changing the `node_modules/react-native` is not enough.

P.S. Android works fine. Of course, the crash happens only with `newArchEnabled: true`

bglgwyng, 2025-07-30:
I suggest that in the new version of VisionCamera, we should expose more primitive methods for the camera session and camera core.
For example, I want to directly create a `CameraSession` on the JS side and attach it to a preview component, then detach it, instead of manipulating the `Camera` component's props.
Also, it would be possible to have a background camera that doesn't have an attached preview component, but just emits frames via an `onFrame`-like callback.
We could introduce something like `CameraCoreView` to provide a more native-like API while still supporting the Camera component by wrapping `CameraCoreView`.

mrousavy, 2025-07-31:
> Web support for react native vision camera with frame processors in threads,

That won't happen unless anyone sponsors the efforts. I don't use Web, and Nitro has nothing to do with Web. So it'd be an entirely different project.

angelo-hub, 2025-08-10:
@mrousavy getting this to work in web would be a massive lift right? Would require some translation layer on top of nitro modules

mrousavy, 2025-08-11:
Yep. Would be quite a lot of effort - I can do it, but I don't have a use-case for it.

no-1ne, 2026-01-29:
just finished listening to your podcast, looking forward to vision camera v5 https://infinite.red/react-native-radio/rnr-351-transforming-packages-to-nitro-with-marc-rousavy

mrousavy, 2026-04-18:
[VisionCamera V5 is out](https://x.com/mrousavy/status/2044798370632368320?s=46&t=1cXOLI7mgsMcSjFIAruAkg) - after >2 years this issue is now finally closed!

All of the points around Frame Processors are now fixed with Nitro's Swift/Kotlin implementation.

> 1. I want to pass multiple callbacks to a native function. This was a limitation with the Bridge, but should now work with TurboModules I think? In my case startRecording() takes both [...]

Also fixed with Nitro.

> 3. New arch doesn't have first class Swift support as far as I know?

Nitro is Swift first. Fixed.

> 4. New arch doesn't support custom "hybrid" objects (so e.g. I could return an in-memory UIImage/Image instance in takePhoto(), instead of writing it to a file and returning a string as that's the only supported type)

`Photo` is now an in-memory type. Fixed.

Nitro made all of it possible

jvivasb20, 2026-04-20:
[image]