margelo/react-native-vision-camera #2622: 🐛 iPhone 12 Front Camera returning device does not support focus, when it works fine on the native iOS camera

ChristopherGabba, 2024-02-29:
### What's happening?

I am trying to use tap to focus (from example app) and it works great on the back camera, but does not work on the front camera (camera facing me). 

However, when I open the iPhone native camera app, the front camera has a good tap to focus feature and it adjusts the lighting and focus based off where I press. 

### Reproduceable Code

```tsx
//Exact same code as example app:

  const onFocusTap = useCallback(
    ({ nativeEvent: event }: GestureResponderEvent) => {
      if (!device?.supportsFocus) {
        console.log("Focus not supported!")
        return
      }
      camera.current?.focus({
        x: event.locationX,
        y: event.locationY,
      })
    },
    [device?.supportsFocus],
  )
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hasTorch": false,
  "maxExposure": 8,
  "supportsLowLightBoost": false,
  "minFocusDistance": 0,
  "isMultiCam": false,
  "name": "Front Camera",
  "minExposure": -8,
  "hasFlash": true,
  "maxZoom": 128.875,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "formats": [],
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": false,
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "hardwareLevel": "full",
  "position": "front",
  "minZoom": 1
}
```

### Device

iPhone 12 (iOS 17.3.1)

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-01:
> Relevant log output
> N/A

I'm sorry if this comes across as rude, but I don't understand why people leave out the log output. This is literally the only piece of information I need to investigate this, yet you leave it out.

Closing this issue for now, sorry. If you add logs I can re-open.

ChristopherGabba, 2024-03-05:
@mrousavy Sorry about that. I had never gotten xcode logs before! Thank you for challenging me, and I believe I figured out how. 

```logs
Bridge call to: deviceContexts
[GESTURE HANDLER] Initialize gesture handler for view <RNSScreenView: 0x12827fe00; frame = (0 0; 390 844); autoresize = W+H; gestureRecognizers = <NSArray: 0x2826cbba0>; animations = { UIPacingAnimationForAnimatorsKey=<CABasicAnimation: 0x2828e1b80>; opacity=<CASpringAnimation: 0x2828e1be0>; }; layer = <CALayer: 0x2828f6880>> reactTag: 365; frame = {{0, 0}, {390, 844}}; layer = <CALayer: 0x2828f6880>
Camera: Back Camera | Format: (3840x2160 photo / 3840x2160@60 video @ 60fps)
VisionCamera.didSetProps(_:): Updating 27 props: [onInitialized, cameraId, enableBufferCompression, onCodeScanned, photoHdr, collapsable, onStarted, lowLightBoost, isActive, pixelFormat, video, enableFpsGraph, onError, fps, onStopped, onViewReady, enableHighQualityPhotos, enableFrameProcessor, enableZoomGesture, orientation, videoHdr, audio, format, flex, exposure, zoom, photo]
VisionCamera.configure(_:): configure { ... }: Waiting for lock...
VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true)
VisionCamera.configureDevice(configuration:): Configuring Input Device...
VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
VisionCamera.configureOutputs(configuration:): Configuring Outputs...
VisionCamera.configureOutputs(configuration:): Adding Photo output...
VisionCamera.configureOutputs(configuration:): Adding Video Data output...
VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
VisionCamera.configureFormat(configuration:device:): Configuring Format (3840x2160 | 3840x2160@60.0 (ISO: 33.0..3168.0, Pixel Formats: [VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.rgb, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.unknown]))...
VisionCamera.configureFormat(configuration:device:): Successfully configured Format!
VisionCamera.onCameraStarted(): Camera started!
VisionCamera.onSessionInitialized(): Camera initialized!
VisionCamera.configure(_:): Beginning AudioSession configuration...
VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
VisionCamera.configureAudioSession(configuration:): Adding Audio input...
VisionCamera.configureAudioSession(configuration:): Adding Audio Data output...
VisionCamera.configure(_:): Committed AudioSession configuration!
[GESTURE HANDLER] Initialize gesture handler for view <RNSScreenView: 0x127754000; frame = (0 0; 390 844); autoresize = W+H; gestureRecognizers = <NSArray: 0x2826f5890>; animations = { UIPacingAnimationForAnimatorsKey=<CABasicAnimation: 0x282f01360>; position=<CASpringAnimation: 0x282f01300>; }; layer = <CALayer: 0x282f22460>> reactTag: 549; frame = {{0, 0}, {390, 844}}; layer = <CALayer: 0x282f22460>
Camera: Front Camera | Format: (3840x2160 photo / 3840x2160@60 video @ 60fps)
VisionCamera.didSetProps(_:): Updating 2 props: [cameraId, format]
VisionCamera.configure(_:): configure { ... }: Waiting for lock...
VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: false)
VisionCamera.configureDevice(configuration:): Configuring Input Device...
VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:1...
VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
VisionCamera.configureOutputs(configuration:): Configuring Outputs...
VisionCamera.configureOutputs(configuration:): Adding Photo output...
VisionCamera.configureOutputs(configuration:): Adding Video Data output...
VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
VisionCamera.configureFormat(configuration:device:): Configuring Format (3840x2160 | 3840x2160@60.0 (ISO: 23.0..2208.0, Pixel Formats: [VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.rgb, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.unknown]))...
VisionCamera.configureFormat(configuration:device:): Successfully configured Format!
VisionCamera.onSessionInitialized(): Camera initialized!
tcp_input [C57.1.1.1:3] flags=[R] seq=3256765643, ack=0, win=0 state=LAST_ACK rcv_nxt=3256765643, snd_una=2470308156
tcp_input [C57.1.1.1:3] flags=[R] seq=3256765643, ack=0, win=0 state=CLOSED rcv_nxt=3256765643, snd_una=2470308156
tcp_input [C57.1.1.1:3] flags=[R] seq=3256765643, ack=0, win=0 state=CLOSED rcv_nxt=3256765643, snd_una=2470308156
tcp_input [C58.1.1.1:3] flags=[R] seq=2265444067, ack=0, win=0 state=LAST_ACK rcv_nxt=2265444067, snd_una=3653737952
tcp_input [C58.1.1.1:3] flags=[R] seq=2265444067, ack=0, win=0 state=CLOSED rcv_nxt=2265444067, snd_una=3653737952
[... cut ...]
tcp_input [C59.1.1.1:3] flags=[R] seq=2720296222, ack=0, win=0 state=CLOSED rcv_nxt=2720296222, snd_una=3467708932
nw_connection_copy_connected_local_endpoint_block_invoke [C91] Client called nw_connection_copy_connected_local_endpoint on unconnected nw_connection
nw_connection_copy_connected_remote_endpoint_block_invoke [C91] Client called nw_connection_copy_connected_remote_endpoint on unconnected nw_connection
nw_connection_copy_protocol_metadata_internal_block_invoke [C91] Client called nw_connection_copy_protocol_metadata_internal on unconnected nw_connection
```

mrousavy, 2024-03-05:
Thanks for posting logs, yes this is exactly what I need, but it doesn't look like you are calling focus here? At least I don't see any focus locks..

ChristopherGabba, 2024-03-05:
@mrousavy Okay so I added more console logs to my code. The camera starts as a back camera (which you will see in the logs). I tapped focus and it logged the focus events. Then I swapped to front camera and it is saying "Device does not support focus!". As stated though, on the native iOS camera app, I can tap all over and get it to focus. 

My focus code:

```typescript
  const onFocusTap = useCallback(
    ({ nativeEvent: event }: GestureResponderEvent) => {
      if (!device?.supportsFocus) {
        console.log("Device does not support focus!!")
        return
      }
      console.log("FOCUSING CAMERA: x:", event.locationX, "y:", event.locationY)
      camera.current?.focus({
        x: event.locationX,
        y: event.locationY,
      })
    },
    [device?.supportsFocus],
  )
```

More logs: 
```logs
Camera: Back Camera | Format: (3840x2160 photo / 3840x2160@60 video @ 60fps)
VisionCamera.didSetProps(_:): Updating 27 props: [onInitialized, cameraId, enableBufferCompression, onCodeScanned, photoHdr, collapsable, onStarted, lowLightBoost, isActive, pixelFormat, video, enableFpsGraph, onError, fps, onStopped, onViewReady, enableHighQualityPhotos, enableFrameProcessor, enableZoomGesture, orientation, videoHdr, audio, format, flex, exposure, zoom, photo]
VisionCamera.configure(_:): configure { ... }: Waiting for lock...
VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true)
VisionCamera.configureDevice(configuration:): Configuring Input Device...
VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
VisionCamera.configureOutputs(configuration:): Configuring Outputs...
VisionCamera.configureOutputs(configuration:): Adding Photo output...
VisionCamera.configureOutputs(configuration:): Adding Video Data output...
VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
VisionCamera.configureFormat(configuration:device:): Configuring Format (3840x2160 | 3840x2160@60.0 (ISO: 33.0..3168.0, Pixel Formats: [VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.yuv, VisionCamera.PixelFormat.rgb, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.unknown, VisionCamera.PixelFormat.unknown]))...
VisionCamera.configureFormat(configuration:device:): Successfully configured Format!
VisionCamera.onCameraStarted(): Camera started!
VisionCamera.onSessionInitialized(): Camera initialized!
VisionCamera.configure(_:): Beginning AudioSession configuration...
VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
VisionCamera.configureAudioSession(configuration:): Adding Audio input...
VisionCamera.configureAudioSession(configuration:): Adding Audio Data output...
VisionCamera.configure(_:): Committed AudioSession configuration!
[GESTURE HANDLER] Initialize gesture handler for view <RNSScreenView: 0x10f878c00; frame = (0 0; 390 844); autoresize = W+H; gestureRecognizers = <NSArray: 0x2829339f0>; layer = <CALayer: 0x28270cac0>> reactTag: 555; frame = {{0, 0}, {390, 844}}; layer = <CALayer: 0x28270cac0>
'FOCUSING CAMERA: x:', 160, 'y:', 381.6666564941406
VisionCamera.focus(point:): Focusing (0.45221167831059306, 0.5737230121116377)...
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0305 14:39:44.975244 1801515008 JSIExecutor.cpp:376] Memory warning (pressure level: TRIM_MEMORY_RUNNING_CRITICAL) received by JS VM, running a GC
[I] <libMMKV.mm:301::-[MMKV onMemoryWarning]> cleaning on memory warning mmkv.default
[I] <MMKV.cpp:308::clearMemoryCache> clearMemoryCache [mmkv.default]
[I] <MemoryFile.cpp:103::close> closing fd[0x15], /var/mobile/Containers/Data/Application/1AAC40AC-AF6C-418A-9DF0-01149577BE07/Documents/mmkv/mmkv.default
'FOCUSING CAMERA: x:', 327, 'y:', 322.3333282470703
VisionCamera.focus(point:): Focusing (0.3819115263590879, 0.2219589257503949)...
'FOCUSING CAMERA: x:', 172.66665649414062, 'y:', 393
VisionCamera.focus(point:): Focusing (0.4656398104265402, 0.5470423243935953)...
```

Swapped to front facing camera:
```typescript
Camera: Front Camera | Format: (3840x2160 photo / 3840x2160@60 video @ 60fps)
VisionCamera.didSetProps(_:): Updating 2 props: [cameraId, format]
VisionCamera.configure(_:): configure { ... }: Waiting for lock...
VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: false)
VisionCamera.configureDevice(configuration:): Configuring Input Device...
VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:1...
VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
VisionCamera.configureOutputs(configuration:): Configuring Outputs...
VisionCamera.configureOutputs(configuration:): Adding Photo output...
VisionCamera.configureOutputs(configuration:): Adding Video Data output...
[... cut ...]
Device does not support focus!!
Device does not support focus!!
nw_connection_copy_connected_local_endpoint_block_invoke [C55] Client called nw_connection_copy_connected_local_endpoint on unconnected nw_connection
nw_connection_copy_connected_remote_endpoint_block_invoke [C55] Client called nw_connection_copy_connected_remote_endpoint on unconnected nw_connection
nw_connection_copy_protocol_metadata_internal_block_invoke [C55] Client called nw_connection_copy_protocol_metadata_internal on unconnected nw_connection
```

ChristopherGabba, 2024-03-15:
@mrousavy Hello Marc, just to follow up, I replicated the exact same thing with my friends iPhone 13 Pro and my iPhone 12. Both say the front camera doesn't support focus, but they both can be focused on the native camera app.

mrousavy, 2024-03-18:
Hey @ChristopherGabba - well honestly I don't really know how this can happen - I am basically just checking the value of [`isFocusPointOfInterestSupported`](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1390436-isfocuspointofinterestsupported) - I don't know why this is `false` in your case.

And I _think_ the `com.apple.avfoundation.avcapturedevice.built-in_video:1` device is the default front camera. Are there any other front camera devices on that phone? Maybe a true depth one? Idk...

ChristopherGabba, 2024-03-18:
@mrousavy Interesting.... I looked up the specs:
[image]
It would appear that it doesn't have an actual focal lens for the front facing camera. Instead they used a software based approach to focus. 

**Explanation I found:**
The iPhone 12's front-facing camera does not have a traditional focus lens like the one you might find in a DSLR camera. Instead, it uses a fixed-focus lens system, which is common in most smartphones' front-facing cameras. This means that the lens does not physically move to focus on subjects at different distances. However, the camera is designed to keep the subject in focus within a typical range of distances suitable for selfies and video calls.

The front-facing camera of the iPhone 12 uses sophisticated software algorithms to ensure that the subject is in clear focus and to adjust the depth of field digitally, creating a sharp image with a blurred background effect in Portrait mode, if desired. This is achieved through a combination of the camera's fixed optics and the phone's processing capabilities.

This may be same the for the iPhone 13 Pro as well. If this is the case, do you have any recommendations on how to solve this issue with the front facing camera for these ranges of iPhones?

mrousavy, 2024-04-22:
Well I have honestly no idea how to workaround this issue - if `focus(..)` doesn't work then there's nothing I can do, I am literally just calling the iOS Camera focus API.

closed 2024-04-22 by mrousavy