margelo/react-native-vision-camera #2625: 🐛 App Crashed While camRef.current.takePhoto() in React Native New Architechture

Vigneshwaran-crypto, 2024-03-02:
### What's happening?

        const file = await cameraRef.current.takePhoto({
          qualityPrioritization: 'quality',
          flash: 'off',
          enableShutterSound: false,
        });

### Reproduceable Code

```tsx
const camType = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera'],
  });

  const camFormat = useCameraFormat(camType, [{photoResolution: 'max'}]);

  const cameraRef = useRef(Camera);

      <Camera
          ref={cameraRef}
          device={camType}
          format={camFormat}
          style={{height: 300, width: 300}}
          isActive={true}
          photo={true}
          onError={err => LOG('camera error :', err)}
          onInitialized={() => LOG('camera onInitialized')}
        />
```

### Relevant log output

```shell
'camera error :', { [unknown/unknown: [unknown/unknown] Event: you must return a valid, non-null value from `getEventData`, or override `dispatch` and `dispatchModern`. Event: cameraStarted]
                             │ name: 'unknown/unknown',
                             │ _code: 'unknown/unknown',
                             │ _message: '[unknown/unknown] Event: you must return a valid, non-null value from `getEventData`, or override `dispatch` and `dispatchModern`. Event: cameraStarted',
                             │ _cause:
                             │ { message: 'Event: you must return a valid, non-null value from `getEventData`, or override `dispatch` and `dispatchModern`. Event: cameraStarted',
                             └ stacktrace: 'com.facebook.react.uimanager.IllegalViewOperationException: Event: you must return a valid, non-null value from `getEventData`, or override `dispatch` and `dispatchModern`. Event: cameraStarted\n\tat com.facebook.react.uimanager.events.Event.dispatch(Event.java:165)\n\tat com.facebook.react.uimanager.events.Event.dispatchModern(Event.java:212)\n\tat com.facebook.react.uimanager.events.FabricEventDispatcher.dispatchEvent(FabricEventDispatcher.java:41)\n\tat com.facebook.react.fabric.interop.InteropEventEmitter.receiveEvent(InteropEventEmitter.java:50)\n\tat com.mrousavy.camera.CameraView_EventsKt.invokeOnStarted(CameraView+Events.kt:26)\n\tat com.mrousavy.camera.CameraView.onStarted(CameraView.kt:242)\n\tat com.mrousavy.camera.core.CameraSession.setRunning(CameraSession.kt:89)\n\tat com.mrousavy.camera.core.CameraSession.configureCaptureRequest(CameraSession.kt:563)\n\tat com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:196)\n\tat com.mrousavy.camera.core.CameraSession$configure$1.invokeSuspend(Unknown Source:15)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat android.os.Handler.handleCallback(Handler.java:942)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loopOnce(Looper.java:211)\n\tat android.os.Looper.loop(Looper.java:300)\n\tat android.os.HandlerThread.run(HandlerThread.java:67)\n' } }
```

### Camera Device

```json
{
  "hardwareLevel": "full",
  "hasFlash": true,
  "hasTorch": true,
  "id": "0",
  "isMultiCam": false,
  "maxExposure": 24,
  "maxZoom": 10,
  "minExposure": -24,
  "minFocusDistance": 10,
  "minZoom": 1,
  "name": "BACK (0)",
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "position": "back",
  "sensorOrientation": "landscape-right",
  "supportsFocus": true,
  "supportsLowLightBoost": false,
  "supportsRawCapture": true
}
```

### Device

POCO M4 Pro

### VisionCamera Version

3.8.2 

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ChristopherGabba, 2024-03-06:
@Vigneshwaran-crypto I've never seen a `useRef` used like this for this package: ` const cameraRef = useRef(Camera);`. I believe your issue may be that the ref should be called as `const cameraRef= useRef<Camera | null>(null)`

I am using vision camera with react-native v73 and expo 50 and it takes a picture perfectly.

benjipott, 2025-01-07:
Same problem on android, take photo crash with new Arch enabled