margelo/react-native-vision-camera #2626: Camera.onError(code-scanner/not-compatible-with-outputs): [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.🐛

ginxx009, 2024-03-04:
### What's happening?

Currently I am having an error on android saying : 

```
E/ReactNativeJS: 'Camera.onError(code-scanner/not-compatible-with-outputs): [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.', { [code-scanner/not-compatible-with-outputs: [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.]
      name: 'code-scanner/not-compatible-with-outputs',
      _code: 'code-scanner/not-compatible-with-outputs',
      _message: '[code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.',
      _cause: undefined }
 ```

### Reproduceable Code

```tsx
<Camera
  style={StyleSheet.absoluteFillObject}
  device={device}
  isActive={scannerActive}
  frameProcessor={frameProcessor}
  codeScanner={detectedBarcode}
/>

here's what I have added so far

on the `grade.properties`

VisionCamera_enableCodeScanner=true

on the `build.gradle`

implementation 'com.google.mlkit:barcode-scanning:17.2.0'

and here's the `package.json`

"react-native-worklets-core": "^0.2.0",
"react-native-vision-camera": "^3.9.0",

By the way it works perfectly on iOS
```

### Relevant log output

```shell
E/CameraSession: Failed to configure CameraSession! Error: [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead., isRunning: false, Config-Diff: Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
    com.mrousavy.camera.core.CodeScannerTooManyOutputsError: [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.
        at com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:324)
        at com.mrousavy.camera.core.CameraSession.access$configureOutputs(CameraSession.kt:41)
        at com.mrousavy.camera.core.CameraSession$configureOutputs$1.invokeSuspend(Unknown Source:15)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
E/CameraView: invokeOnError(...):
W/System.err: com.mrousavy.camera.core.CodeScannerTooManyOutputsError: [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.
W/System.err:     at com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:324)
W/System.err:     at com.mrousavy.camera.core.CameraSession.access$configureOutputs(CameraSession.kt:41)
W/System.err:     at com.mrousavy.camera.core.CameraSession$configureOutputs$1.invokeSuspend(Unknown Source:15)
W/System.err:     at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
W/System.err:     at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
W/System.err:     at android.os.Handler.handleCallback(Handler.java:873)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
W/System.err:     at android.os.Looper.loop(Looper.java:193)
W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:65)
E/ReactNativeJS: 'Camera.onError(code-scanner/not-compatible-with-outputs): [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.', { [code-scanner/not-compatible-with-outputs: [code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.]
      name: 'code-scanner/not-compatible-with-outputs',
      _code: 'code-scanner/not-compatible-with-outputs',
      _message: '[code-scanner/not-compatible-with-outputs] CodeScanner can only be enabled when both video and frameProcessor are disabled! Use a Frame Processor Plugin for code scanning instead.',
      _cause: undefined }
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 12,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 8,
  "minExposure": -12,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Android(lollipop)

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-04:
The error is exactly what it says - CodeScanner can only be enabled when both video and frameProcessor are disabled. Use a Frame Processor Plugin for code scanning instead.

What is unclear about this?

closed 2024-03-04 by mrousavy

ginxx009, 2024-03-04:
Hi @mrousavy I forgot to mention here that I tried also to disable the frameProcessor and tried to scan a QR and it says here on the logcat

```
W/ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than MaxImages buffers
W/ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than MaxImages buffers
W/ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than MaxImages buffers
W/ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than MaxImages buffers
W/ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than MaxImages buffers
W/ImageReader_JNI: Unable to acquire a buffer item, very likely client tried to acquire more than MaxImages buffers
```