margelo/react-native-vision-camera #2631: Android 11 camera stretched 🐛

GioMaugeri, 2024-03-05:
### What's happening?

Hi, 

I'm using react-native-vision-camera on my app and I noticed that the camera in android 11 is stretched, but not in android 13. I'm using version 3.9.0 of the package.

I noticed that if use the component in this way for example:

<Camera device={device} isActive={true} style={{width: 500, height: 500}} />

And I change the height and I save, the camera is ok, but when I close it and I reopen it is again stretched.
    
I also installed the shadow lens app suggested in readme and I found the same error if can help.
 
Can you help me with that?

Thanks.

### Reproduceable Code

```tsx
<Camera
    codeScanner={codeScanner}
    style={StyleSheet.absoluteFill}
    device={device}
    isActive={true}
    audio={false}
    photo={false}
    video={false}
    onTouchEnd={() => setEnableOnCodeScanned(true)}
 />
```

### Relevant log output

```shell
03-05 10:44:34.043 13319 13319 I PreviewView: Creating PreviewView...
03-05 10:44:34.044 13319 13319 I PreviewView: PreviewView is 0x0, rendering 1411x720 content (LANDSCAPE_LEFT). Resizing to: 0x0 (COVER)
03-05 10:44:34.044   728 23194 E qdmetadata: paramType 2048 not supported
03-05 10:44:34.045 13319 13319 I CameraView: Updating CameraSession...
03-05 10:44:34.047 13319 16938 I CameraSession: configure { ... }: Waiting for lock...
03-05 10:44:34.047 13319 16938 I CameraSession: configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
03-05 10:44:34.047 13319 16938 I CameraSession: Configuring inputs for CameraSession...
03-05 10:44:34.047 13319 16938 D PersistentCameraCaptureSession: --> setInput(0)
03-05 10:44:34.048 13319 16938 I CameraSession: Destroying previous outputs...
03-05 10:44:34.048 13319 16938 I CameraSession: Creating outputs for Camera #0...
03-05 10:44:34.052 13319 16938 I CameraSession: Adding 1280x720 CodeScanner Output in YUV_420_888...
03-05 10:44:34.052 13319 17271 W DynamiteModule: Local module descriptor class for com.google.mlkit.dynamite.barcode not found.
03-05 10:44:34.053 13319 16938 D PersistentCameraCaptureSession: --> setOutputs([VIDEO (1280 x 720 CodeScanner for [QR, EAN_13] (35))])
03-05 10:44:34.053 13319 16938 I CameraSession: Successfully configured Session with 1 outputs for Camera #0!
03-05 10:44:34.053 13319 16938 D PersistentCameraCaptureSession: --> setRepeatingRequest(...)
03-05 10:44:34.053 13319 16938 D PersistentCameraCaptureSession: --> setIsActive(false)
03-05 10:44:34.053 13319 16938 D PersistentCameraCaptureSession: Configure() with isActive: false, ID: 0, device: null, session: null
03-05 10:44:34.053 13319 16938 I PersistentCameraCaptureSession: Creating new device...
03-05 10:44:34.053 13319 16938 I CameraManager: Camera #0: Opening...
03-05 10:44:34.053 13319 13319 W sionCamera.main: type=1400 audit(0.0:20234265): avc: denied { read } for name="u:object_r:vendor_persist_camera_prop:s0" dev="tmpfs" ino=19384 scontext=u:r:untrusted_app:s0:c99,c258,c512,c768 tcontext=u:object_r:vendor_persist_camera_prop:s0 tclass=file permissive=0
03-05 10:44:34.059   728 23194 E qdmetadata: paramType 2048 not supported
03-05 10:44:34.060 13319 16938 E libc    : Access denied finding property "persist.vendor.camera.privapp.list"
03-05 10:44:34.061   980 28813 I CameraService: CameraService::connect call (PID -1 "com.halog", camera ID 0) for HAL version default and Camera API version 2
03-05 10:44:34.061   980 28813 I Camera2ClientBase: Camera 0: Opened. Client: com.halog (PID 13319, UID 10611)
03-05 10:44:34.062   980 28813 I CameraDeviceClient: CameraDeviceClient 0: Opened
03-05 10:44:34.062 13319 17128 D TransportRuntime.SQLiteEventStore: Storing event with priority=DEFAULT, name=FIREBASE_ML_SDK for destination cct 
03-05 10:44:34.064   779 11579 I vendor.qti.hardware.servicetracker@1.2-service: unbindService is called for service : com.google.android.gms/.chimera.PersistentApiService and for client com.halog
03-05 10:44:34.064   779 11579 I vendor.qti.hardware.servicetracker@1.2-service: size of client connections for client: com.halogafter removal is 2
03-05 10:44:34.066  1337  2123 I ActivityTaskManager: START u0 {act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.permissioncontroller cmp=com.google.android.permissioncontroller/com.android.permissioncontroller.permission.ui.GrantPermissionsActivity (has extras)} from uid 10611
03-05 10:44:34.066  1337  1359 D EventSequenceValidator: Transition from ACTIVITY_CANCELLED to INTENT_STARTED
03-05 10:44:34.066  1337  2123 V ActivityTaskManager: Calling mServicetracker.OnActivityStateChange with flag falsestateINITIALIZING
03-05 10:44:34.066   980 28813 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
03-05 10:44:34.066  1337  2123 V ActivityTaskManager: Calling mServicetracker.OnActivityStateChange with flag truestateINITIALIZING
03-05 10:44:34.066   779 11579 I vendor.qti.hardware.servicetracker@1.2-service: Activity state is changed to : 0
[... 90 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Xiaomi Mi A3

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

frenbergFNX, 2024-03-05:
Did you try 3.9.1? There might be a fix for this issue in there but I have not verified my self. Had same issue on android 10 device

GioMaugeri, 2024-03-05:
> Did you try 3.9.1? There might be a fix for this issue in there but I have not verified my self. Had same issue on android 10 device

Yes I tried but I got same error

oscar-b, 2024-03-05:
https://github.com/mrousavy/react-native-vision-camera/issues/2583

vincenzoiacovone, 2024-03-05:
I'm using class component
Tired of trying and trying workarounds, version changes, debugging..
Sure that after the first rendering all works well..
Switch on caveman mode:

added this style in the Camera component:
```
    style={{
        ...StyleSheet.absoluteFill,
        width: Dimensions.get('screen').width,
        height: Dimensions.get('screen').height + this.state.update,
      }}
```

and this in componentDidMount: 
```
    let context = this;
    setTimeout(function () {
      context.setState({update: 1});
    }, 500);
```

frenbergFNX, 2024-03-05:
```
    const [hack, doHack] = useState(0);

    return (
	    <Camera
		    ref={cameraRef}
		    onInitialized={() => doHack(1)}
		    style={[StyleSheet.absoluteFill, { width: width + hack }]}

```

This looks like its working on a functional component 👍

GioMaugeri, 2024-03-06:
> ```
>     const [hack, doHack] = useState(0);
> 
>     return (
> 	    <Camera
> 		    ref={cameraRef}
> 		    onInitialized={() => doHack(1)}
> 		    style={[StyleSheet.absoluteFill, { width: width + hack }]}
> ```
> 
> This is working on a functional component 👍 (I've tested on samsung galaxy s10 with Android 10 and Oneplus 7 running Android 12) 🎉

This worked also for me both for android 11 and android 13 thanks.

GioMaugeri, 2024-03-06:
> ```
>     const [hack, doHack] = useState(0);
>
>     return (
> 	    <Camera
> 		    ref={cameraRef}
> 		    onInitialized={() => doHack(1)}
> 		    style={[StyleSheet.absoluteFill, { width: width + hack }]} [...]

I covered the component that contains the camera with a condition of a boolean value, but with this trick now the app sometimes crash.

This is the code that call the component:

` {cameraVisibile && <ScannerScreen onChangeCameraVisible={onChangeCameraVisible} onQrCheck={onQrCheck} />}`

And this is the component of before:

 ```
<Camera
        codeScanner={codeScanner}
        device={device}
        isActive={true}
        audio={false}
        photo={false}
        video={false}
        onInitialized={() => setHack(1)}
        style={[StyleSheet.absoluteFill, {width: Dimensions.get('screen').width + hack}]}
        onTouchEnd={() => setEnableOnCodeScanned(true)}
      />

```

mrousavy, 2024-03-06:
Duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2583

ShreyasSathyanarayana, 2024-03-08:
the hack is working, I was also facing the same issue. Tq for the help..........