margelo/react-native-vision-camera #2632: 🐛 CameraRuntimeError:  Unknown / Unknown:  Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed"

ChristopherGabba, 2024-03-05:
### What's happening?

I'm using a custom countdown timer component to invoke a callback when the timer hits 0 to start the camera recording. When the timer hits 0, sometimes the camera records perfectly, other times it throws the error above. 

### Reproduceable Code

```tsx
CAMERA TEST PAGE:
import { observer } from "mobx-react-lite"
import React, { FC, useCallback, useEffect, useMemo, useRef, useState } from "react"
import { Dimensions, View, ViewStyle } from "react-native"
import { AppStackScreenProps } from "app/navigators"
import { colors } from "../../theme"
import {
  Camera,
  CameraRuntimeError,
  useCameraDevice,
  useCameraFormat,
} from "react-native-vision-camera"
import { useIsForeground } from "app/hooks/useIsForeground"

import { useIsFocused } from "@react-navigation/native"
import Video, { IgnoreSilentSwitchType, ResizeMode } from "react-native-video"
import { Button } from "app/components"
import { PreVideoThumbnailBlur } from "./PreVideoThumbnailBlur"
import { CountdownTimer } from "./CountdownTimer"
import { TimerStatus } from "./timer.types"

const { width, height } = Dimensions.get("window")
interface CameraDebugProps extends AppStackScreenProps<"CameraDebug"> {}

export const CameraDebugScreen: FC<CameraDebugProps> = observer(function RecordingScreen(
  _props, // @demo remove-current-line
) {
  const { navigation } = _props

  // #region - initializations
  const cameraRef = useRef<Camera | null>(null)
  const [initialCountTimerStatus, setInitialCountdownTimerStatus] =
    useState<TimerStatus>("unstarted")

  const [cameraReady, setCameraReady] = useState(false)
  const [videoReady, setVideoReady] = useState(false)
  const [isVertical, setIsVertical] = useState(false)
  const isFocused = useIsFocused()
  const isForeground = useIsForeground()

  const isActive = useMemo(() => {
    return isFocused && isForeground
  }, [isFocused, isForeground])

  const device = useCameraDevice("front")
  const [targetFps] = useState(30)
  const cameraAspectRatio = 130 / 80

  const format = useCameraFormat(device, [
    { videoStabilizationMode: "standard" },
    { fps: targetFps },
    { videoAspectRatio: cameraAspectRatio },
    { videoResolution: { width: 640, height: 480 } },
    { photoAspectRatio: cameraAspectRatio },
    { photoResolution: { width: 640, height: 480 } },
  ])

  // #endregion

  useEffect(() => {
    if (cameraReady && videoReady) {
      setInitialCountdownTimerStatus('play')
    }
[... 320 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "hardwareLevel": "full"
}
```

### Device

iPhone 12 Physical Device

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-06:
What exactly is Error `-10868` again? I remember seeing this somewhere

ChristopherGabba, 2024-03-06:
@mrousavy I did a bunch of googling and docs reading on it, but truthfully everyone says that "it doesn't provide any helpful info".

Hate to be that guy, but I asked GPT-4 as a last resort and the response was: 
<details>
<summary>GPT Response</summary>
The AVFoundationErrorDomain Code=-11800 is an error that occurs in the context of iOS development when dealing with media playback, and it generally indicates that "The operation could not be completed". It's a broad error that can be caused by various issues related to media loading and playback, often indicating that the media file could not be loaded due to issues with its format, encoding, or the way it is being accessed.

Several common reasons for this error could be:

Insecure HTTP URLs being blocked by App Transport Security in iOS, requiring HTTPS instead, or needing exceptions configured in Info.plist if HTTP must be used.
Incorrect or missing file extensions for the media resources, which can lead to the system being unable to recognize the file format.
Issues with the media file encoding or how it's being served from the backend (such as incorrect Content-Type headers or streaming configurations).
Developers have found that verifying the URL, ensuring that the file has the correct extension, and confirming that the backend is properly configured to serve media files can help to resolve this issue. If the issue occurs only in a simulator and not on a real device, it might also be linked to the simulator's limitations or bugs.
</details>

I know that it isn't coming from the react-native-video package because it's being thrown in the onCameraError package. The resolution to this issue may be just "hey fix your shit timer code" lol. But figured it may be some edge case you could protect for in the react-native-vision-camera package. I spent a good amount of time making a reproducible demo for you and adding logs this time, hopefully it doesn't take you long to dig into.

ChristopherGabba, 2024-03-06:
@mrousavy Okay so this issue has plagued me now for a few weeks and I've tried a lot of updates. 

1. I rewrote the timer logic and it's still occurring.
2. I just changed the react-native-video package to expo-av's video package and the error is gone. 

I'm going to assume it's some sort of conflict between react-native-vision-camera and react-native-video. Maybe the AVAudioSessions or something?

mrousavy, 2024-03-06:
Good point, yea it could be the audio sessions.

mrousavy, 2024-04-22:
Closing this for now, as I think I am doing nothing wrong in my audio session - maybe RN Video doesn#t clean up after they are done with audio.

closed 2024-04-22 by mrousavy