margelo/react-native-vision-camera #2633: ✨ Exposing available video codes on device

ISnowFoxI, 2024-03-06:
### What feature or enhancement are you suggesting?

Having a function/hook to get the available codes for a particular device would be great. We had issues with old devices where they wouldn't support the `h265` codec and the app would crash upon recording start (example: iPad mini 4 and iPad Air 2).

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Currently, I was not able to find any workarounds.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-03-06:
Good feature request.

drLeonymous, 2024-03-06:
A dirty workaround, just for debugging time, can be:

In CameraView add:

```
private val codecs = MediaCodecList(MediaCodecList.ALL_CODECS).codecInfos.filter {
    it.isEncoder && it.supportedTypes[0].startsWith("video")
  }
  private val availableVideoCodecs: List<String> = codecs.map { it.name }
```

Then in init add a log:
`Log.i(TAG, "Available Video Codecs: $availableVideoCodecs")
` 

Open LogCat and, filter for TAG name, here CameraView and ... check Available Video Codecs

`Available Video Codecs: [c2.android.avc.encoder, OMX.google.h264.encoder, c2.android.h263.encoder, OMX.google.h263.encoder, c2.android.hevc.encoder, c2.android.mpeg4.encoder, OMX.google.mpeg4.encoder, c2.android.vp8.encoder, OMX.google.vp8.encoder, c2.android.vp9.encoder, OMX.google.vp9.encoder]
`

[image]

;)

mrousavy, 2024-03-07:
Nice find @drLeonymous - I definitely want to expose that API to JS!

eleretic, 2024-03-12:
I would suggest falling back to H264 when H265 is unavailable, that would be much safer

mrousavy, 2024-04-22:
Instead of exposing it, you can just pass `h265` and it will automatically fall back to `h264` if `h265` is not supported. Added in VisionCamera 4.0.0

closed 2024-04-22 by mrousavy