margelo/react-native-vision-camera #2634: 🐛On Android using videoCodec: 'h265' produce Recording failed! .. "java.io.IOException: prepare failed

drLeonymous, 2024-03-06:
### What's happening?

Trying to 
`use  camera.current.startRecording({ videoCodec: 'h265', ...options}) `
throwing 
`Recording failed! {"cause": {"message": "prepare failed.", "stacktrace": "java.io.IOException: prepare failed.`

Without videoCodec - OK
With videoCodec: 'h265' - OK
Tested in v3.6.1, v3.9.0, v3.9.0-allBetas, v3.9.1 with emulator or Samsung device 

### Reproduceable Code

```tsx
const startRecording = useCallback(() => {
    try {
    ...
      camera.current.startRecording({
        videoCodec: 'h265',
        flash: flash,
        onRecordingError: error => {
          console.error('Recording failed!', error);
          onStoppedRecording();
        },
       ...
```

### Relevant log output

```shell
Recording failed! {"cause": {"message": "prepare failed.", "stacktrace": "java.io.IOException: prepare failed.
	at android.media.MediaRecorder._prepare(Native Method)
	at android.media.MediaRecorder.prepare(MediaRecorder.java:1320)
	at com.mrousavy.camera.core.RecordingSession.start(RecordingSession.kt:99)
	at com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:438)
	at com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:36)
	at com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:103)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.os.HandlerThread.run(HandlerThread.java:67)
"}, "code": "capture/unknown", "message": "An unknown error occurred while trying to start a video recording! prepare failed.", "userInfo": null}
 LOG  stopped recording video!
 DEBUG  state: ACTIVE
 DEBUG  state: END
 LOG  calling stopRecording()...
 ERROR  failed to stop recording! [capture/no-recording-in-progress: [capture/no-recording-in-progress] There was no active video recording in progress! Did you call stopRecording() twice?]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 9,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -9,
  "name": "BACK (0)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
 LOG  {
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 9,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "telephoto-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -9,
  "name": "BACK (0)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel_2_XL_API_33:5554, Samsung Note10

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ISnowFoxI, 2024-03-06:
What samsung device are you using? Some devices don't support `h265` video codes, so that might be why there is an error.

drLeonymous, 2024-03-06:
Check the bug report. Same error in emulator as well

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy