margelo/react-native-vision-camera #2635: 🐛  Frame Processor Error: runAtTargetFps is not a function (it is undefined), js engine: VisionCamera

VNDRN, 2024-03-06:
### What's happening?

Using runAtTargetFps results in an error being thrown in the frameProcessor: 

> Frame Processor Error: runAtTargetFps is not a function (it is undefined), js engine: VisionCamera

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
    "worklet";
    runAtTargetFps(5, () => {
      "worklet";
      console.log("sadly. not working");
    });
  }, []);

  const format = useMemo(() => {
    const desiredWidth = 1920;
    const desiredHeight = 1080;
    if (device) {
      for (let index = 0; index < device.formats.length; index++) {
        const format = device.formats[index];
        if (format) {
          if (format.videoWidth == desiredWidth && format.videoHeight == desiredHeight) {
            return format;
          }
        }
      }
    }
    return undefined;
  }, [device?.formats]);

  useEffect(() => {
    const handlePermissions = async () => {
      const permission = await requestPermission();
      await rpm();
    };
    if (!hasPermission || !hpm) {
      handlePermissions();
    }
  }, [hasPermission, requestPermission]);

  return (
    <View style={styles.container}>
      {hasPermission ? (
        <Camera
          style={{ width: cameraSize, height: cameraSize }}
          device={device}
          isActive={true}
          enableFpsGraph
          frameProcessor={frameProcessor}
          format={format}
        />
      ) : (
        <Text>Permission not granted</Text>
      )}
      <Text style={{ padding: 8, fontSize: 32, color: "red", backgroundColor: "black" }}>{result.value}</Text>
    </View>
  );
}
```

### Relevant log output

```shell
Frame Processor Error: runAtTargetFps is not a function (it is undefined), js engine: VisionCamera
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 12,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 8,
  "minExposure": -12,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

onePlus Nord

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

VNDRN, 2024-03-06:
Yes! I've updated the reproducable code with the import from rnvc. I've also added my package.json

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy