margelo/react-native-vision-camera #2636: 🐛 Video recording does not record audio on some phones

redundancer, 2024-03-06:
### What's happening?

I implemented video recording, but it does not record audio on some devices

### Reproduceable Code

```tsx
const cameraRef = useRef<Camera>(null);
const frontCamera = useCameraDevice('front');

const [currentCamera, setCurrentCamera] = useState(frontCamera);
const format = useCameraFormat(currentCamera, Templates.Snapchat)

<Camera
        ref={cameraRef}
        device={currentCamera}
        isActive={true}
        style={{flex: 1}}
        format={format}
        fps={30}
        video={true}
        audio={true}
        resizeMode="contain"
      />

const startRecording = useCallback(() => {
    cameraRef.current?.startRecording({
      videoBitRate: 'normal',
      videoCodec: 'h265',
      fileType: 'mp4',
      onRecordingError,
      onRecordingFinished,
    });
  }, [onRecordingError, onRecordingFinished]);
```

### Relevant log output

```shell
I logged the cameraRef.current:

CAMERA REF:  {"_reactInternalInstance": {}, "_reactInternals": {"_debugHookTypes": null, "_debugNeedsRemount": false, "_debugOwner": {"_debugHookTypes": [Array], "_debugNeedsRemount": false, "_debugOwner": [FiberNode], "_debugSource": undefined, "actualDuration": 1567.8377629816532, "actualStartTime": 66696569.853409, "alternate": null, "child": [FiberNode], "childLanes": 0, "deletions": null, "dependencies": [Object], "elementType": [Function RecordVideo], "flags": 8390657, "index": 0, "key": null, "lanes": 0, "memoizedProps": [Object], "memoizedState": [Object], "mode": 2, "pendingProps": [Object], "ref": null, "return": [FiberNode], "selfBaseDuration": 1549.6784360036254, "sibling": null, "stateNode": null, "subtreeFlags": 9439745, "tag": 0, "treeBaseDuration": 1556.697657994926, "type": [Function RecordVideo], "updateQueue": [Object]}, "_debugSource": undefined, "actualDuration": 7.062550999224186, "actualStartTime": 66698119.708305, "alternate": null, "child": {"_debugHookTypes": null, "_debugNeedsRemount": false, "_debugOwner": [Circular], "_debugSource": undefined, "actualDuration": 5.792342998087406, "actualStartTime": 66698120.974555, "alternate": null, "child": null, "childLanes": 0, "deletions": null, "dependencies": null, "elementType": "CameraView", "flags": 512, "index": 0, "key": null, "lanes": 0, "memoizedProps": [Object], "memoizedState": null, "mode": 2, "pendingProps": [Object], "ref": [Object], "return": [Circular], "selfBaseDuration": 0.01895800232887268, "sibling": null, "stateNode": [ReactNativeFiberHostComponent], "subtreeFlags": 0, "tag": 5, "treeBaseDuration": 0.01895800232887268, "type": "CameraView", "updateQueue": null}, "childLanes": 0, "deletions": null, "dependencies": null, "elementType": [Function Camera], "flags": 1049089, "index": 0, "key": null, "lanes": 0, "memoizedProps": {"audio": true, "device": [Object], "format": [Object], "fps": 30, "isActive": true, "resizeMode": "contain", "style": [Object], "video": true}, "memoizedState": {"isRecordingWithFlash": false}, "mode": 2, "pendingProps": {"audio": true, "device": [Object], "format": [Object], "fps": 30, "isActive": true, "resizeMode": "contain", "style": [Object], "video": true}, "ref": {"current": [Circular]}, "return": {"_debugHookTypes": null, "_debugNeedsRemount": false, "_debugOwner": [FiberNode], "_debugSource": undefined, "actualDuration": 18.02906597405672, "actualStartTime": 66698119.664034, "alternate": null, "child": [Circular], "childLanes": 0, "deletions": null, "dependencies": null, "elementType": "RCTView", "flags": 0, "index": 0, "key": null, "lanes": 0, "memoizedProps": [Object], "memoizedState": null, "mode": 2, "pendingProps": [Object], "ref": null, "return": [FiberNode], "selfBaseDuration": 0.04275999963283539, "sibling": null, "stateNode": [ReactNativeFiberHostComponent], "subtreeFlags": 9439745, "tag": 5, "treeBaseDuration": 6.897815987467766, "type": "RCTView", "updateQueue": null}, "selfBaseDuration": 1.2610940039157867, "sibling": {"_debugHookTypes": null, "_debugNeedsRemount": false, "_debugOwner": [FiberNode], "_debugSource": undefined, "actualDuration": 10.76250497251749, "actualStartTime": 66698126.788461, "alternate": null, "child": [FiberNode], "childLanes": 0, "deletions": null, "dependencies": null, "elementType": [Object], "flags": 1048577, "index": 1, "key": null, "lanes": 0, "memoizedProps": [Object], "memoizedState": null, "mode": 2, "pendingProps": [Object], "ref": null, "return": [FiberNode], "selfBaseDuration": 0.13895799964666367, "sibling": null, "stateNode": null, "subtreeFlags": 9439233, "tag": 11, "treeBaseDuration": 5.575003981590271, "type": [Object], "updateQueue": null}, "stateNode": [Circular], "subtreeFlags": 512, "tag": 1, "treeBaseDuration": 1.2800520062446594, "type": [Function Camera], "updateQueue": {"baseState": [Object], "effects": null, "firstBaseUpdate": null, "lastBaseUpdate": null, "shared": [Object]}}, "context": {}, "displayName": "Camera", "isNativeViewMounted": false, "lastFrameProcessor": undefined, "onCodeScanned": [Function bound onCodeScanned], "onError": [Function bound onError], "onInitialized": [Function bound onInitialized], "onStarted": [Function bound onStarted], "onStopped": [Function bound onStopped], "onViewReady": [Function bound onViewReady], "props": {"audio": true, "device": {"formats": [Array], "hardwareLevel": "full", "hasFlash": false, "hasTorch": false, "id": "1", "isMultiCam": false, "maxExposure": 12, "maxZoom": 8, "minExposure": -12, "minFocusDistance": 0, "minZoom": 1, "name": "FRONT (1)", "neutralZoom": 1, "physicalDevices": [Array], "position": "front", "sensorOrientation": "landscape-right", "supportsFocus": false, "supportsLowLightBoost": false, "supportsRawCapture": true}, "format": {"autoFocusSystem": "none", "fieldOfView": 81.76763631551827, "maxFps": 30, "maxISO": 1600, "maxZoom": 8, "minFps": 1, "minISO": 100, "photoHeight": 480, "photoWidth": 864, "pixelFormats": [Array], "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 480, "videoStabilizationModes": [Array], "videoWidth": 864}, "fps": 30, "isActive": true, "resizeMode": "contain", "style": {"flex": 1}, "video": true}, "ref": {"current": {"_children": [Array], "_internalFiberInstanceHandleDEV": [FiberNode], "_nativeTag": 2133, "viewConfig": [Object]}}, "refs": {}, "state": {"isRecordingWithFlash": false}, "updater": {"enqueueForceUpdate": [Function enqueueForceUpdate], "enqueueReplaceState": [Function enqueueReplaceState], "enqueueSetState": [Function enqueueSetState], "isMounted": [Function isMounted]}}
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 12,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": false,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -12,
  "name": "FRONT (1)",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Fairphone 4 (/e/OS v1.20, Android 12)

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mauriceKalevra, 2024-03-06:
+1 on phone  Pixel 3a

mrousavy, 2024-03-06:
> I logged the cameraRef.current:

Why?

The log section explicitly tells you to put adb logcat logs in there. I need those, otherwise I cannot do anything.

closed 2024-03-06 by mrousavy

redundancer, 2024-03-07:
> > I logged the cameraRef.current:
> 
> Why?
> 
> The log section explicitly tells you to put adb logcat logs in there. I need those, otherwise I cannot do anything.

Thank you for the feedback. I updated the issue with the adb logcat logs.