margelo/react-native-vision-camera #2638: Camera image stretched but fixed when  height  value updates.💭

Sfeeen, 2024-03-07:
### Question

I'm using  version 3.9.0 of the vision  camera. I made an app to take images with the camera and it works normal on  a  Redmi Xiaomi device. But when I use a samsung galaxy A12 for testing/developing the image I get from the camera is stretched. This is how my camera is defined. cameraheight  = 16 / 9  * screen.width
```

        <Camera
          ref={myCamera}
          photo={true}
          device={device}
          focusable={false}
          isActive={cameraLoaded}
          format={format}
          style={{height: cameraheight + 1}}
          onError={handleCameraError}
          onInitialized={() => {
            console.log("camera ready!")
            setIsCameraReady(true)
          } }
        />
```

### What I tried

When I make any change to the style property of the camera, e.g. add 1 to the cameraheight. The images refreshes and is suddenly not stretched anymore. When  the camera  gets unloaded and loaded again the image is stretched again

### VisionCamera Version

3.9.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Dingenis, 2024-03-07:
I think this might be a duplicate of #2583

lucksp, 2024-03-10:
I too am seeing this same issue.  #2583 is closed, we should reopen or keep this now.

I get an error message, which I do not understand since I am using the `useCameraDevice` hook:

> Camera.onError(device/no-device): [device/no-device] No device was set! Use `useCameraDevice(..)` or `Camera.getAvailableCameraDevices()` to select a suitable Camera device.

```js
  const cameraRef = useRef<Component<AnimatedProps<CameraProps>>>(null);
  const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera', 'wide-angle-camera'],
  });
  const isFocused = useIsFocused();
  const isForeground = useIsForeground();
  const isActive = isFocused && isForeground;

  useEffect(
    function checkPermissions() {
      const status = Camera.getCameraPermissionStatus();
      setCameraPermission(status);
      if (cameraPermission === 'granted') return;
      Camera.requestCameraPermission();
    },
    [cameraPermission]
  );

{!!device && (
            <Reanimated.View>
              <Camera
                animatedProps={animatedProps}
                device={device}
                isActive={isActive}
                photo
                ref={cameraRef}
                style={styles.cameraContainer}
              />
            </Reanimated.View>
      )}
```

All I did was upgrade my RNVC package to `^3.9.1` since I also updated react native to `0.73.4` as part of the Expo SDK50 update.

lucksp, 2024-03-12:
> I think this might be a duplicate of #2583

Also a dupe of #2142

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

j3kin000, 2024-03-20:
> Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see [...]

The stretched issue is fixed but then the onCodeScanned wont be trigger when scanning qr/barcodes

lucksp, 2024-03-20:
Anyone here using Expo with Android SDK (34)?

mrousavy, 2024-03-21:
> The stretched issue is fixed but then the onCodeScanned wont be trigger when scanning qr/barcodes

I just fixed the Code Scanner issue in the latest V4 beta.

Sfeeen, 2024-03-22:
Tested with the V4 beta and works normally!

closed 2024-03-22 by Sfeeen

moafandi0, 2024-03-25:
> Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see [...]

I was having a similar issue. The frame width and height were coming out equal when they are not. I I was also getting 10 FPS on a OnePlus5 device and now I'm getting 17. (I'm using a native Frame Processor).  Great job man!! I love this library

mrousavy, 2024-03-25:
Thanks guys, great to hear that V4 now works perfectly! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

pranavb2402, 2024-07-17:
Can someone provide the patch for the issue so that I don't need to upgrade the SDK version.