margelo/react-native-vision-camera #2639: App crash when I close the component that contains camera💭

GioMaugeri, 2024-03-07:
### Question

Hi,

I have a component that is visible only when a boolean value is true. Inside this component there is this:

```
 <>
      <View style={{flex: 1}}>
        <Camera
          codeScanner={codeScanner}
          device={device}
          isActive={true}
          style={[StyleSheet.absoluteFill]}
          onTouchEnd={() => setEnableOnCodeScanned(true)}
        />

        <TouchableOpacity
          onPress={() => setCameraOn(false)}
          style={{alignItems: 'center', position: 'absolute', zIndex: 1, top: '90%', alignSelf: 'center'}}>
          <View
            style={{
              backgroundColor: '#FFF',
              borderRadius: 50,
              width: 50,
              height: 50,
              justifyContent: 'center',
              alignItems: 'center',
            }}>
            <Text style={{color: '#000'}}>X</Text>
          </View>
        </TouchableOpacity>
      </View>
    </>
```

And this is how I show the component:

` {cameraOn && <ScannerScreen onQrCheck={onQrCheck} />}`
 
When I close the component pressing the button or scanning the qr code, sometimes the app crash. I think that is because I created too much time the instance of camera, and so the app crash, but in camera component I don't found a prop that make the camera not visible, so I can't find another way. 

These are the logs that I take from adb logcat command when the app crash:

```
03-07 11:11:39.772   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: total connections for client : com.google.android.googlequicksearchbox:searchare :4
03-07 11:11:39.773  1337  2120 V ActivityTaskManager: Calling mServicetracker.OnActivityStateChange with flag truestateFINISHING
03-07 11:11:39.774   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: Activity state is changed to : 7
03-07 11:11:39.774  1337  2120 V ActivityTaskManager: Calling mServicetracker.OnActivityStateChange with flag falsestateFINISHING
03-07 11:11:39.775  1337  2120 V ActivityTaskManager: Calling mServicetracker.OnActivityStateChange with flag truestateDESTROYING
03-07 11:11:39.775  1337  2120 V ActivityTaskManager: Calling mServicetracker.OnActivityStateChange with flag falsestateDESTROYING
03-07 11:11:39.774   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: Activity state is changed to : 7
03-07 11:11:39.775   728 23194 E qdmetadata: paramType 2048 not supported
03-07 11:11:39.775   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: Activity state is changed to : 8
03-07 11:11:39.775   728 23194 E qdmetadata: paramType 2048 not supported
03-07 11:11:39.775   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: Activity state is changed to : 8
03-07 11:11:39.779   728 23194 E qdmetadata: paramType 2048 not supported
03-07 11:11:39.781 20639 20639 D BoundBrokerSvc: onBind: Intent { act=com.google.android.gms.inappreach.service.START dat=chimera-action: cmp=com.google.android.gms/.chimera.GmsApiService }
03-07 11:11:39.781 20639 20639 D BoundBrokerSvc: Loading bound service for intent: Intent { act=com.google.android.gms.inappreach.service.START dat=chimera-action: cmp=com.google.android.gms/.chimera.GmsApiService }
03-07 11:11:39.789 14930 14930 I GoogleInputMethodService: GoogleInputMethodService.onFinishInput():3058 
03-07 11:11:39.791   728 23194 E qdmetadata: paramType 2048 not supported
03-07 11:11:39.792 14930 14930 I GoogleInputMethodService: GoogleInputMethodService.updateDeviceLockedStatus():2136 repeatCheckTimes = 0, locked = false
03-07 11:11:39.793 14930 14930 I GoogleInputMethodService: GoogleInputMethodService.onStartInput():1882 onStartInput(EditorInfo{EditorInfo{packageName=com.android.launcher3, inputType=0, inputTypeString=NULL, enableLearning=false, autoCorrection=false, autoComplete=false, imeOptions=0, privateImeOptions=null, actionName=UNSPECIFIED, actionLabel=null, initialSelStart=-1, initialSelEnd=-1, initialCapsMode=0, label=null, fieldId=-1, fieldName=null, extras=null, hintText=null, hintLocales=[]}}, false)
03-07 11:11:39.795 14930 14930 I GoogleInputMethodService: GoogleInputMethodService.updateDeviceLockedStatus():2136 repeatCheckTimes = 1, locked = false
03-07 11:11:39.795   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: bindService is called for service : com.google.android.gms/.chimera.PersistentDirectBootAwareApiService and for client com.google.android.gms.persistent
03-07 11:11:39.795   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: total connections for service : com.google.android.gms/.chimera.PersistentDirectBootAwareApiServiceare :25
03-07 11:11:39.795   779 14325 I vendor.qti.hardware.servicetracker@1.2-service: total connections for client : com.google.android.gms.persistentare :4
03-07 11:11:39.795  2207  2207 I fpod    : updateFingerprintListeningState
03-07 11:11:39.795  2207  2207 I fpod    : mFingerprintRunningState = 0
03-07 11:11:39.795  2207  2207 D KeyguardFace: shouldListenForFace ( return false, Face is not default condition)
[... 58 lines cut ...]
03-07 11:11:40.033  3462  3462 I CNSS    : Failed to send nl message
03-07 11:11:40.039   728 23194 E qdmetadata: paramType 2048 not supported
03-07 11:11:40.040  3462  3462 I CNSS    : Failed to send nl message
03-07 11:11:40.044  3462  3462 I chatty  : uid=1000(system) /system/vendor/bin/cnss_diag identical 1 line
03-07 11:11:40.045  3462  3462 I CNSS    : Failed to send nl message
03-07 11:11:40.056   728   728 E qdmetadata: paramType 2048 not supported
03-07 11:11:40.158   728   728 E qdmetadata: paramType 2048 not supported
03-07 11:11:40.571 22787 22787 W charge_logger: type=1400 audit(0.0:20845983): avc: denied { remove_name } for name="charge_logger_a.csv" dev="sda15" ino=150 scontext=u:r:charge_logger:s0 tcontext=u:object_r:charge_logger_data_file:s0 tclass=dir permissive=0
03-07 11:11:40.617  3462  3462 I CNSS    : Failed to send nl message
```

Can you suggest me a way to resolve this problem? 

Thanks. 

### VisionCamera Version

3.9.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

GioMaugeri, 2024-03-07:
I resolve it in this way:

I removed the condition to show ScannerScreen component and I added this style to the View before camera component:

 <View style={{flex: 1, display: cameraOn ? 'flex' : 'none'}}>
 
 In this way I can use the same camera and adding this style the camere isn't stretched anymore, because the style refresh.
 
 I setted setEnableOnCodeScanned to true each time that I open the camera, so each time that cameraOn is true.

Hope it can help someoene.

closed 2024-03-07 by GioMaugeri

Sakura-echos, 2024-03-20:
it's not work for me