margelo/react-native-vision-camera #2641: 💭 Capturing iPhone photos in max resolution

stevengoldberg, 2024-03-08:
### Question

Not sure if this is a bug, a feature request, or a question, sorry!

1. On an iPhone 14 Pro or 15 Pro, I can enable resolution control in the camera settings. This allows me to take 24 or 48MP photos in HEIF or JPEG using the wide angle camera. But when I pass an array of physical devices in `vision camera`, like this:

```
const physicalDevices =
        cameraPosition === 'back'
            ? [
                  'ultra-wide-angle-camera',
                  'wide-angle-camera',
                  'telephoto-camera',
              ]
            : ['wide-angle-camera']
    const device = useCameraDevice(cameraPosition, { physicalDevices })
```

There is no format available with a resolution higher than 3024x4032. However, when I include only `wide-angle-camera` in the `physicalDevices` array, more formats are available, including some with 6048x8064 resolution. Is this expected? Is it possible to use a multi-camera device but still use the 48MP resolution when the wide angle camera is active?

2. When I pass only `wide-angle-camera` and select the highest resolution format, which looks like this:

```
autoFocusSystem: "phase-detection"
fieldOfView: 72.044113
maxFps: 30
maxISO: 12320
maxZoom: 189
minFps: 1
minISO: 55
photoHeight: 6048
photoWidth: 8064
pixelFormats: (9) ['yuv', 'yuv', 'rgb', 'unknown', 'yuv', 'unknown', 'yuv', 'unknown', 'rgb']
supportsDepthCapture: false
supportsPhotoHdr: false
supportsVideoHdr: false
videoHeight: 3024
videoStabilizationModes: (2) ['auto', 'off']
videoWidth: 4032
```

The result of `Camera.takePhoto` (taken at neutral zoom) only has 3024x4032 resolution. Is this expected? Is it possible to take photos in higher resolution?

### VisionCamera Version

3.9.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-04-22:
High res would be RAW, I think.

closed 2024-04-22 by mrousavy

stevengoldberg, 2024-04-23:
Not a high-priority issue IMO, but I don't believe that's the case. The iPhone 15 can capture 24 or 48 MP photos as HEIF or JPEG. I just checked with vision camera 4.0.0-beta.8 and everything is still as I reported it above.

mrousavy, 2024-04-23:
48 MP is HEIF - 24 MP is JPEG.
Currently only JPEG is supported, HEIF isn't.
Then it's not RAW but rather HEIF that needs to be supported.
Can you create a feature request?