margelo/react-native-vision-camera #2642: 💭 ISO and taking sharp photos

stevengoldberg, 2024-03-08:
### Question

I'm revisiting the issue of how to take the sharpest photo possible while displaying the result as quickly as possible, and I'm confused about the results I'm seeing.

The following are all cropped segments of photos taken on an iPhone 14 Pro camera at neutral zoom. In all cases I am passing `enableHighQualityPhotos` to the `Camera` component and `photoQualityPrioritization` is set to `'quality'`.

This is the result from the native iOS camera app:

[image]

The original photo was 3024x4032, ISO 1000, 24mm, f.178, 1/30s exposure.

This is the result from my `vision-camera` app using this format:

```
autoFocusSystem: "phase-detection"
fieldOfView: 101.021591
maxFps: 60
maxISO: 5472
maxZoom: 171.875
minFps: 1
minISO: 57
photoHeight: 3024
photoWidth: 4032
pixelFormats: (9) ['yuv', 'yuv', 'rgb', 'unknown', 'yuv', 'unknown', 'yuv', 'unknown', 'rgb']
supportsDepthCapture: false
supportsPhotoHdr: false
supportsVideoHdr: false
videoHeight: 768
videoStabilizationModes: (3) ['auto', 'off', 'standard']
videoWidth: 1024
```
[image]

The image is much blurrier. It's the same resolution, same camera, same everything except the ISO is 3200 and the exposure is 1/90s.

To take sharper photos, I've been adding `{ iso: 'max' }` to the format filter. Doing so selects the following format:

```
autoFocusSystem: "phase-detection"
fieldOfView: 103.62532
maxFps: 30
maxISO: 12768
maxZoom: 189
minFps: 1
minISO: 57
photoHeight: 3024
photoWidth: 4032
pixelFormats: (9) ['yuv', 'yuv', 'rgb', 'unknown', 'yuv', 'unknown', 'yuv', 'unknown', 'rgb']
supportsDepthCapture: false
supportsPhotoHdr: false
supportsVideoHdr: false
videoHeight: 3024 videoStabilizationModes: (2) ['auto', 'off']
videoWidth: 4032
```

Taking a photo with this format is noticeably slower — slower than using the native app, I think, but I can't measure it to be sure. [Here is a screen recording](https://imgur.com/a/bJ38Lf3) demonstrating the difference in capture speed between the two formats. But the result looks similar to the native app:

[image]

And sure enough, this photo has the same specs as the one taken with the native app — ISO 1000, 1/30s exposure.

And just to be thorough, I took one with ShadowLens as well (after raising the exposure back to 0):

[image]

There's no lens information available, but it looks roughly in between the two results taken in my app.

Now, what confuses me is: why does selecting the format with the higher max ISO result in a sharper photo with a lower ISO value? These ISO values are nowhere close to the min or max stated by the formats. Is there any way to produce the sharp result without the capture slowing down so much?

### VisionCamera Version

3.9.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

stevengoldberg, 2024-03-11:
Thanks for the reply — that makes sense re the perceived difference in capture speed. 

The thing I’m still confused about is why choosing the format with a higher max iso results in a photo with a *lower* ISO according to the EXIF. Why do I need a max ISO of 12768 to take a photo with 1000 ISO? Is there some other difference between the formats that isn’t being reported, and the max ISO difference is merely coincidental?

mrousavy, 2024-03-11:
Might also be the case, yes. I don't know what differences these are though

mrousavy, 2024-03-19:
Here's some additional format details from all cameras on my iPhone 15 Pro:

```
4032x3024 | 192x144@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 192x144@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
3696x3024 | 352x288@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
3696x3024 | 352x288@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 480x360@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 480x360@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 640x480@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 640x480@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
2016x1512 | 640x480@60.0 | Binned: true | EDR: false | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.5e-05..1.0 | Focus: phase-detection | Studio: false
2016x1512 | 640x480@60.0 | Binned: true | EDR: false | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 960x540@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 960x540@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 1024x768@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4032x3024 | 1024x768@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1280x720@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1280x720@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1280x720@30.0 | Binned: false | EDR: false | Tone Mapping: true (HLG_BT2020) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1280x720@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1280x720@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1280x720@60.0 | Binned: false | EDR: false | Tone Mapping: true (HLG_BT2020) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
2112x1188 | 1280x720@60.0 | Binned: true | EDR: false | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.5e-05..1.0 | Focus: phase-detection | Studio: false
2112x1188 | 1280x720@60.0 | Binned: true | EDR: false | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.5e-05..1.0 | Focus: phase-detection | Studio: false
2016x1512 | 1440x1080@60.0 | Binned: true | EDR: false | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.5e-05..1.0 | Focus: phase-detection | Studio: false
2016x1512 | 1440x1080@60.0 | Binned: true | EDR: false | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@30.0 | Binned: false | EDR: false | Tone Mapping: true (HLG_BT2020) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBHLG_BT2020LOG) | ISO: 55.0..5280.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
4224x2376 | 1920x1080@60.0 | Binned: false | EDR: false | Tone Mapping: true (HLG_BT2020) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
[... cut ...]
3840x2160 | 3840x2160@60.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBHLG_BT2020LOG) | ISO: 55.0..5280.0 | Exposure: 2.3e-05..1.0 | Focus: phase-detection | Studio: false
8064x6048 | 4032x3024@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGB) | ISO: 55.0..12320.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
8064x6048 | 4032x3024@30.0 | Binned: false | EDR: true | Tone Mapping: true (sRGBP3) | ISO: 55.0..12320.0 | Exposure: 1.5e-05..1.0 | Focus: phase-detection | Studio: false
```

mrousavy, 2024-04-22:
heyo - bumping this @stevengoldberg

stevengoldberg, 2024-04-23:
I'm afraid I don't have any more insight on this — I did skim through Apple's camera documentation, but didn't find anything that seemed relevant. In my app I'm back to using the `iso: 'max'` filter to capture with max quality, which generally seems to match the output of the native camera app.

TungE610, 2024-04-23:
> I'm afraid I don't have any more insight on this — I did skim through Apple's camera documentation, but didn't find anything that seemed relevant. In my app I'm back to using the `iso: 'max'` filter to capture with max quality, which generally seems to match the output of the native camera app.

Can you show me full format object ?

stevengoldberg, 2024-04-23:
> Can you show me full format object ?

Which one? I posted format objects in the original post of this thread.

TungE610, 2024-04-23:
The object which helps capture images with max quality, I tried your above objects but the image looks quite blurry and the speed also slows down every time I add format props. Thank you