margelo/react-native-vision-camera #2644: 🐛 Error while switching the camera from front to back and vice versa, when video recording is in-progress

MuhammadFaisal215, 2024-03-08:
### What's happening?

🐛 Error while switching the camera from front to back and vice versa, when video recording is in-progress.

Following error occurs when i switch the camera while video is recording:

{"cause": {"message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "stacktrace": "com.mrousavy.camera.InactiveSourceError: [capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:64)
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:52)
	at androidx.camera.video.Recorder$RecordingRecord.lambda$updateVideoRecordEvent$6$androidx-camera-video-Recorder$RecordingRecord(Recorder.java:2752)
	at androidx.camera.video.Recorder$RecordingRecord$$ExternalSyntheticLambda5.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:268)
	at android.app.ActivityThread.main(ActivityThread.java:8019)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:627)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:997)
"}, "code": "capture/inactive-source", "message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "userInfo": null}.

### Reproduceable Code

```tsx
<CameraVision
            ref={camera}
            style={StyleSheet.absoluteFill}
            device={device}
            audio={hasMicrophonePermission}
            isActive={
              !showPhotoPreview && !showVideoEditor && appState === 'active'
            }
            photo
            video
            enableZoomGesture
            orientation="portrait"
            enableHighQualityPhotos
          />
```

### Relevant log output

```shell
{"cause": {"message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "stacktrace": "com.mrousavy.camera.InactiveSourceError: [capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:64)
	at com.mrousavy.camera.CameraView_RecordVideoKt$startRecording$1.accept(CameraView+RecordVideo.kt:52)
	at androidx.camera.video.Recorder$RecordingRecord.lambda$updateVideoRecordEvent$6$androidx-camera-video-Recorder$RecordingRecord(Recorder.java:2752)
	at androidx.camera.video.Recorder$RecordingRecord$$ExternalSyntheticLambda5.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:268)
	at android.app.ActivityThread.main(ActivityThread.java:8019)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:627)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:997)
"}, "code": "capture/inactive-source", "message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "userInfo": null}
```

### Camera Device

```json
<CameraVision
            ref={camera}
            style={StyleSheet.absoluteFill}
            device={device}
            audio={hasMicrophonePermission}
            isActive={
              !showPhotoPreview && !showVideoEditor && appState === 'active'
            }
            photo
            video
            enableZoomGesture
            orientation="portrait"
            enableHighQualityPhotos
          />
```

### Device

All devices

### VisionCamera Version

2.16.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.