margelo/react-native-vision-camera #2647: 🐛  Camera invades statusbar in android

YoonJeongLulu, 2024-03-11:
### What's happening?

[image]

s shown in the picture above, the camera invades the status bar only on Android.

The code I wrote is as follows, but I added svg to mask part of the camera only in the center.

What I want is for the camera to only exist in the masked area. I need help.

I tried wrapping it in SafeAreaView and adding width and height to the Camera Component's style, but it didn't help.

* ScannerFrame means maskingSVG

### Reproduceable Code

```tsx
import { Pressable, SafeAreaView, View, ViewProps } from "react-native";
import { Text } from "@src/app/component/Text";
import { Camera } from "react-native-vision-camera";
import colors from "@src/constant/colors";
import CloseIcon from "@icon/24/line/times.svg";
import { useTranslation } from "react-i18next";
import { useSafeAreaInsets } from "react-native-safe-area-context";
import { useScanner } from "./useScanner";
import { ScannerFrame } from "./ScannerFrame";

export interface BaseHeaderProps extends ViewProps {
  className?: string;
  paddingTopBgColor?: string;
}

export function BaseHeader({
  children,
  className,
  style,
  paddingTopBgColor,
  ...props
}: BaseHeaderProps) {
  const insets = useSafeAreaInsets();
  return (
    <>
      <View
        style={{
          paddingTop: insets.top,
          backgroundColor: paddingTopBgColor,
        }}
      />
      <View className={className} {...props} style={style}>
        {children}
      </View>
    </>
  );
}

export function Scanner() {
  const {
    device,
    isFocused,
    isCameraAvailable,
    onCodeScanned,
    closeScanner,
    onCameraLayout,
    cameraViewSize,
    isInValidCode,
  } = useScanner();

  const { t } = useTranslation();

  return (
    <SafeAreaView>
      <BaseHeader />
      <View className="w-full h-full relative" onLayout={onCameraLayout}>
        <Pressable
          className="absolute right-[8px] top-[12px] w-[40px] h-[40px] z-10 flex items-center justify-center"
          onPress={closeScanner}>
          <CloseIcon
            width="24px"
            height="24px"
            color={colors.primary["over-white"]}
          />
        </Pressable>
        {device && isCameraAvailable ? (
          <View
            style={{
              flex: 1,
              position: "relative",
              backgroundColor: isCameraAvailable ? "none" : "bg-black",
            }}>
            {isCameraAvailable && (
              <Camera
                className="flex-1"
                device={device}
                isActive={isFocused}
                codeScanner={{
                  codeTypes: ["qr"],
                  onCodeScanned,
                }}
              />
            )}
            {isCameraAvailable && (
              <ScannerFrame
                width={cameraViewSize.width}
                height={cameraViewSize.height}
                isInvaild={isCameraAvailable && isInValidCode}
              />
            )}
          </View>
        ) : (
          <View className="flex-1 bg-over-gray-800" />
        )}
        <View className="w-full h-[228px] bg-primary-over-white p-[20px]">
          <Text en="O24b" ko="K24b" className="pt-[20px] pb-[16px]">
            {t("OWA-21-021.title")}
          </Text>
          <View className="bg-over-gray-25 px-[16px] py-[12px] rounded-[8px]">
            <Text en="O13r" ko="K13r" className="text-over-gray-500">
              {t("OWA-21-021.desc")}
            </Text>
          </View>
        </View>
      </View>
    </SafeAreaView>
  );
}
```

### Relevant log output

```shell
no log.
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy A24, SM-A245N

### VisionCamera Version

"react-native-vision-camera": "^3.9.0"

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

FranzPichardoR, 2024-07-31:
@mrousavy Hey, we are still experiencing this issue.  I am on version 4.2.1. The exact same thing that is happening to OP is happening to me.

Here are the logs:
`2024-07-31 15:45:57.013   276-5341  JPG                     and...e.camera.provider@2.4-service  D  L 491, jpg_sem_init: jpg_sem_init
2024-07-31 15:45:57.013   276-5341  JPG_OEM                 and...e.camera.provider@2.4-service  D  L 230, sprd_jpeg_init:  jpeg handle 0xae5f8380
2024-07-31 15:45:57.013   276-5341  JPG_OEM                 and...e.camera.provider@2.4-service  D  L 239, sprd_jpeg_init: done 0
2024-07-31 15:45:57.033   276-276   ae_sprd_adpt            and...e.camera.provider@2.4-service  E  4112, ae_parser_otp_info: ae otp_info_ptr is NULL
2024-07-31 15:45:57.033   276-276   ae_sprd_adpt            and...e.camera.provider@2.4-service  E  4164, ae_parser_otp_info: ae ae_otp_info_ptr = 0x0, module_info_ptr = 0x0. Parser fail !
2024-07-31 15:45:57.034   276-276   flash_lib               and...e.camera.provider@2.4-service  I  591, flash_init: FLASH VERSION: FLASH VERSION : 20181225-0900
2024-07-31 15:45:57.034   276-276   ae_lib_misc             and...e.camera.provider@2.4-service  I  121, ae_misc_init: AE2.1 branch: trunk VERSION : 20180526_1700_1_1
2024-07-31 15:45:57.035   276-276   deflicker               and...e.camera.provider@2.4-service  E  flicker detection library version:ver2.1a_20180130
2024-07-31 15:45:57.041   276-276   awb_adpt                and...e.camera.provider@2.4-service  E  948, _awb_parser_otp_info: awb otp info ptr is NULL
2024-07-31 15:45:57.041   276-276   awb_adpt                and...e.camera.provider@2.4-service  E  1018, _awb_parser_otp_info: awb awb_otp_info_ptr = 0x0, module_info_ptr = 0x0. Parser fail !
2024-07-31 15:45:57.041   276-276   AWBLIB                  and...e.camera.provider@2.4-service  E  awb version: AWB VERSION : 0x20180312-1000
2024-07-31 15:45:57.041   276-276   AWBLIB                  and...e.camera.provider@2.4-service  E  awb stat: 50x37
2024-07-31 15:45:57.041   276-276   AWBLIB                  and...e.camera.provider@2.4-service  E  awb otp: random(0,0,0) golden(0,0,0)
2024-07-31 15:45:57.048   276-276   ATM                     and...e.camera.provider@2.4-service  E  atm version: ATM VERSION : 0x20170830-1600
2024-07-31 15:45:57.049   276-276   lsc_ctrl                and...e.camera.provider@2.4-service  E  250, _lsc_parser_otp: lsc otp_info_ptr is NULL
2024-07-31 15:45:57.049   276-276   lsc_ctrl                and...e.camera.provider@2.4-service  E  328, _lsc_parser_otp: lsc module_info_ptr = 0x0. Parser fail !
2024-07-31 15:45:57.050   276-276   <no-tag>                and...e.camera.provider@2.4-service  I  [ALSC]: L 911, lsc_adv_init_alg2: LSC_VERSION: W2018.05.03_update_alsc_version !
2024-07-31 15:45:57.058  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.058  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.059  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.060  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.060  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.061  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.062  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.062  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.062  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.062  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.063  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.063  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.063  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.064  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.064  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.064  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.066  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.066  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.066  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  number_of_cameras:1
2024-07-31 15:45:57.067  4296-5012  Camera-JNI              com.suncard.sunchatmobile            I  scan_packages:
2024-07-31 15:45:57.067  4296-4383  CameraDevices           com.suncard.sunchatmobile            I  Camera #0 is now unavailable.
2024-07-31 15:45:57.067  4296-4388  Camera2CameraImpl       com.suncard.sunchatmobile            D  {Camera@7aa5233[id=0]} Use case androidx.camera.core.Preview-28799934-e468-45bd-917c-c20a42b7b5af154723875 ACTIVE
[... cut ...] de=1280, flags=4, usage=0x3, size=1382400, width=1280, height=720, internalWidth=1280, internalHeight=720, stride=1280, base=0xa1c80000, writeOwner=0, attr_base=0xffffffff, yuv_info=1, fd=-1, buf_idx=0, offset=0, min_pgsz=4096, phyaddr=0
2024-07-31 15:45:57.713  4296-5328  CameraView              com.suncard.sunchatmobile            I  invokeOnAverageFpsChanged(0.0)
2024-07-31 15:45:58.712  4296-5328  CameraView              com.suncard.sunchatmobile            I  invokeOnAverageFpsChanged(0.0)`