margelo/react-native-vision-camera #2648: Video recording does not record audio🐛

redundancer, 2024-03-11:
### What's happening?

I implemented video recording, but it does not record audio on some devices

### Reproduceable Code

```tsx
const cameraRef = useRef<Camera>(null);
const frontCamera = useCameraDevice('front');

const [currentCamera, setCurrentCamera] = useState(frontCamera);
const format = useCameraFormat(currentCamera, Templates.Snapchat)

<Camera
        ref={cameraRef}
        device={currentCamera}
        isActive={true}
        style={{flex: 1}}
        format={format}
        fps={30}
        video={true}
        audio={true}
        resizeMode="contain"
      />

const startRecording = useCallback(() => {
    cameraRef.current?.startRecording({
      videoBitRate: 'normal',
      videoCodec: 'h265',
      fileType: 'mp4',
      onRecordingError,
      onRecordingFinished,
    });
  }, [onRecordingError, onRecordingFinished]);
```

### Relevant log output

```shell
Here are the adb logcat outputs when using the camera vision example app

[14:41:02] I | ReactNativeJS ▶︎ Running "VisionCameraExample" with {"rootTag":11}

[14:41:02] I | ReactNativeJS ▶︎ Loading react-native-worklets-core...

[14:41:02] I | ReactNativeJS ▶︎ Worklets loaded successfully

[14:41:02] I | ReactNativeJS ▶︎ Re-rendering Navigator. Camera: denied | Microphone: denied

[14:41:03] I | ReactNative ▶︎ [GESTURE HANDLER] Initialize gesture handler for root view com.facebook.react.ReactRootView{10319 V.E...... ......ID 0,0-1080,2208 #b}

[14:41:14] I | ReactNativeJS ▶︎ Requesting camera permission...

[14:41:16] I | ReactNativeJS ▶︎ Camera permission status: granted

[14:41:17] I | ReactNativeJS ▶︎ Requesting microphone permission...

[14:41:18] I | ReactNativeJS ▶︎ Microphone permission status: granted

[14:41:20] I | ReactNativeJS ▶︎ Camera: BACK (0) | Format: (3952x1824 photo / 2340x1080@30 video @ 30fps)

[14:41:20] I | ReactNativeJS ▶︎ Camera initialized!

[14:41:22] I | ReactNativeJS ▶︎ 11268112970423: 2340x1080 yuv Frame (portrait)

[14:41:23] I | ReactNativeJS ▶︎ 11268150161568: 2340x1080 yuv Frame (portrait)

[14:41:23] I | ReactNativeJS ▶︎ 11268183494120: 2340x1080 yuv Frame (portrait)

[14:41:23] I | ReactNativeJS ▶︎ 11268214017223: 2340x1080 yuv Frame (portrait)
[... 683 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Fairphone 4 (/e/OS v1.20, Android 12)

### VisionCamera Version

 3.8.2, 3.9.0, 3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-11:
Did you try with codec 264?

redundancer, 2024-03-11:
Yes, it does not work either

mauriceKalevra, 2024-03-11:
+1 pixel 3a

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

mrousavy, 2024-03-19:
amazing. If you appreciate my time and dedication for VisionCamera, please consider sponsoring me on GitHub :)

closed 2024-03-19 by mrousavy