margelo/react-native-vision-camera #2650: 💭 Is it not possible to get the binary of the QR code?

Yuuki47, 2024-03-12:
### Question

Hello
I am a pharmacist in Japan.

I would like to create an app that reads divided QR codes that contain medication history, which are distributed in Japan.

However, the read text data does not include information such as how many parts it has been divided into or what number the read data is.

However, we discovered that by analyzing the binary of a QR code, we can obtain information such as how many parts it is divided into and what data number it is.
↓Sorry for the Japanese site...
https://qiita.com/Toshi_Tab/items/e528145b9e4f813a3cf9

Is it not possible to obtain the QR code binary with react-native-vision-camera?

Thank you for reading to the end!
I'm not good at English, so I use a translation function to convert Japanese to English.
I apologize if my expressions are difficult to understand.
Thank you.

C.C-Yuuki

### What I tried

↓A sample of the QR code that I actually want to read
[image]

### VisionCamera Version

"^3.8.2"

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

xulihang, 2024-03-19:
[Dynamsoft Barcode Reader](https://github.com/tony-xlh/vision-camera-dynamsoft-document-normalizer) can return the bytes of barcodes.

Then, we can use this reading library to get the data: https://www.npmjs.com/package/barcode-data-reader 

The QR codes you provided use the "Structured Append" mode which stores data in several QR codes and this library can process it. You can use this online demo to have a try: https://tony-xlh.github.io/barcode-data-reader/

Yuuki47, 2024-03-24:
Mr.xulihang
thank you for your reply😁
I'll try out the library you suggested❗

mrousavy, 2024-04-02:
feature request created: https://github.com/mrousavy/react-native-vision-camera/issues/2701

closed 2024-04-02 by mrousavy