margelo/react-native-vision-camera #2651: Video={true}, the camera does not load. 4.0.0-beta.5

sberkeo, 2024-03-12:
### What's happening?

When you try to change the camera when video={true}, the camera does not load. It works fine only when photo={true}.

### Reproduceable Code

```tsx
<Camera
                    key={cameraKey}
                    style={
                        isCameraInitialized
                            ? {
                                position: "absolute",
                                top: 0,
                                left: 0,
                                right: 0,
                                bottom: 0,
                                ...StyleSheet.absoluteFill, zIndex: isVisionCameraActive ? 995 : -9, flex: 1, width: windowWidth, height: windowHeight
                            }
                            : {
                                width: 0,
                                height: 0,
                            }
                    }
                    //style={{ ...StyleSheet.absoluteFill, zIndex: isVisionCameraActive ? 995 : -9, flex: 1, width: windowWidth, height: windowHeight, }}
                    ref={cameravision}
                    device={device}
                    photo={true}
                    //video={true}
                    //audio={true}
                    //pixelFormat="native"
                    //resizeMode="contain"
                    fps={fps}
                    videoHdr={hdrMode}
                    photoHdr={hdrMode}
                    lowLightBoost={device.supportsLowLightBoost ? lowlightMode : false}
                    videoStabilizationMode={format.videoStabilizationModes.includes('cinematic-extended') ? videostabilizationMode : 'off'}
                    enableHighQualityPhotos
                    enableZoomGesture
                    isActive={isVisionCameraActive}
                    focusable={true}
                    orientation={'portrait'}
                    format={format}
                    onInitialized={() => {

                        if (!isCameraInitialized) {
                            // Kamera başlatılmamışsa ve platform iOS değilse
                            setTimeout(() => {
                                setIsCameraInitialized(true);
                            }, 50);
                        }

                    }}

                />
```

### Relevant log output

```shell
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.CameraView_EventsKt.sendEvent(CameraView+Events.kt:124)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.CameraView_EventsKt.invokeOnError(CameraView+Events.kt:69)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.CameraView.onError(CameraView.kt:278)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.core.CameraSession$configureCamera$3.invoke(CameraSession.kt:387)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.core.CameraSession$configureCamera$3.invoke(CameraSession.kt:375)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.core.CameraSession.configureCamera$lambda$8(CameraSession.kt:375)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.core.CameraSession.$r8$lambda$4Z3XBXeZe7nK1umYnjsgdpFZMVY(Unknown Source:0)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at com.mrousavy.camera.core.CameraSession$$ExternalSyntheticLambda1.onChanged(Unknown Source:2)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at androidx.lifecycle.LiveData.considerNotify(LiveData.java:133)
03-12 23:01:09.688 13544 13544 E unknown:com.facebook.react.uimanager.UIManagerHelper:  at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:151)
[... 140 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy A70 (Android 11)

### VisionCamera Version

4.0.0-beta.5

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-14:
Fixed in https://github.com/mrousavy/react-native-vision-camera/commit/5283020840751d174a3f82a97ceea78b875cc5d5! 💪🥳

closed 2024-03-14 by mrousavy

sberkeo, 2024-03-14:
Thank you for your solution, but unfortunately the problem still persists.

mrousavy, 2024-03-14:
well, I don't know. I already spent too much time on this (about 2 weeks just for this bug) so I'm not sure what else to try. Also it seems like no one wants to pay/sponsor to get this fixed

mrousavy, 2024-03-21:
Awesome, great to hear - thanks!