margelo/react-native-vision-camera #2657: [Android] - v3.9.1 Crash on component unmount when frame processor is running🐛

ldstein, 2024-03-18:
### What's happening?

I'm in the process of upgrading an App from RNVC v3.4.0. 

The app implements a custom Frame Processor plugin using MLKit for barcode scanning. Nothing advanced - just some QR code detection plus boundary co-ordinates used for overlay drawing.

Upgrading RNVC to v3.9.1 introduced an intermittent crash when either:
- The frame processor prop is toggled between a fn and null value.
- The camera component is unmounted

To demonstrate the crash, I have added a minimal Barcode frame processor to the Example App:
https://github.com/ldstein/react-native-vision-camera/tree/crash-test

There is a new icon (Green bug icon) which toggles rendering the Camera component.  Continuously tapping the bug icon will eventually lead to a crash.

Appreciate any pointers if the issue is with my frame processor implementation. Has been stable in RNVC 3.40, but that was many moons ago.
https://github.com/ldstein/react-native-vision-camera/blob/crash-test/package/example/android/app/src/main/java/com/mrousavy/camera/example/ExampleQRScannerPlugin.java

### Reproduceable Code

```tsx
https://github.com/ldstein/react-native-vision-camera/tree/crash-test
```

### Relevant log output

```shell
2024-03-18 18:04:24.786  1592-1659  ActivityTaskManager     system_server                        W  Activity top resumed state loss timeout for ActivityRecord{7361abc u0 com.mrousavy.camera.example/.MainActivity t-1 f}}
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 7,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 3, Pixel 3 XL

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

jimsideout, 2024-03-19:
@mrousavy this appears to be resolved in the beta. Thank you!

ldstein, 2024-03-19:
Hi @mrousavy 

Still seeing a crash in [v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7) when Camera component is unmounted and frame processor is running. 

But it no longer crashes in  [v4.0.0-beta.8](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.8)! Camera component unmount also appears much faster/smooter.

Thanks!

mrousavy, 2024-03-21:
Great to hear!

mrousavy, 2024-03-25:
Thanks guys, great to hear that V4 now works perfectly! If you appreciate my work, please consider [💖 sponsoring me on GitHub 💖](http://github.com/sponsors/mrousavy/) so I can keep maintaining this library, fixing bugs and building new features! :)

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy