margelo/react-native-vision-camera #2663: 🐛 [Android] Flash is removed before the actual capture, resulting in a dark photo

AndriyZahura, 2024-03-18:
### What's happening?

Hi! First of all, great lib, thanks a lot for what you're doing! 

My issue is:
When I set the `torch='on'` on my Samsung, and do `takePhoto` (either with or without `{flash: 'on'}`), the torch/flash is present, but I believe it gets off just before the capture, which results in a dark photo. In case when torch is `on`, it also gets turned off, and then turns on again after the photo capture. You can see the result on the video under. In the video, you can see the torch, but believe me, it's almost the same when I use the `flash: 'on'` (except that with flash it takes extra time to capture, see the https://github.com/mrousavy/react-native-vision-camera/issues/2577).

**Important**
For me, this is only reproducible on some Androids. For me, it's my Samsung Galaxy S21 FE(SM-G990B/DS) and Xiaomi Note 9 Pro. It works just fine on another android - Nothing Phone 1, , and on iOS. Tomorrow, I can test it on another Samsung Galaxy - S20

I was wondering if I am doing something wrong, because my issue is basically a duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2239. But in that case, the reporter managed to fix the dark photos by simply doing the `flash: 'on'`, which doesn't work for me. Will be grateful for any reply! 

The video:

https://github.com/mrousavy/react-native-vision-camera/assets/55994435/05151bde-7dcd-49d6-8f2f-036f62dff437

### Reproduceable Code

```tsx
const refCamera = useRef<Camera>(null);
  const additionalCameraStyle = IS_ANDROID
    ? {
        opacity: isCarouselVisible ? 0 : 1,
        zIndex: -1,
      }
    : {zIndex: -1};
  const device = useCameraDevice('back');

//Basically, torch is on when you press the flash icon. I did try the same with torch on all the time, the result is the same, so you can ignore this
  const showTorch = IS_ANDROID && flashMode === FlashMode.ENABLED && !isCarouselVisible ? 'on' : 'off';

//This is simplified, please tell if you need more info
  const onMakePhoto = async () => {
    const photo: PhotoFile = await refCamera.current.takePhoto({
      //I tried both 'on', 'off', and 'auto', nothing helps
      flash: IS_ANDROID ? 'off' : flashMode,
      enableShutterSound: false,
      //I tried both 'quality', 'speed', and 'balanced', nothing helps
      qualityPrioritization: 'balanced',
    });
  };

//I did try to remove all the unnecessary props, didn't help
            <Camera
              ref={refCamera}
              photo={true}
              style={{
                ...StyleSheet.absoluteFillObject,
                ...additionalCameraStyle,
                marginTop: IS_ANDROID && isCarouselVisible ? MARGIN_CAMERA_STATUSBAR : 0,
              }}
              device={device}
              enableZoomGesture={true}
              torch={showTorch}
              isActive={isVisible}
              onError={onCameraError}
              onLayout={() => setIsLoadingCamera(true)}
              onInitialized={() => setIsLoadingCamera(false)}
            />

                <TouchableOpacity
                  disabled={disabledFlash}
                  activeOpacity={1}
                  onPress={onMakePhoto}
                  style={styles.makePhotoButton}>
                  <Box style={styles.makePhotoInnerButton} />
                </TouchableOpacity>
```

### Relevant log output
[... 247 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy S21 FE(SM-G990B/DS)

### VisionCamera Version

^3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

impodracer, 2024-03-19:
I have just the same issue with v3.9.1 on Pixel 2 XL / Android 11 😢

mrousavy, 2024-03-19:
Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see if that fixes the issue for you? 😅 

```
yarn add react-native-vision-camera@beta
```

You might need to increase the `compileSdk` to the newest Android SDK (34) if you are behind.

AndriyZahura, 2024-03-19:
> Hey! I just released a new V4 beta ([v4.0.0-beta.7](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0-beta.7)) where I fixed a bunch of issues! 💪🚀 Can you test that and see [...]

Thanks a lot, it does fix it indeed! I will also try on another device which was not working. Can you tell me, approximately, when will the v4.0.0 be released? Or what precise commit fixes this issue? I can patch the library locally just to fix that issue, since I can't upgrade to SDK 34 atm.

daimonkor, 2024-03-20:
But please check `codeScanner`, beta version failed with current feature (ignore codes)

AndriyZahura, 2024-03-21:
> Thanks a lot, it does fix it indeed! I will also try on another device which was not working. Can you tell me, approximately, when will the v4.0.0 be released? Or what precise commit fixes this [...]

Update: 4.0.0-beta.8 fixed the issue for Nothing Phone 1 (previously, it was working fine only with torch, the flash was giving a dark image, now it's okay with flash). But it still results in a dark picture for my Samsung S21 FE, both from flash and torch. I guess it's heavy dependent on the phone and camera device

mrousavy, 2024-03-21:
> Thanks a lot, it does fix it indeed! I will also try on another device which was not working. Can you tell me, approximately, when will the v4.0.0 be released? Or what precise commit fixes this issue? I can patch the library locally just to fix that issue, since I can't upgrade to SDK 34 atm.

Great to hear everything works fine! A lot of research went into this. V4 will take a few more days or weeks of testing.

>  I can't upgrade to SDK 34 atm.

Why not? This is a simple non-breaking change in every app. Everytime Android releases a new SDK, you have to bump your target SDK to that new SDK to use the latest features anyways.

daimonkor, 2024-03-21:
> @daimonkor I have no idea what you meant with that message, what does "beta version failed with current feature" mean?
> 
> There was a bug that Images are not being released properly, I fixed that in latest beta.

it is simple - nothing happed, ignore codeScanner handler when I tried show qr code, no error, no warning. But if I returned 3.9.1 - codeScanner works great. Please try check hello world qr code scanner with help 4.0.0 betas builds. thanks a lot for help.

mrousavy, 2024-03-21:
please try check latest V4 beta.

daimonkor, 2024-03-21:
> please try check latest V4 beta.

Greate, it works good, thanks a lot 4.0.0-beta.10

arslanafzal077, 2024-03-21:
i tried beta version but there is black screen no preview .so i could not go further to check if the issue still persist or not

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

minhkwan, 2024-05-05:
Thanks a lot for this library.
My application is in the production phase and it is using 3.9.2. 
I'm afraid that 4.0 is just newly released, it will have some bugs that I / we can't anticipate. So do you have any pr / commit to fix this in 3.9.2? Or it was fixed by rewriting all the CameraX which you do in 4.0.0?
Tks!

mrousavy, 2024-05-06:
Nice to hear VisionCamera being used in production everywhere.

This was fixed in V4, not in V3.