margelo/react-native-vision-camera #2668: Do I need Firebase to use Face Detection plug in?

frodriguez-hu, 2024-03-19:
### Question

Hey! One question! Do I need firebase in order to use this plugin? 

`https://github.com/rodgomesc/vision-camera-face-detector`

I am also trying to use `face-api` with this library and this one:
`https://github.com/mrousavy/react-native-blob-jsi-helper` 
In order to transform resized images buffers in blobs so In can send them to the face detection api from face api, is that possible?

### What I tried

-

### VisionCamera Version

3.9.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

frodriguez-hu, 2024-03-19:
I realized that I do not need firebase cause MLKit is free.

But my other question is remaining, is it possible to use `face-api`?

mrousavy, 2024-04-22:
Yes you can build a native Frame Processor plugin to use any native Obj-C/Swift/Java/Kotlin library.